openwrt/openwrt #9378: stub error

Theta1009, 2022-03-02:
hi guys
i have this problem can some one give me solution ?

Create index file './feeds/telephony.index'
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... failed.

Build dependency: Failed to install stub

/home/test/openwrt/include/prereq.mk:9: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/test/openwrt/include/toplevel.mk:179: recipe for target 'staging_dir/host/.prereq-build' failed

ynezz, 2022-03-08:
Your host is missing `/bin/true` binary, for example on Debian it's part of `coreutils` package, check https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem

closed 2022-03-08 by ynezz

EyitopeIO, 2022-07-24:
That's not the case when using Ubuntu 22. I have confirmed the binary is actually in `/bin/true`, yet I cannot proceed.

```
eyi@Nihinlola:~/Openwrt/x-wrt$ sudo apt install coreutils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
coreutils is already the newest version (8.32-4.1ubuntu1).
The following packages were automatically installed and are no longer required:
  chromium-codecs-ffmpeg-extra gstreamer1.0-vaapi libgstreamer-plugins-bad1.0-0 libminizip1
  libva-wayland2 python-pkg-resources python-setuptools
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 93 not to upgrade.
eyi@Nihinlola:~/Openwrt/x-wrt$ uname -r
5.15.0-40-generic
eyi@Nihinlola:~/Openwrt/x-wrt$ 

```

bersbersbers, 2022-09-10:
This fixed it for me (building 21.02.3): https://github.com/openwrt/openwrt/commit/6a5b4228e30244b44a49f523dea66caf3fbe3307

bersbersbers, 2022-12-29:
For those building 21.02.5, this has been backported in bd84d517eb95df96d6e9dccac387a135a6cc4065

systemcrash, 2023-01-09:
Just got this error on a fresh clone and `make menuconfig` for 22.03.3. 

My Dockerfile for the buildhost uses `FROM debian:buster`...

SirLouen, 2023-03-31:
> For those building 21.02.5, this has been backported in [bd84d51](https://github.com/openwrt/openwrt/commit/bd84d517eb95df96d6e9dccac387a135a6cc4065)

I've tried backporting to 19.07 and I still see this error

```
Create index file './feeds/telephony.index'
Prerequisite check failed. Use FORCE=1 to override.
gmake: *** [/home/mcamargo/src/openwrt/include/toplevel.mk:177: staging_dir/host/.prereq-build] Error 1
Updating feed 'freifunk' from 'https://github.com/freifunk/openwrt-packages.git;openwrt-19.07' ...
Already up to date.
```