openwrt/openwrt #9379: [Totolink X500R] no wireless channel selection in 5GHz AX mode

dfateyev, 2022-03-02:
The 5GHz wireless interface settings in Totolink X5000R in AX mode don't have a wireless channel option:

---
[image]

---
Therefore, after changing and saving any wireless parameters, the channel parameter cannot be set and stored to config.

Missing wireless channel parameter prevents any further interface settings manipulations:

---
[image]

---
The workaround is to check and manually set channel settings before and after interface parameters change, but better to have it fixed eventually.
Can be reproduced under clean 21.02.1 and 21.02.2 install.

dfateyev, 2022-03-06:
@981213 probably you have some information about it (as the initial commiter)?

981213, 2022-03-06:
I doubt if OpenWrt 21.02 has proper 11ax support. I remember 11ax support wasn't done when 21.02 branched off.
I'm using this router with the master snapshot image and wireless channel shows up properly in LuCI.

dfateyev, 2022-03-06:
> I doubt if OpenWrt 21.02 has proper 11ax support. I remember 11ax support wasn't done when 21.02 branched off. I'm using this router with the master snapshot image and wireless channel shows up properly in LuCI.

Thank you for the information. I managed to use 802.11ax in 21.02 specifying all needed parameters in the "wireless" config directly. Will wait for official 22.x builds for full 802.11ax support out of box.