openwrt/openwrt #9392: wireless-regdb: update to version 2022.02.18 will cause some problems for MCS staying in 0

andrewlsa, 2022-03-04:
https://github.com/openwrt/openwrt/commit/19a90262df89f8419b058cf9a00dc9e8d465088b
pls review this commit.

trounticonan, 2022-03-05:
Is it related to [this commit in wireless-regdb](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/commit/?id=e427ff2a592e26fc1e8336769b9a1ad223f6f697)?
What happens if you try another country code?

andrewlsa, 2022-03-05:
functions back to normal when I reverted this commit by exclusion.

andrewlsa, 2022-03-05:
> What happens if you try another country code?
it works!

[image]

I used to changing it to CN for 160mhz.
The problem solved when changed back to Driver Default.

trounticonan, 2022-03-05:
CC @FlyGoat (Author of commit e427ff2 in wireless-regdb)

andrewlsa, 2022-03-05:
But I still think it would cause some troubles for those who using 160mhz while CN needed.

FlyGoat, 2022-03-06:
No idea what's going wrong. I was only trying to correct the database with current regulations.

hauke, 2022-03-06:
Channel 36 is at 5180 MHz, this channel now needs DFS in China. 
Which driver are you using?

andrewlsa, 2022-03-06:
> No idea what's going wrong. I was only trying to correct the database with current regulations.

It’s working fine now while I correct country code to world 00 or another one except China.

andrewlsa, 2022-03-06:
> Channel 36 is at 5180 MHz, this channel needs DFS in China now.
> Which driver are you using?

5180 not DFS. I’ve changed it to driver default, not CN. It works fine.

FlyGoat, 2022-03-06:
> Channel 36 is at 5180 MHz, this channel needs DFS in China now. Which driver are you using?

Not mandatory if you transmitting power is below 20 dBm.

FlyGoat, 2022-03-06:
> > Channel 36 is at 5180 MHz, this channel needs DFS in China now.
> > Which driver are you using?
> 
> 5180 not DFS. I’ve changed it to driver default, not CN. It works fine.

If it's not working with CN then it's still a bug.

mans0n, 2022-03-13:
Sorry, I could not reproduce the problem with my devices.
It took about 1 minute to turn on wlan in DFS channel, but after that the MCS was fine.

andrewlsa, 2022-03-13:
> Sorry, I could not reproduce the problem with my devices.
> It took about 1 minute to turn on wlan in DFS channel, but after that the MCS was fine.

Pls try country code set to CN.

mans0n, 2022-03-13:
I tried both CN and KR but the result was same.

Could you check:
* whether channel width affects (VHT20, VHT80, VHT160)
* whether DFS channel affects (channel 36, channel 149)
* whether reverting 19a90262df89 affects the above results

daiyeqi, 2023-02-07:
> Channel 36 is at 5180 MHz, this channel needs DFS in China now. Which driver are you using?

https://en.wikipedia.org/wiki/List_of_WLAN_channels#China

China MIIT expanded allowed channels as of 31 December 2012 to add UNII-1, 5150–5250 GHz, UNII-2, 5250–5350 GHz (DFS/TPC), similar to European standards EN 301.893 V1.7.1.[[68]](https://en.wikipedia.org/wiki/List_of_WLAN_channels#cite_note-79) China MIIT expanded allowed channels as of 3 July 2017 to add UNII-3, 5725—5850 MHz. [[69]](https://en.wikipedia.org/wiki/List_of_WLAN_channels#cite_note-80)