openwrt/openwrt #9398: mt7622: 5G Wifi fails when using 160MHz channel  and selecting a channel from 36 to 48 in wireless configuration.

okibcn, 2022-03-05:
Using a Linksys E8450 (mediatek mt7622) 5G Wifi fails when using 160MHz channel in the low contiguous band (CH36-64) using **CH36 to CH48** as the configured channel in the configuration. However, the 160MHz connection from CH36 to CH64 works when selecting any channel from CH52 to CH64. There is no problem when configuring channels from 100 to 128.

In all 8 cases, the system log is the same, but wifi is not available to the clients when using the lower 4 channels in the configuration. Maybe the fact that the lower 4 IEEE channels are not DFS could be something.

In any case, this is something new, as you can see it happening for instance in today's SNAPSHOT r19040-247eaa4416 but not in older SNAPSHOT r18646-3869ccbcc8.

The system  log reports the expected information but 5G is not working in this case:

```
Sat Mar  5 01:15:33 2022 daemon.notice hostapd: wlan1: interface state HT_SCAN->DFS
Sat Mar  5 01:15:33 2022 daemon.notice hostapd: wlan1: DFS-CAC-START freq=5180 chan=36 sec_chan=1, width=2, seg0=50, seg1=0, cac_time=60s
Sat Mar  5 01:15:34 2022 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Mar  5 01:16:35 2022 daemon.notice hostapd: wlan1: DFS-CAC-COMPLETED success=1 freq=5180 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5250 cf2=0
Sat Mar  5 01:16:35 2022 daemon.warn hostapd: Can't set DFS state for freq 5180 MHz
Sat Mar  5 01:16:35 2022 daemon.warn hostapd: Can't set DFS state for freq 5200 MHz
Sat Mar  5 01:16:35 2022 daemon.warn hostapd: Can't set DFS state for freq 5220 MHz
Sat Mar  5 01:16:35 2022 daemon.warn hostapd: Can't set DFS state for freq 5240 MHz
Sat Mar  5 01:16:35 2022 kern.info kernel: [ 2845.222641] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Sat Mar  5 01:16:35 2022 kern.info kernel: [ 2845.229167] br-lan: port 6(wlan1) entered blocking state
Sat Mar  5 01:16:35 2022 kern.info kernel: [ 2845.234530] br-lan: port 6(wlan1) entered forwarding state
Sat Mar  5 01:16:35 2022 daemon.notice netifd: Network device 'wlan1' link is up
Sat Mar  5 01:16:35 2022 daemon.notice hostapd: wlan1: interface state DFS->ENABLED
Sat Mar  5 01:16:35 2022 daemon.notice hostapd: wlan1: AP-ENABLED
```

However, When using any channel 52, 56, 60 or 64 in the config file everything works fine, and the system log is the same:

```
Sat Mar  5 01:05:10 2022 daemon.notice hostapd: wlan1: interface state HT_SCAN->DFS
Sat Mar  5 01:05:10 2022 daemon.notice hostapd: wlan1: DFS-CAC-START freq=5280 chan=56 sec_chan=-1, width=2, seg0=50, seg1=0, cac_time=60s
Sat Mar  5 01:05:10 2022 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Mar  5 01:06:15 2022 daemon.notice hostapd: wlan1: DFS-CAC-COMPLETED success=1 freq=5280 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5250 cf2=0
Sat Mar  5 01:06:15 2022 daemon.warn hostapd: Can't set DFS state for freq 5180 MHz
Sat Mar  5 01:06:15 2022 daemon.warn hostapd: Can't set DFS state for freq 5200 MHz
Sat Mar  5 01:06:15 2022 daemon.warn hostapd: Can't set DFS state for freq 5220 MHz
Sat Mar  5 01:06:15 2022 daemon.warn hostapd: Can't set DFS state for freq 5240 MHz
Sat Mar  5 01:06:15 2022 kern.info kernel: [ 2225.676669] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Sat Mar  5 01:06:15 2022 kern.info kernel: [ 2225.683161] br-lan: port 6(wlan1) entered blocking state
Sat Mar  5 01:06:15 2022 kern.info kernel: [ 2225.688499] br-lan: port 6(wlan1) entered forwarding state
Sat Mar  5 01:06:15 2022 daemon.notice netifd: Network device 'wlan1' link is up
Sat Mar  5 01:06:16 2022 daemon.notice hostapd: wlan1: interface state DFS->ENABLED
Sat Mar  5 01:06:16 2022 daemon.notice hostapd: wlan1: AP-ENABLED
```

It is strange, as the configuration works in the Netgear R7800.

This is the /etc/config/wireless that duplicates the issue:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/18000000.wmac'
        option channel '1'
        option band '2g'
        option htmode 'HT40'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '2G'
        option encryption 'sae-mixed'
        option key 'PasswordHere'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option country 'US'
        option cell_density '0'
        option htmode 'HE160'
        option channel '36'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid '5G'
        option encryption 'sae-mixed'
        option key 'PasswordHere'
```

So far the easy workaround is to avoid CH36 to CH48 when using the 160MHz channel width until the problem was fixed. However, since the default channel is 36 it is quite common to report that 160MHz is not working when actually is a bug in new snapshots. I got the router yesterday, so I don't know when the problem arose.

okibcn, 2022-03-05:
It seems that there is already a fix in process in the mt76 driver repo: https://github.com/openwrt/mt76/commit/64c74dc93f68566cd2c199d2951482ee55ca8b9a

I look forward to seeing this new version applied soon to the master OpenWrt

neheb, 2022-03-09:
Have you tested master mt76?