openwrt/openwrt #9403: Intel i225 (Stepping B3, SLNMH) is not working

jellybean13, 2022-03-06:
The device can not be loaded with errors:

[image]

From [this issue in Gentoo](https://forums.gentoo.org/viewtopic-p-8685676.html?sid=610a30b6da8c9b88fcef8e6aa4501455), this error is from Linux kernel modules. Also, this issue was fixed in Linux 5.15.x, which means either upgrading kernel sources or backport fixes from newer kernel sources can handle this issue.

Borromini, 2022-03-06:
Did you try backporting the 5.15 driver to see if it works?

jellybean13, 2022-03-06:
I attempted to backport kernel module, but I failed with compilation error.

Related commit:
https://github.com/torvalds/linux/commit/fc9df2a0b520d7d439ecf464794d53e91be74b93

jellybean13, 2022-03-09:
After applying patches from Linux Ubuntu 20.04.x HWE kernel (linux 5.11.22), it works.

[image]

Using patches:
[Kernel patches for igc: backport igc drivers from Linux Ubuntu 20.04.x HWE kernel (linux 5.11.22) to OpenWrt 21.02 SNAPSHOT (currently Linux 5.4.182).](https://github.com/jellybean13/Actions-OpenWrt/blob/x86_64/overlay/target/linux/x86/patches-5.4/300-backport-igc-driver.patch)
[Makefile patches: Fix missing dependencies.](https://github.com/jellybean13/Actions-OpenWrt/blob/x86_64/patches/002-kmod-igc-fix-missing-dependencies.patch)

ahayes, 2022-09-24:
@jellybean13 Have you tested that adapter as a DHCP client yet, by chance? I am having issues on an i226-V with OpenMPTCProuter and someone else just jumped in with the same problem on an i225-V b3. Have a peek at https://github.com/Ysurac/openmptcprouter/issues/2548 if interested.

khdegraaf, 2023-10-20:
See https://github.com/Ysurac/openmptcprouter/issues/3005 for what I believe is the cause of DHCP instability here.  I found a fix/work-around for anyone interested in this.