openwrt/openwrt #9406: AVM FRITZ!Box 7330 SL: After installation not reachable at 192.168.1.1 (from 21.02.0 on)

lupo-on-github, 2022-03-06:
Installing current stable (21.02.2) on an AVM FRITZ!Box 7330 SL, following the guide at [https://openwrt.org/toh/avm/fritz.box.wlan.7330](https://openwrt.org/toh/avm/fritz.box.wlan.7330), proceeds as to be expected, including the concluding reboot.

But after that, the device cannot be reached at 192.168.1.1, neither on LAN1 nor on LAN2, neither via browser, ssh nor by ping.

I've tested older versions; up to 19.07.9 worked, 21.02.0 fails.

Can I provide more information? 
I do not have the equipment to access the serial port of the device.

lupo-on-github, 2022-03-07:
Moved [here](https://forum.openwrt.org/t/avm-fritz-box-7330-sl-after-installation-not-reachable-at-192-168-1-1-from-21-02-0-on/122073).

blackmonkDMO, 2022-03-15:
Hi!

Same problem here. I managed to connect an USB/Serial TTL converter to the device; this is the boot log:

```
[    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 Wed Feb 16 20:292
[    0.000000] SoC: AR9 rev 1.2
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001954c (MIPS 34Kc)
[    0.000000] MIPS: machine is AVM FRITZ!Box 7320
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyLTQ0,115200
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00018110
[    0.000000] Readback ErrCtl register=00018110
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56764K/65536K available (5686K kernel code, 215K rwdata, 768K rodata, 1184K init, 203K bss, 8772K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from 0x80688a28 with crng_init=0
[    0.000000] CPU Clock: 393MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9721189849 ns
[    0.000013] sched_clock: 32 bits at 196MHz, resolution 5ns, wraps every 10922685437ns
[    0.007989] Calibrating delay loop... 260.86 BogoMIPS (lpj=521728)
[    0.046067] pid_max: default: 32768 minimum: 301
[    0.051039] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.058252] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.074639] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.084305] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.091278] pinctrl core: initialized pinctrl subsystem
[    0.100824] NET: Registered protocol family 16
[    0.112572] pinctrl-xway 1e100b10.pinmux: Init done
[    0.131408] dma-xway 1e104100.dma: Init done - hw rev: 6, ports: 5, channels: 20
[    0.142198] PCI host bridge /fpi@10000000/pci@e105400 ranges:
[    0.147895]  MEM 0x0000000018000000..0x0000000019ffffff
[    0.153132]   IO 0x000000001ae00000..0x000000001affffff
[    0.197131] usbcore: registered new interface driver usbfs
[    0.202658] usbcore: registered new interface driver hub
[    0.208033] usbcore: registered new device driver usb
[    0.213505] PCI host bridge to bus 0000:00
[    0.217524] pci_bus 0000:00: root bus resource [mem 0x18000000-0x19ffffff]
[    0.224429] pci_bus 0000:00: root bus resource [io  0x1ae00000-0x1affffff]
[    0.231372] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.238225] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.247701] pci 0000:00:0e.0: [168c:ff1d] type 00 class 0x020000
[    0.253651] pci 0000:00:0e.0: reg 0x10: [mem 0x00000000-0x0000ffff]
[    0.262183] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.268741] pci 0000:00:0e.0: BAR 0: assigned [mem 0x18000000-0x1800ffff]
[    0.285273] clocksource: Switched to clocksource MIPS
[    0.291743] thermal_sys: Registered thermal governor 'step_wise'
[    0.292272] NET: Registered protocol family 2
[    0.302840] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.311256] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.319584] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.327259] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.334364] TCP: Hash tables configured (established 1024 bind 1024)
[    0.341045] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.347534] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.354868] NET: Registered protocol family 1
[    0.359233] PCI: CLS 0 bytes, default 32
[... 439 lines cut ...]
```
Even in this setup, the traffic seems to flow only in one direction (outbound of the eth0 interface). I can confirm that, because I can see the ARP requests on my directly connected machine (Fritz!Box/eth0 <--> troubleshooting_machine):

```
14:56:07.011332 24:65:11:a3:77:ca > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.1.2 tell 192.168.1.1, length 46
14:56:07.011411 dc:0e:a1:24:5c:89 > 24:65:11:a3:77:ca, ethertype ARP (0x0806), length 42: Ethernet (len 6), IPv4 (len 4), Reply 192.168.1.2 is-at dc:0e:a1:24:5c:89, length 28
```

The ARP reply never arrives to the Fritzbox 7330 SL Device. The counters for the RX@eth0 remains always at 0:

```
    eth0 Link encap:Ethernet HWaddr 24:65:11:A3:77:CA
    inet addr:192.168.1.1 Bcast:192.168.1.255 Mask:255.255.255.0
[... cut ...]
    TX packets:198 errors:0 dropped:0 overruns:0 carrier:0
    collisions:0 txqueuelen:1000
    RX bytes:0 (0.0 B) TX bytes:12876 (12.5 KiB)
```

I know this is a pretty old device, but I try to use every bit of technology to its fullest life cycle; and this piece of technolgy still serves me (and i think others) very well; for now I will stick with OpenWRT 19.07.9 on this particular one.

If I can assist any eventual developer with my limited technical capabilities in fixing this problem, I will do it. Thank you!

blackmonkDMO, 2022-05-02:
Update, 2 May 2022, I tried the following versions:

OpenWRT 19.07.10
* LAN1 - works at 1Gbps, only if the physical interface its atached during power-on/boot; then you can disconnect and reconnect the cable, it will continue to work. But, if you connect the cable to an 100mbps device, it will not work. So 1gbps only, if the interface is connected to a 1gbps line at boot-up.
* LAN1 - works at 100Mbps, only if the physical interface its atached during power-on/boot; then you can disconnect and reconnect the cable, it will continue to work. But, if you connect the cable to an 1Gbps device, it will not work. So 100Mbps only, if the interface is connected to a 100Mbps line at boot-up.
* Not tested LAN1 at 10mbps.
* LAN2 - no link, no matter what.

OpenWRT 21.02.3 and OpenWRT 22.03.0-rc1
* LAN1 - linkup but no traffic passing; no matter speed (100mbps or 1gbps) or if the cable is connected at boot-up time or not.
* LAN2 - no link, no matter what.

I suspect this problem has something to do with the fact that this device has 2 ethernet phy chips: AR8035 (1gbps) and AR8030 (100 mbps); they supposedly use the same mdio interface (?).

I did the following experiment:
* I gitcloned the OpenWRT repository, then checkout the 21.02.3 branch;
* I manually edited ar9_avm_fritz7320.dts file by copying the &gsw section AND the state_default: pinmux section from the ar9_avm_fritz7312.dts (which uses the 8030 ethernet phy).
* I build the image for the 7320 device and flashed it to the 7330SL.
* That didn't solved the problem (obviously) but the i noticed something strange: both ports, LAN1 and LAN2 are negotiating link-up with my 100mbps unmanaged switch.
* Furthermore I kind of getting full-duplex conectivityy with both ports conected to this unmanaged switch: LAN1@7330SL is doing just RX and LAN2@7330SL is doing just TX. This really worked and i was shocked. I was able to ping and acces the device (SSH) through this weird setup.

I have no idea how to further investigate this, because i know next to nothing about linux device-tree and buses, mdios, etc. I don't even know if this is fixable with modifications in the dts file for the device only, or deeper inspection in the drivers are necessary.

Hope this helps someone.

abajk, 2022-05-03:
The Lantiq ARX100 needs its own driver for the built-in switch. There is probably no reason to put effort into fixing bugs in the current driver, since the patches available in openwrt will not be accepted upsptream in this form anyway.