openwrt/openwrt #9407: recursive dependencies caused by @IPV6 sub-dependency and PROVIDES

champtar, 2022-03-07:
We have multiple packages showing `recursive dependency detected!`
`ip6tables` is provided by both `ip6tables-legacy` and `ip6tables-nft`, both have `DEPENDS:=@IPV6`

Adding `@IPV6` to `DEPENDS` fixes the recursive dependency error, should every packages depending on `ip6tables` add `@IPV6` or can this be fixed another way ?

Here as an example  `shorewall6-lite` (from `tmp/.config-package.in`)
```
...
        config PACKAGE_shorewall6-lite
                tristate "shorewall6-lite.......................................... Shorewall6 Lite"
                default y if DEFAULT_shorewall6-lite
                default m if ALL
                depends on !(PACKAGE_ip6tables-nft<PACKAGE_shorewall6-lite) || IPV6
                select PACKAGE_libpthread if USE_GLIBC
                select PACKAGE_librt if USE_GLIBC
                select PACKAGE_ip-tiny if PACKAGE_ip-full<PACKAGE_shorewall6-lite
                select PACKAGE_libc
                select PACKAGE_ip6tables-legacy if PACKAGE_ip6tables-nft<PACKAGE_shorewall6-lite
                select PACKAGE_shorewall-core
...
```
`depends on` is wrong, it should just be `depends on IPV6`

justinkb, 2022-03-08:
olsrd, luci-app-olsr, luci-app-olsr-services, luci-app-olsr-viz are also among the affected

champtar, 2022-03-13:
@jow- @nbd168 @NeoRaider I see you 3 fixed some PROVIDES issues in the past, if you have any opinions before we just add `@IPV6` or `+IPV6:ip6tables` to all packages that create a circular dependency

champtar, 2022-03-14:
https://github.com/openwrt/packages/pull/18057 / https://github.com/openwrt/routing/pull/819

champtar, 2022-03-14:
Closing for now as all problematic packages are now fixed

closed 2022-03-14 by champtar

justinkb, 2022-03-21:
```
tmp/.config-package.in:101813:error: recursive dependency detected!
tmp/.config-package.in:101813:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft
tmp/.config-package.in:102620:  symbol PACKAGE_iptables-nft is selected by PACKAGE_opennds
tmp/.config-package.in:100843:  symbol PACKAGE_opennds depends on PACKAGE_iptables-legacy
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:101813:error: recursive dependency detected!
tmp/.config-package.in:101813:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft
tmp/.config-package.in:102620:  symbol PACKAGE_iptables-nft is selected by PACKAGE_opennds
tmp/.config-package.in:100843:  symbol PACKAGE_opennds depends on PACKAGE_iptables-legacy
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

back after your recent commits?

justinkb, 2022-03-23:
should I open a new issue for this or do we re-open this one for this very similar issue?

bluewavenet, 2022-03-25:
@justinkb 
opennds v9.7.0-2 specifically requires iptables-nft, so has:
`  DEPENDS:=+iptables-nft .....`
To enforce running with iptables-nft, it also has:
`  CONFLICTS:=....... iptables-legacy`

There don't seem to be any problems with the build-bots and it builds fine with SDK.

Is it related to the problem with the snapshot version of iptables.ipk? It installs iptables-legacy instead of defaulting to iptables-nft, which would be the logical default on snapshot and later where iptables-legacy is deprecated.

bluewavenet, 2022-03-25:
@justinkb 
It is this line that is causing the problem for you:
`tmp/.config-package.in:101813:  symbol PACKAGE_iptables-legacy is selected by PACKAGE_iptables-nft`

justinkb, 2022-03-25:
It gives an error at the start of make, but I don't use the package, so the
build works fine.

Do the build bot logs not show this error?

bluewavenet, 2022-03-25:
@justinkb 

> Do the build bot logs not show this error?

No

justinkb, 2022-03-25:
Do the build bots do clean git checkouts from scratch every time, and
subsequent feed updates and installs? If so, that's weird. I can reliably
reproduce this here during my builds targeting E8450

bluewavenet, 2022-03-25:
@justinkb 

> Do the build bots do clean git checkouts from scratch every time, and
> subsequent feed updates and installs?

I don't know.
I use the SDK to build the package and do not have any problems.

Is your .config-package.in getting updated?
The relevant part in mine is:
```
menu "Captive Portals"
	config PACKAGE_opennds
		tristate "opennds............................... Open public network gateway daemon"
		default y if DEFAULT_opennds
		default m if ALL
		select PACKAGE_libc
		select PACKAGE_libmicrohttpd-no-ssl
		depends on IPTABLES_NFTABLES
		select PACKAGE_kmod-ipt-nat
		select PACKAGE_libpthread if USE_GLIBC
		select PACKAGE_kmod-ipt-conntrack
		select PACKAGE_iptables-nft
		select PACKAGE_librt if USE_GLIBC
		help
		 openNDS is a Captive Portal solution that offers an instant way to provide restricted access to the Internet.
		 	With little or no configuration, a dynamically generated and adaptive splash page sequence is automatically served.
		 	Internet access is granted by either a click to continue button, or after credential verification.
		 	The package incorporates the FAS API allowing many flexible customisation options.
		 	The creation of sophisticated third party authentication applications is fully supported.
		 	Internet hosted https portals can be utilised to inspire maximum user confidence.
		 https://github.com/opennds/opennds
		 Rob White <rob@blue-wave.net>

endmenu
```

justinkb, 2022-03-25:
```
	config PACKAGE_opennds
		tristate "opennds............................... Open public network gateway daemon"
		default y if DEFAULT_opennds
		default m if ALL
		select PACKAGE_iptables-nft
		select PACKAGE_libc
		depends on IPTABLES_NFTABLES
		select PACKAGE_libpthread if USE_GLIBC
		select PACKAGE_kmod-ipt-conntrack
		select PACKAGE_libmicrohttpd-no-ssl
		select PACKAGE_librt if USE_GLIBC
		select PACKAGE_kmod-ipt-nat
		depends on m || (PACKAGE_nodogsplash != y)
		depends on m || (PACKAGE_iptables-legacy != y)
		help
		 openNDS is a Captive Portal solution that offers an instant way to provide restricted access to the Internet.
		   With little or no configuration, a dynamically generated and adaptive splash page sequence is automatically served.
		   Both client driven Captive Portal Detection (CPD) and gateway driven Captive Portal Identification (CPI - RFC 8910 and RFC 8908) are supported.
		   Internet access is granted by either a click to continue button, or after credential verification as a result of filling in a login form.
		   The package incorporates the FAS API allowing many flexible customisation options.
		   The creation of sophisticated third party authentication applications is fully supported.
		   Internet hosted https portals can be implemented with no security errors, to inspire maximum user confidence.
		   This version requires iptables-nft.
		 https://github.com/opennds/opennds
		 Rob White <rob@blue-wave.net>
```
here

bluewavenet, 2022-03-25:
@justinkb 

> I'm building from openwrt master. what about you?

I'm using the latest SDK for master.

> can you delete the generated tmp/.config-package.in and have it be regenerated?

Yes, no problems.

justinkb, 2022-03-25:
I have no idea as to the differences between building using the SDK vs just
building the whole firmware image from the source code base. The second use
case is where I'm seeing the error, which is definitely something that
needs fixing. You should consult with Etienne on the proper fix, I have no
real knowledge how the build system works in terms of dependency resolution

bluewavenet, 2022-03-25:
@justinkb @champtar 

> I have no real knowledge how the build system works in terms of dependency resolution

Neither do I.

**In summary**, I very much suspect it is something to do with the fact that, **on current master**:
*  Running `opkg install iptables` results in **iptables(legacy**) being installed,

where logically

 * It should be **iptables(nftables)** as the legacy version is deprecated.

Somehow the build-bots and SDK build correctly and the opennds package produced installs and runs correctly.