openwrt/openwrt #9410: [mktplinkfw2] *** error: images are too big

rubydesign, 2022-03-07:
Hi,
this is just about the message, it would be nice if it contained an indication of how much too big.

I am on image-builder 21.02.2 and trying to build a notoriously difficult 4/32 device (wr841*). So it is really about every kb

So ideally it would say image is too big by 15k

(as it stops the build i can't check/know after the error. And so have to make many many runs to creep up on it from below)
* build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/tmp/openwrt-21.02.2-ramips-mt76x8-tplink_tl-wr841n-v14-squashfs-sysupgrade.bin

Borromini, 2022-03-07:
Does it not have a --verbose switch or similar. The buildroot will only show excess size with added verbosity, off the top of my head.

rubydesign, 2022-03-07:
I am using imagebuilder with the make command. 
`make help` has no mention to verbosity flags and make does not take --verbose

closed 2022-03-08 by ynezz