openwrt/openwrt #9411: [ipv6] Outbound connection uses LAN's interface's ip not wan's

sanosis, 2022-03-07:
When using curl (curl --interface wan_interface https://api6.ipify.org)
I get lan's interface ip (or set when i'm updating dynamic dns)
This changes if another ipv6 WAN interface is present.

sanosis, 2022-03-09:
I've tested this with a setup from ISP (in my case an IP was assigned o the router + range via PD, but I could also change it), but the same situation happens with he.net tunnel - standard ip (::2) or standard + assigned from the prefix… in both cases the situation is +/- the same.
Adding a default route with src address fixes this - automatically created routing table has only ranges.

tiagogaspar8, 2022-03-18:
Just to confirm, you have an actual IPv6 global address on the WAN?
Might be of the hours but can you show me the routes you add and and the ones present?

sanosis, 2022-04-28:
I *had* a dual stack connection (ipv6 PD + ipv4 behind isp's nat), I've reverted to ipv4 only after struggling with problems originating from ipv6+ipv4. No ipv6 connection to the world on the router is a common theme in my experiments with ipv6.
Lately I had similar problems with ipv6 on pppoe - ipv6 public prefix was given to clients on the network, but router was limited to only link local, after forcing ipv6 assignment it got ipv6 on pppoe (different prefix address prefix::1) but no curl/http connection - it started working after adding a default ipv6 route.
In standard setup I get:
- wan: ipv6 only link local, no default ipv6 route on the router
- lan: ipv6 public address, on a simple setup (1 wan, 1 lan) this is used for outgoing connections from the router (doesn't work well with wan/lan firewall rules), on more complicated setups (2-3 wans, 15 lans) I usually get no connection, but sometimes lan address was used (as in name=lan interface)

tiagogaspar8, 2022-06-16:
> but router was limited to only link-local

On WAN or LAN interface?

> this is used for outgoing connections from the router

In that example, you mention that the WAN has no IPv6 address, which makes this expected behaviour.
What Could be an issue is if you have a global in WAN and global in LAN and outgoing connections were going through LAN instead of WAN. You mention this is the case with the he.net tunnel, that could be lots of things, do you have the interface marked as WAN? is it unspecified? can you see the routing table to see what routes are preferred?

sanosis, 2023-07-06:
It's been some time,but I'm back to experimenting with ISPs with ipv6 (usually with ds-lite).

The standard OpenWRT's behavior is not to set GUA on wan (like in upstream connection/wan port on router) and use LAN (as in interface name) as the main connection address. If for some reason LAN has no GUA the system will use GUA from other interface (wan's GUA as the last choice). ds-lite will set up the ipv4 tunnel with LAN's GUA, but if that does not exists ds-lite's ipv4 will not work. This setup works for basic home setup, but once you start creating more complex setups it starts braking down:
1. no lan's gua - no ds-lite (automatic at least)
2. firewall rules - dynamic dns results in setting lan's gua as aaaa - so opening access to the router needs to include modifying the lan firewall zone