openwrt/openwrt #9412: realtek/rtl838x: ethtool --show-eee lan1 causes gs108t-v3 to drop offline

stintel, 2022-03-07:
```
root@sw2:~# ethtool --show-eee lan1
EEE Settings for lan1:
        EEE status: enabled - inactive
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  100baseT/Full 
                                    1000baseT/Full 
        Link partner advertised EEE link modes:  Not reported
root@sw2:~# ethtool --show-eee lan1client_loop: send disconnect: Broken pipe
```

Don't have serial console so I don't know what exactly is happening.

Borromini, 2022-03-07:
I distinctly remember fiddling with EEE on my ZyXELs made them sh*t the bed too. Will check later.

Hurricos, 2022-03-08:
I cannot reproduce this on a ZyXEL GS1900-24HP v1 built atop [this branch](https://github.com/Hurricos/openwrt/tree/add_gs1900-24hp-v1) -- maybe has to do with bmork's [dsa patch](https://github.com/Hurricos/openwrt/commit/9e325f79497db65f2f8edf055fa81490e05435f7)?

```console
root@gs1900-24hp:~# ethtool --show-eee lan13
EEE Settings for lan13:
	EEE status: disabled
	Tx LPI: disabled
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  Not reported
	Link partner advertised EEE link modes:  Not reported
root@gs1900-24hp:~# ethtool --show-eee lan11
EEE Settings for lan11:
	EEE status: disabled
	Tx LPI: disabled
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  Not reported
	Link partner advertised EEE link modes:  Not reported
root@gs1900-24hp:~# ethtool --show-eee lan1
EEE Settings for lan1:
	EEE status: disabled
	Tx LPI: disabled
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  Not reported
	Link partner advertised EEE link modes:  Not reported
root@gs1900-24hp:~#
```

... etc. I tried every port eventually.

I didn't try any links that were both link-up AND directly connected to the RTL8382M (the only port linked-up was lan2, which is connected through one of the RTL8321 switches connected via QSGMII).

Hurricos, 2022-04-02:
I `--set-eee eee on` before running `set-eee`, but cannot reproduce this.

Hurricos, 2022-04-03:
OK, one more update:

Toggling the EEE state of a port using `set-eee` does bring the link down for a bit:

```txt
root@OpenWrt:/# ethtool --set-eee lan4 eee off
[20401.426341] Disabled EEE for port 3
[20401.430269] In rtl8218b_set_eee, port 3, enabled 0
[20401.518841] rtl8218b_set_eee done
[20401.533701] RTL8380 Link change: status: 1, ports 8
[20401.557428] rtl83xx-switch switch@1b000000 lan4: Link is Down
[20401.571546] switch: port 21(lan4) entered disabled state
[20401.581873] rtl83xx-switch switch@1b000000 lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[20401.593420] switch: port 21(lan4) entered blocking state
[20401.599388] switch: port 21(lan4) entered forwarding state
root@OpenWrt:/# [20402.197433] rtl83xx-switch switch@1b000000 lan4: Link is Down
[20402.204126] switch: port 21(lan4) entered disabled state
ip r[20404.991966] RTL8380 Link change: status: 1, ports 8
[20405.318794] rtl83xx-switch switch@1b000000 lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[20405.328528] switch: port 21(lan4) entered blocking state
[20405.334639] switch: port 21(lan4) entered forwarding state
```

bmork, 2022-06-04:
> if anyone has a similar device with access to serial console, please be so kind to try and reproduce the issue.

I do, and see the same.  Doing show-eee on lan1 breaks networking on all active ports on the Netgear GS108Tv3.  show-eee works fine on other ports.  It also works on lan1 on the ZyXEL GS1900-10HP, which should be similar hardware except for the PD circuitry.

From GS108Tv3. console with lan1 and lan2 connected:
```
root@OpenWrt:/# ethtool --show-eee lan1
EEE Settings for lan1:
        EEE status: enabled - active
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full 
                                   1000baseT/Full 
        Advertised EEE link modes:  100baseT/Full 
                                    1000baseT/Full 
        Link partner advertised EEE link modes:  100baseT/Full 
                                                 1000baseT/Full 
root@OpenWrt:/# [3699674.824702] rtl83xx-switch switch@1b000000 lan1: Link is Down
[3699674.831440] switch: port 1(lan1) entered disabled state
[3699675.067970] rtl83xx-switch switch@1b000000 lan2: Link is Down
[3699675.074832] switch: port 2(lan2) entered disabled state
```

MaZderMind, 2024-10-13:
That does seem to be fixed in current master (probably as a side effect of the 6.6 rebase). This issue should be closed.

stintel, 2024-10-13:
Cannot reproduce on GS108Tv3 running master 170ecbecfc73a9c8f6be9cb9ee67d6e5dbe3f402.

closed 2024-10-13 by stintel