openwrt/openwrt #9414: Linksys EA9500 issues

Tentpig, 2022-03-07:
I have attempted to flash 21.02.1 and 21.02.2 to a Linksys EA9500. While both flashed successfully, I am encountering errors which I can only classify as bugs in both releases.

First, in 21.02.2, after flashing, I am unable to connect via ethernet to 192.168.1.1, nor does it appear the Internet port attempts to get a DHCP address. When I watch the serial console during boot, I see the following message:

```
[    4.148865] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    4.162578] init: - preinit -
Failed to connect to the switch. Use the "list" command to see which switches are available.
Failed to connect to the switch. Use the "list" command to see which switches are available.
Failed to connect to the switch. Use the "list" command to see which switches are available.
Failed to connect to the switch. Use the "list" command to see which switches are available.
[    5.341711] bgmac_bcma bcma0:3 eth0: Link is Up - 1Gbps/Full - flow control off
[    5.349060] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

I downgraded to 21.02.1 and was able to get past this problem. However, 21.02.1 appears to have issues with configuring the wireless radios. It appears I can use any channel 112 or below on radios 0 and 2, but anything above 112 will yield the following error on the console:

```
[  171.434382] ieee80211 phy2: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55414, -52
[  173.372503] device wlan2 left promiscuous mode
[  173.377053] br-lan: port 2(wlan2) entered disabled state
[  175.314988] br-lan: port 2(wlan2) entered blocking state
[  175.320322] br-lan: port 2(wlan2) entered disabled state
[  175.325915] device wlan2 entered promiscuous mode
[  175.332565] ieee80211 phy2: brcmf_run_escan: error (-52)
[  175.337897] ieee80211 phy2: brcmf_cfg80211_scan: scan error (-52)

```

Once the scan error appears, the only way to clear it is a reboot. Even changing the channel to something 'valid' will still generate the error.

I set my country code to US and am using 40Mhz widths.

Thanks, Vinny

ynezz, 2022-03-08:
@rmilecki do you've any clue?

tryingbe, 2022-05-24:
I came across the same issue on the 21.02.x, so I tried 22.03.x

openwrt-22.03.0-rc1-bcm53xx-generic-linksys_ea9500-squashfs.trx works. 

openwrt-22.03.0-rc2-r19374-34b6abf5a8-bcm53xx-generic-linksys_ea9500-squashfs.trx doesn't work.

neheb, 2026-02-28:
still an issue?