openwrt/openwrt #9420: MT7622: Fix FDB learning bugs when VLAN filtering is enabled causes performance loss

dietcoke73, 2022-03-07:
Fix FDB learning bugs when VLAN filtering is enabled was merged in ee6ba216d8ba1b02154c287e64d709a8bc7b0054 but causes a performance drop on LAN ports.  Multiple users discussing this issue in Forum topic https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/1785 

TL;DR summary - when mulitple devices are connected to the switch, the performance of the switch drops to the lowest speed of the connected devices.  So for example a 1000mbps capable device receives data only at the 10mbps data rate.

Reverting this commit ee6ba216d8ba1b02154c287e64d709a8bc7b0054 tested by multiple users corrects the problem as the cost of reintroducing the original issue.

Vladdrako, 2022-03-09:
Related patches are:
```
765-v5.15-net-dsa-mt7530-disable-learning-on-standalone-ports.patch
766-v5.15-net-dsa-mt7530-enable-assisted-learning-on-CPU-port.patch
```
With the rest no problems. I manually edited the 767 patch to apply. Also applied [this](https://github.com/LGA1150/openwrt/commit/80a9611cea1a4310b46b7ee777a98a99802514fd) patch.

justinkb, 2022-03-09:
@dengqf6 seems to be the author of the upstream kernel patch

@stintel 

I don't understand where the logic in the backport for these two diff chunks comes from:

```
--- a/drivers/net/dsa/mt7530.c
+++ b/drivers/net/dsa/mt7530.c
@@ -1163,6 +1163,8 @@ mt7530_port_bridge_join(struct dsa_switc
                           PCR_MATRIX_MASK, PCR_MATRIX(port_bitmap));
        priv->ports[port].pm |= PCR_MATRIX(port_bitmap);
 
+       mt7530_clear(priv, MT7530_PSC_P(port), SA_DIS);
+
        mutex_unlock(&priv->reg_mutex);
 
        return 0;
@@ -1260,6 +1262,8 @@ mt7530_port_bridge_leave(struct dsa_swit
                           PCR_MATRIX(BIT(MT7530_CPU_PORT)));
        priv->ports[port].pm = PCR_MATRIX(BIT(MT7530_CPU_PORT));
 
+       mt7530_set(priv, MT7530_PSC_P(port), SA_DIS);
+
        mutex_unlock(&priv->reg_mutex);
 }
```

I don't see this behavior reflected anywhere in the original patch. I think the first chunk is the culprit (and the second one is useless if you nuke the first chunk from the patch). What am I missing?

Reading the linked discourse thread, I see quarky and LGA1150 have also identified this part of the patch is the problem. LGA1150 objected to quarky's suggestion (identical to mine) to delete those two chunks entirely with the following

> However enabling address learning back by default is not an option, as the [documentation](https://docs.kernel.org/networking/dsa/dsa.html) says: The switch driver is responsible for initial setup of the standalone ports with address learning disabled

But isn't that exactly what the rest of the patch does (the two other chunks).

Vladdrako, 2022-03-09:
I tried quarky's patch:
<pre>
--- a/drivers/net/dsa/mt7530.c
+++ b/drivers/net/dsa/mt7530.c
@@ -1193,7 +1193,7 @@
 			   PCR_MATRIX_MASK, PCR_MATRIX(port_bitmap));
 	priv->ports[port].pm |= PCR_MATRIX(port_bitmap);
 
-	mt7530_clear(priv, MT7530_PSC_P(port), SA_DIS);
+	//mt7530_clear(priv, MT7530_PSC_P(port), SA_DIS);
 
 	/* Set to fallback mode for independent VLAN learning */
 	mt7530_rmw(priv, MT7530_PCR_P(port), PCR_PORT_VLAN_MASK,
@@ -1304,7 +1304,7 @@
 			   PCR_MATRIX(BIT(dsa_upstream_port(ds, port))));
 	priv->ports[port].pm = PCR_MATRIX(BIT(dsa_upstream_port(ds, port)));
 
-	mt7530_set(priv, MT7530_PSC_P(port), SA_DIS);
+	//mt7530_set(priv, MT7530_PSC_P(port), SA_DIS);
 
 	/* When a port is removed from the bridge, the port would be set up
 	 * back to the default as is at initial boot which is a VLAN-unaware
@@ -2733,7 +2733,7 @@
 			   PCR_MATRIX_CLR);
 
 		/* Disable learning by default on all ports */
-		mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
+		//mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
 
 		if (dsa_is_cpu_port(ds, i)) {
 			ret = mt753x_cpu_port_enable(ds, i);
@@ -2906,7 +2906,7 @@
 			   PCR_MATRIX_CLR);
 
 		/* Disable learning by default on all ports */
-		mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
+		//mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
 
 		mt7530_set(priv, MT7531_DBG_CNT(i), MT7531_DIS_CLR);
 
</pre>
and LGA1150 variant:
<pre>
```
--- a/drivers/net/dsa/mt7530.c
+++ b/drivers/net/dsa/mt7530.c
@@ -1193,7 +1193,7 @@ mt7530_port_bridge_join(struct dsa_switch *ds, int port,
 			   PCR_MATRIX_MASK, PCR_MATRIX(port_bitmap));
 	priv->ports[port].pm |= PCR_MATRIX(port_bitmap);

-	mt7530_clear(priv, MT7530_PSC_P(port), SA_DIS);
+	mt7530_rmw(priv, MT7530_PSC_P(port), SA_CNT_EN, MAC_SA_LRN);

 	/* Set to fallback mode for independent VLAN learning */
 	mt7530_rmw(priv, MT7530_PCR_P(port), PCR_PORT_VLAN_MASK,
@@ -1300,7 +1300,7 @@ mt7530_port_bridge_leave(struct dsa_switch *ds, int port,
 			   PCR_MATRIX(BIT(MT7530_CPU_PORT)));
 	priv->ports[port].pm = PCR_MATRIX(BIT(MT7530_CPU_PORT));

-	mt7530_set(priv, MT7530_PSC_P(port), SA_DIS);
+	mt7530_rmw(priv, MT7530_PSC_P(port), MAC_SA_LRN, SA_CNT_EN);

 	/* When a port is removed from the bridge, the port would be set up
 	 * back to the default as is at initial boot which is a VLAN-unaware
@@ -2714,7 +2714,7 @@ mt7530_setup(struct dsa_switch *ds)
 			   PCR_MATRIX_CLR);

 		/* Disable learning by default on all ports */
-		mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
+		mt7530_rmw(priv, MT7530_PSC_P(i), MAC_SA_LRN, SA_CNT_EN);

 		if (dsa_is_cpu_port(ds, i)) {
 			ret = mt753x_cpu_port_enable(ds, i);
@@ -2887,7 +2887,7 @@ mt7531_setup(struct dsa_switch *ds)
 			   PCR_MATRIX_CLR);

 		/* Disable learning by default on all ports */
-		mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
+		mt7530_rmw(priv, MT7530_PSC_P(i), MAC_SA_LRN, SA_CNT_EN);

 		mt7530_set(priv, MT7531_DBG_CNT(i), MT7531_DIS_CLR);

--- a/drivers/net/dsa/mt7530.h
+++ b/drivers/net/dsa/mt7530.h
@@ -294,6 +294,8 @@ enum mt7530_port_mode {

 /* Register for port security control */
 #define MT7530_PSC_P(x)			(0x200c + ((x) * 0x100))
+#define  MAC_SA_LRN			GENMASK(19, 8)
+#define  SA_CNT_EN			BIT(5)
 #define  SA_DIS				BIT(4)

 /* Register for port vlan control */
</pre>
with no luck. But maybe I'm wrong somewhere. @justinkb Can you test?

justinkb, 2022-03-09:
I think the issue with quarky's patch is that he/you commented out ALL the mt7530_set(priv, MT7530_PSC_P(i), SA_DIS) instances, even those that are necessary

justinkb, 2022-03-09:
I just did a build with the first two chunks of the 765-...patch removed and the issue is gone. I'm still not sure if this is the right fix, but as I said, I can't find any corresponding logic in the original patch, so not sure why these two lines in bridge_join and bridge_leave ended up in the backport?

LGA1150, 2022-03-10:
> I can't find any corresponding logic in the original

It's in .port_bridge_flags upstream. DSA core calls it when a port joins/leaves a bridge.

Vladdrako, 2022-03-10:
> I just did a build with the first two chunks of the 765-...patch removed and the issue is gone.

Just deleting chunks didn't fix the issue for me. I need to drop 765/766 patches completely to fix it.

justinkb, 2022-03-11:
I only diagnosed the issue over WAN. If I remove the two chunks, upload speed over WAN is fine, but it's capped over LAN instead. If I keep the two chunks, on WAN the upload speed is hampered while it's fine over LAN (IIRC). weird issue.

I'm gonna take your approach and drop 76{5,6} for now and adapt 767

sppmasterspp, 2022-03-18:
Do you think that this issue can be related anyhow with the seemingly very similar effect of poor WAN performance during a LAN traffic with 100FD Mbps connected device/s.
https://github.com/openwrt/openwrt/issues/9202

quarkysg, 2022-03-18:
@justinkb When I did not set the SA_DIS bit (i.e. remaining as the default switch port behaviour) for all the switch ports, including the CPU port, the issue I encountered went away on my E8450.  It's odd that it did not resolve it for you.

I have been studying the mt7530.c codes, and also reading the DSA documentation in the Linux kernel tree.  It seems like the switch CPU port's learning should be enabled (according to documentation.)

In any case, when switch ports are added into a bridge and vlan filtering is turned on, the switch ports SA_DIS bit will be cleared (i.e. SA learning is enabled) for the bridge's switch ports.  But the patch causing the regression set the SA_DIS bit (causing SA learning to be disabled) for all ports (including the CPU ports) when the switch is initialised.  When the bridge is setup and vlan filtering enabled, the CPU port's SA_DIS bit remained set.  So end result is that all switch ports in the bridge has SA learning enabled but the CPU port's SA learning is disabled.  This somehow is causing the regression from what I can tell.  In this instance, for me at least, when I upload another firmware to the router, I'm also hit with the 10mbps link, as one of the switch port's link speed is at 10mbps.

justinkb, 2022-03-22:
just did a fresh build from master, problem is still present (no reason to assume it wasn't, but doesn't hurt to verify)

dangowrt, 2022-03-22:
https://github.com/openwrt/openwrt/issues/9514 reports the same problem on ramips MT7621 hardware, I've edited the issue title accordingly.

stintel, 2022-03-22:
I'm not familiar with this code at all, I just merged the patches because I tested them on my ramips/mt7621 hardware and noticed no issues. The only thing I can do to solve the problem for people encountering it is reverting the patch. If people agree I do that, please confirm which hashes I need to revert. If not, please don't ping me again about this problem.

dl3it, 2022-03-22:
With this commit (https://github.com/openwrt/openwrt/commit/ee6ba216d8ba1b02154c287e64d709a8bc7b0054) devices roaming on WIFI don't get a network connection, until the fdb entry in mit7530 has expired; so no DHCP answer, and so on. This behavior is pretty annoying, as you have to wait for >1min, or so, until you get network access.
Deleting 765-..., 766-..., and modifying 767-... (as mentioned earlier) fixes this problem for me. I have no idea, what side effects are caused by these modifications.
I see no speed issues.

It might be a good idea to revert this commit until all effects of these changes are understood and the behavior is considered to be intended.

Vladdrako, 2022-03-24:
@LGA1150 Didn't help, unfortunately. (I restored 765/6/7 patches and applied your one)

LGA1150, 2022-03-24:
@Vladdrako 
```diff
--- a/drivers/net/dsa/mt7530.c
+++ b/drivers/net/dsa/mt7530.c
@@ -1021,9 +1021,9 @@ mt753x_cpu_port_enable(struct dsa_switch *ds, int port)
 	mt7530_write(priv, MT7530_PCR_P(port),
 		     PCR_MATRIX(dsa_user_ports(priv->ds)));

-	/* Set to fallback mode for independent VLAN learning */
-	mt7530_rmw(priv, MT7530_PCR_P(port), PCR_PORT_VLAN_MASK,
-		   MT7530_PORT_FALLBACK_MODE);
+	/* Set default PVID to 4095 */
+	mt7530_rmw(priv, MT7530_PPBV1_P(port), G0_PORT_VID_MASK,
+		   G0_PORT_VID(4095));

 	return 0;
 }
@@ -2128,9 +2128,6 @@ mt7530_setup(struct dsa_switch *ds)
 		mt7530_rmw(priv, MT7530_PCR_P(i), PCR_MATRIX_MASK,
 			   PCR_MATRIX_CLR);

-		/* Disable learning by default on all ports */
-		mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
-
 		if (dsa_is_cpu_port(ds, i)) {
 			ret = mt753x_cpu_port_enable(ds, i);
 			if (ret)
@@ -2138,6 +2135,9 @@ mt7530_setup(struct dsa_switch *ds)
 		} else {
 			mt7530_port_disable(ds, i);

+			/* Disable learning by default on all user ports */
+			mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
+
 			/* Set default PVID to 0 on all user ports */
 			mt7530_rmw(priv, MT7530_PPBV1_P(i), G0_PORT_VID_MASK,
 				   G0_PORT_VID_DEF);
@@ -2296,9 +2296,6 @@ mt7531_setup(struct dsa_switch *ds)
 		mt7530_rmw(priv, MT7530_PCR_P(i), PCR_MATRIX_MASK,
 			   PCR_MATRIX_CLR);

-		/* Disable learning by default on all ports */
-		mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
-
 		mt7530_set(priv, MT7531_DBG_CNT(i), MT7531_DIS_CLR);

 		if (dsa_is_cpu_port(ds, i)) {
@@ -2308,6 +2305,9 @@ mt7531_setup(struct dsa_switch *ds)
 		} else {
 			mt7530_port_disable(ds, i);

+			/* Disable learning by default on all user ports */
+			mt7530_set(priv, MT7530_PSC_P(i), SA_DIS);
+
 			/* Set default PVID to 0 on all user ports */
 			mt7530_rmw(priv, MT7530_PPBV1_P(i), G0_PORT_VID_MASK,
 				   G0_PORT_VID_DEF);
```

dl3it, 2022-03-24:
with this patch, the device (Xiaomi 4A Gig) is unreachable over the network. Roaming also does not work.

Vladdrako, 2022-03-24:
@LGA1150 Nothing changes :(

atiensivu, 2022-03-25:
Would it make sense to make a special case for the CPU port or specific models known to have problems? I'm not intimately familiar with the mt* driver internals but it sounds like learning is at least partially required to not get stuck in this scenario. 

Documentation may say to disable learning but documentation doesn't necessarily need to be followed, except for trying to upstream changes.

quarkysg, 2022-03-30:
I added debug logs to the mt7530 driver, specifically to the part where the CPU will add FDB entries (i.e. mt7530_port_fdb_add/del()) into the switch ARL table.

I do not see CPU adding any FDB entries into the switch's ARL table for host connected to the switch ports, regardless of whether the switch port is bridged or not.  I only see the DSA framework adding FDB entries for WiFi clients and those from the OpenVPN tap interface that I added to the bridge, and it is only added for the CPU port.  Also I do not see any flags in the DSA framework to tell it whether a switch port's FDB should be managed by DSA instead of the switch.  I would assume that if a switch SA learning is disabled, it should be handled by the DSA framework.

So when SA learning is disabled for standalone ports, all incoming frames from the stand-alone port(s) will always go thru the switch's forwarding algorithm, since there's no ARL entry.  Maybe this is why the switch drops to the lowest link speed?

rossiniscarface, 2022-03-30:
Just on question: how is vlan filtering activated? In Luci on the Bridge VLAN filtering —> Enable VLAN filtering?
If yes, than this is strange because its not checked in my config.

spolack, 2022-04-03:
> With this commit (https://github.com/openwrt/openwrt/commit/ee6ba216d8ba1b02154c287e64d709a8bc7b0054) devices roaming on WIFI don't get a network connection, until the fdb entry in mit7530 has expired;

I can confirm the issue on Linksys E8450,  breaking Roaming for completly,

jebrossard, 2022-04-03:
I am having this issue as well.  The MT7621 ethernet port with the tagged VLANs defined becomes essentially useless from a few minutes to four days after a restart. https://forum.openwrt.org/t/mt7621-vlan-problem/124058

closed 2022-04-03 by jow-

justinkb, 2022-04-03:
Hopefully the switch to 5.15 kernel will go relatively smoothly so all these Frankenstein backport hacks are no longer even necessary

dl3it, 2022-04-03:
Thanks for reverting....

@justinkb 
I tried a build with 5.15. The device (XIAOMI 4A Gig) was unreachable over wired network... WIFI connection was possible, but of course no DHCP and no connection to the rest of the world. This might need some further refinement :-)

quarkysg, 2022-04-04:
Would anyone be interested to try out my [patch ](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/1916?u=quarky) to the mt7530 driver?  I tried to change the way VLAN is configured for the switch.

rossiniscarface, 2022-04-04:
Is this also reverted for the 22.03 branch?

Matthaiks, 2022-04-05:
Netgear R6220 & openwrt-7d9db1760307 (without "kernel: backport MT7530 VLAN fix")

WAN (PPPoE) is partially (pinging works but other stuff like web browsing does not) unreachable via Ethernet when HW NAT is enabled. No problems via Wi-Fi.

d-me3, 2022-04-11:
> Is this also reverted for the 22.03 branch?

I was not able to find this revert on the 22.03 branch (and I've also faced this issue on a Xiaomi AX6S runnning a recent build of 22.03).

I believe this commit should also be reverted on the 22.03 branch.

pesa1234, 2022-04-28:
Hi, we have very similiar issue on mvebu linksys wrt series https://forum.openwrt.org/t/wrt1900acs-only-10mbps-in-upload-dsa-issue/92609

sppmasterspp, 2022-05-23:
I own Netgear R7800 and I've reported a bug here https://github.com/openwrt/openwrt/issues/9202
R7800 poor download rate when client is connected at 100Mbps.
The exact steps to reproduce the bug are here.
[Netgear R7800 exploration (IPQ8065, QCA9984) - #3187 by sppmaster](https://forum.openwrt.org/t/netgear-r7800-exploration-ipq8065-qca9984/285/3187)

Today I've benchmarked a Belkin RT3200 with the same tests with a 100Mbps client connected to the LAN switch port of RT3200. It still runs factory firmware. I can hardly reach 200Mbps down and 20Mbps upload speeds.
I can only say that I see the same decrease in the WAN throughput. The same issue is present as with R7800. So probably the bug is not connected to the specific hardware as both are based on different SoC.
Is it clear what exactly cause this erratic behaviour.

commit 3e0daca6 (2022-04-03): Revert "kernel: backport MT7530 VLAN fix"

There are various reports on Github and in the forum that this commit
causes multiple problems.

This reverts commit ee6ba216d8ba1b02154c287e64d709a8bc7b0054.

Fixes: #9420
commit f029912c (2022-04-19): Revert "kernel: backport MT7530 VLAN fix"

There are various reports on Github and in the forum that this commit
causes multiple problems.

This reverts commit ee6ba216d8ba1b02154c287e64d709a8bc7b0054.

Fixes: #9420

(cherry picked from commit 3e0daca6447c3d5b9eb6d24ecb8e52f256f385cc)