openwrt/openwrt #9423: lantiq: arcadyan_vgv7510kw22-brn-squashfs-factory.bin and arcadyan_vgv7519-brn-squashfs-factory.bin exceed IMAGE_SIZE

jospezial, 2022-03-08:
I was just looking for the reason why
arcadyan_vgv7510kw22-brn-squashfs-factory.bin
and
arcadyan_vgv7519-brn-squashfs-factory.bin
are missing in https://downloads.openwrt.org/snapshots/targets/lantiq/xrx200/ .
They exist in https://downloads.openwrt.org/releases/21.02.2/targets/lantiq/xrx200/

IMAGE_SIZE := 7168k for Device/arcadyan_vgv7510kw22-brn is exceeded.
Despite the snapshots don't have luci, even the nor-sysupgrade image in snapshots is ~500kB bigger than in releases/21.02.2.

What makes the big difference?
Kernel 5.4-->5.10? BPF? Firewall?

@hauke @mkresin @blocktrron @abajk @kestrel1974 @xdarklight 
Any ideas how to make it smaller?

abajk, 2022-03-08:
You can try to remove the unnecessary firmware for gphy (https://github.com/openwrt/openwrt/blob/master/target/linux/lantiq/xrx200/config-5.10#L13). This should reduce the kernel size by 100kB.

jospezial, 2022-03-08:
will test the difference in my next build
```Symbol: KERNEL_KALLSYMS [=y]                                                                                                                                                                                                                                                                 │  
  │ Type  : bool                                                                                                                                                                                                                                                                                 │  
  │ Defined at config/Config-kernel.in:272                                                                                                                                                                                                                                                       │  
  │   Prompt: Compile the kernel with symbol table information                                                                                                                                                                                                                                   │  
  │   Location:                                                                                                                                                                                                                                                                                  │  
  │     -> Global build settings                                                                                                                                                                                                                                                                 │  
  │ (1)   -> Kernel build options```

hauke, 2022-03-08:
Someone could remove the gphy firmware form the kernel binary and add it to the user space. This should work with the DSA driver. Then we can pack only the needed firmware binaries.

xdarklight, 2022-03-14:
Even removing the GPHY firmware () and saving 100kB will probably result in a non-working device because there's not enough space left for the overlay.

I see two options:
- replacing brnboot with u-boot to get ~15MB
- (I have not tried this, see it as an idea) replacing Code_Image_0 and Code_Image_1 with LZMA loader (identical copies) then placing a "kernel" partition after Code_Image_0's LZMA loader and a "rootfs" partition after Code_Image_1's loader. That should leave ~7MB for the kernel and another ~7MB for "rootfs" (squashfs image) + "rootfs_data" (overlay)