openwrt/openwrt #9428: C99 float operations are not full support based on glibc openwrt

leyao-daily, 2022-03-09:
Is C99 float operations are not full support based on glibc openwrt? When using glibc built image, it report C99 float operations are not supported.

neheb, 2022-03-09:
What target?

leyao-daily, 2022-03-09:
x86_64 and openssl/evp.h can not be found.
symbol PEM_read_RSAPrivateKey not found

neheb, 2022-03-09:
So many details missing.

Sounds like you're trying to add some package.

leyao-daily, 2022-03-09:
I just build glibc and choose some coreutils-* into the openwrt image, and I import it as a docker image, then I compile some application. In its confiuration check, it report the C99 float unsupported error.
I check the symbol and find there is no libmvec under /lib but /usr/lib/gcc/.../.../..

neheb, 2022-03-09:
ping @hauke @dedeckeh I can't see libmvec anywhere.

leyao-daily, 2022-03-10:
@neheb 
I find the libmvec, it's under `/usr/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/`, and copy this file to `/lib` solve C99 not supported issue.
What confused me is that, why the library under  `/usr/lib/gcc/x86_64-openwrt-linux-gnu/7.4.0/` can not be used unless we copy it to /lib.

leyao-daily, 2022-03-10:
And it is not supported to recongnize 'SD' 'DD' 'TD'
error: unable to emulate 'SD'
error: unable to emulate 'DD'
error: unable to emulate 'TD'
SD - Single Decimal Floating; 4 byte (32-bit) decimal float-point
DD - Double Decimal Floating; 8 byte (64-bit) decimal float-point
TD - Tetra Decimal Floating; 4 byte (128-bit) decimal float-point