openwrt/openwrt #9430: icmpv6 match not working with ip6tables-nft

champtar, 2022-03-09:
I am currently testing the mwan3 for nftables.
Since it is not possible to rewrite into native nftables in the short time, I am using iptables-nft translation.
So far everything seems to work for ipv4.
But I have problems loading the icmp6 rules.
https://github.com/openwrt/packages/blob/de9fce568b4d1a2a12cd67312449b0825880c89b/net/mwan3/files/lib/mwan3/mwan3.sh#L274-L277

I always get the follwing output in the logs:
`ip6tables-nft-restore v1.8.7 (nf_tables): Couldn't load match icmp6:No such file or directory  Error occurred at line: 11 Try ip6tables-nft-restore -h or'ip6tables-nft-restore --help' for more information.`

The message is generate in iptables in the following line.
https://git.netfilter.org/iptables/tree/libxtables/xtables.c#n760

As I understand it, the file `libxt_icmp6.so` is missing on the target.
And that is also true. This is not generated during compilation.
If I look at the iptables packet then this module is in builtin.

`BUILTIN_MODULES="standard icmp tcp udp comment set SET limit mac multiport comment LOG nf_log_common nf_log_ipv4 TCPMSS REJECT time mark MARK icmp6 REJECT state CT conntrack SNAT DNAT MASQUERADE REDIRECT"`

I am now at a bit of a loss as to what I can do now

_Originally posted by @feckert in https://github.com/openwrt/packages/issues/16818#issuecomment-1062781558_

feckert, 2022-03-09:
@champtar thankst for open this here.
```
-$(eval $(if $(NF_KMOD),,$(call nf_add,IPT_IPV6,CONFIG_IP6_NF_IPTABLES, ip6t_icmp6)))
+$(eval $(if $(NF_KMOD),,$(call nf_add,IPT_IPV6_EXTRA,CONFIG_IP6_NF_IPTABLES, ip6t_icmp6)))
```
Then the file `libip6t_icmp6.so` gets build. Unfortunately, I have not yet seen that it is also copied into the package for installation. I have to manually copy the new file to the system in the `/usr/lib/iptables` folder.
After this change, the mwan3 does not complain.

champtar, 2022-03-09:
Many modules are built in depending on what you select. Can you removed this modification and try a clean build (`make clean`)
Not sure iptable kmods and iptables always get properly rebuilt when needed

feckert, 2022-03-09:
@champtar I must say I have forgotten to switch on ip6tables-extra in `make menuconfig`. After that i have the following package structure.
```
ipkg-x86_64/ip6tables-extra/
├── CONTROL
│   ├── control
│   ├── postinst
│   └── prerm
└── usr
    └── lib
        └── iptables
            └── libip6t_icmp6.so
```

But I am not sure yet if this is the solution for the problem.
I don't understand why the object `libip6t_icmp6.o` is not static linked into xtables-nft-multi.
With my change I have an `libip6t_icmp6.so`

champtar, 2022-03-13:
`ip6tables-nft -m icmp6 --help` fails, `ip6tables-legacy -m icmp6 --help` works
`libip6t_icmp6.o` is in `libiptext6.so`, problem is that we don't call `init_extensions6()`

In multiple iptables source file we have
```
#if defined(ALL_INCLUSIVE) || defined(NO_SHARED_LIBS)
        init_extensions();
        init_extensions4();
#endif
```

I think we just need to backport this patch https://git.netfilter.org/iptables/commit/?id=e727ccad036e2cdba3339536c65c7ceef43c0740

=> here the PR https://github.com/openwrt/openwrt/pull/9458

champtar, 2022-03-13:
@feckert PR now merged, please confirm it fixes everything for you when you get a chance

closed 2022-03-13 by champtar

feckert, 2022-03-14:
@champtar thanks for this fix. Works now :+1: