openwrt/openwrt #9440: Kernel panics, Router constantly goes into recovery mode

t3dium, 2022-03-10:
My router works fine until it randomly goes into recovery mode and this happens at least once every day. When it does go into recovery mode i then have to use ethernet, ssh and delete the pstore log, then reboot which temporarily fixes it.

**Version:** latest **Device:** belkin rt3200

I've tried re-flashing openwrt several times, and also updated to the latest version using the luci tool, but still have this issue. I've pasted cat /sys/fs/pstore* below:

```
<6>[    9.165285] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
<6>[    9.187663] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
<6>[    9.312883] PPP generic driver version 2.4.2
<6>[    9.318118] NET: Registered protocol family 24
<14>[    9.329880] kmodloader: done loading kernel modules from /etc/modules.d/*
<6>[   12.599785] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   12.603364] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   12.603562] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   12.603587] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<6>[   12.607618] mt7530 mdio-bus:00 lan1: configuring for phy/gmii link mode
<6>[   12.611842] 8021q: adding VLAN 0 to HW filter on device lan1
<6>[   12.616785] br-lan: port 1(lan1) entered blocking state
<6>[   12.616800] br-lan: port 1(lan1) entered disabled state
<6>[   12.617695] device lan1 entered promiscuous mode
<6>[   12.617703] device eth0 entered promiscuous mode
<6>[   12.631432] mt7530 mdio-bus:00 lan2: configuring for phy/gmii link mode
<6>[   12.631987] 8021q: adding VLAN 0 to HW filter on device lan2
<6>[   12.636769] br-lan: port 2(lan2) entered blocking state
<6>[   12.636782] br-lan: port 2(lan2) entered disabled state
<6>[   12.637882] device lan2 entered promiscuous mode
<6>[   12.643568] mt7530 mdio-bus:00 lan3: configuring for phy/gmii link mode
<6>[   12.644158] 8021q: adding VLAN 0 to HW filter on device lan3
<6>[   12.649494] mt7530 mdio-bus:00 lan3: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.652472] br-lan: port 3(lan3) entered blocking state
<6>[   12.652484] br-lan: port 3(lan3) entered disabled state
<6>[   12.654805] device lan3 entered promiscuous mode
<6>[   12.658568] br-lan: port 3(lan3) entered blocking state
<6>[   12.658583] br-lan: port 3(lan3) entered forwarding state
<6>[   12.658702] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
<6>[   12.664290] mt7530 mdio-bus:00 lan4: configuring for phy/gmii link mode
<6>[   12.664863] 8021q: adding VLAN 0 to HW filter on device lan4
<6>[   12.669830] mt7530 mdio-bus:00 lan4: Link is Up - 100Mbps/Full - flow control rx/tx
<6>[   12.672226] br-lan: port 4(lan4) entered blocking state
<6>[   12.672239] br-lan: port 4(lan4) entered disabled state
<6>[   12.674276] device lan4 entered promiscuous mode
<6>[   12.676534] br-lan: port 4(lan4) entered blocking state
<6>[   12.676547] br-lan: port 4(lan4) entered forwarding state
<6>[   12.682258] mt7530 mdio-bus:00 wan: configuring for phy/gmii link mode
<6>[   12.682801] 8021q: adding VLAN 0 to HW filter on device wan
<6>[   12.687548] mt7530 mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.689538] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
<6>[   15.043015] br-lan: port 5(wlan1) entered blocking state
<6>[   15.043029] br-lan: port 5(wlan1) entered disabled state
<6>[   15.043246] device wlan1 entered promiscuous mode
<6>[   15.043330] br-lan: port 5(wlan1) entered blocking state
<6>[   15.043338] br-lan: port 5(wlan1) entered forwarding state
<6>[   15.045045] br-lan: port 5(wlan1) entered disabled state
<6>[   15.170457] br-lan: port 6(wlan0) entered blocking state
<6>[   15.170470] br-lan: port 6(wlan0) entered disabled state
<6>[   15.170709] device wlan0 entered promiscuous mode
<6>[   15.170795] br-lan: port 6(wlan0) entered blocking state
<6>[   15.170804] br-lan: port 6(wlan0) entered forwarding state
<6>[   15.419884] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   15.419925] br-lan: port 1(lan1) entered blocking state
<6>[   15.419935] br-lan: port 1(lan1) entered forwarding state
<6>[   16.045992] br-lan: port 6(wlan0) entered disabled state
<6>[   22.615047] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[   22.615184] br-lan: port 5(wlan1) entered blocking state
<6>[   22.615194] br-lan: port 5(wlan1) entered forwarding state
<6>[   23.036756] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<6>[   23.036895] br-lan: port 6(wlan0) entered blocking state
<6>[   23.036903] br-lan: port 6(wlan0) entered forwarding state
<6>[   54.623461] br-lan: port 3(lan3) entered disabled state
[... 173 lines cut ...]
<7>[160817.599667] x19: dead0000000000f0 x18: ffffffc010cc7570
<7>[160817.605058] x17: 000000000000001e x16: 0000000b7caccada
<7>[160817.610449] x15: 000031cd89e2066e x14: ffffffc0108083e0
<7>[160817.615840] x13: 0000000000000357 x12: ffffffffffffffff
<7>[160817.621231] x11: 0000000000000040 x10: ffffff8000ce1914
<7>[160817.626622] x9 : ffffff8002a7857c x8 : 0000000000000000
<7>[160817.632013] x7 : ffffffffffffffff x6 : 0000000000000000
<7>[160817.637404] x5 : 0000000000000002 x4 : 0000000000000000
<7>[160817.642795] x3 : ffffff800034ee00 x2 : 0000000000000000
<7>[160817.648186] x1 : dead000000000100 x0 : 00000000ffffffff
<7>[160817.653576] Call trace:
<7>[160817.656103]  nf_flow_offload_ip_hook+0x778/0x1340 [nf_flow_table]
<7>[160817.662276]  nf_flow_table_offload_setup+0x468/0x660 [nf_flow_table]
<7>[160817.668709]  process_one_work+0x1d4/0x370
<7>[160817.672796]  worker_thread+0x178/0x4e0
<7>[160817.676625]  kthread+0x120/0x124
<7>[160817.679933]  ret_from_fork+0x10/0x18
<0>[160817.683589] Code: d1004013 eb0002bf 54000580 52800014 (f9401263)
<4>[160817.689759] ---[ end trace d2e0a6d64d92b633 ]---
<0>[160817.701479] Kernel panic - not syncing: Oops: Fatal exception
<2>[160817.707308] SMP: stopping secondary CPUs
<0>[160817.711313] Kernel Offset: disabled
<0>[160817.714881] CPU features: 0x0000002,04002004
<0>[160817.719226] Memory Limit: none
```

ynezz, 2022-03-10:
> nf_flow_offload_ip_hook+0x778/0x1340 [nf_flow_table]

```
(gdb) l *nf_flow_offload_ip_hook+0x778
0x30b8 is in nf_flow_offload_tuple (net/netfilter/nf_flow_table_offload.c:794).
```

```c
 792         down_read(&flowtable->flow_block_lock);
 793         list_for_each_entry(block_cb, block_cb_list, list) {
 794                 err = block_cb->cb(TC_SETUP_CLSFLOWER, &cls_flow,
 795                                    block_cb->cb_priv);
 796                 if (err < 0)
 797                         continue;
 798 
 799                 i++;
 800         }
```

justinkb, 2022-03-10:
is this a snapshot build from the official build bot or something you built yourself? isn't this that old bug that was caused by nftables hack in combination with firewall3? or is it just superficially similar?

t3dium, 2022-03-12:
I can confirm the issue was related to hardware offloading as I haven’t had a kernel panic since disabling it.

> is this a snapshot build from the official build bot or something you built yourself?

I’m using [dangowrt’s image](https://github.com/dangowrt/linksys-e8450-openwrt-installer) as recommended by the openwrt wiki. Although, someone said that the issue wasn’t related to his port in the issues sections there.

> nftables hack in combination with firewall3

This might be the case as I believe firewall 4 only just got added on his latest pre-release version, for the time being I’ll close this issue and just disable offloading. However once his latest pre-release gets officially released I may come back and test to see if that fixes the issue.

ynezz, 2022-03-12:
This looks like a real bug, it should be fixed.

Vladdrako, 2022-03-13:
> I can confirm the issue was related to hardware offloading

[Related](https://forum.openwrt.org/t/possible-kernel-5-10-regression-issue-with-mt7621-and-sw-hw-offload-enabled/115505/110) I think. This bug is really hurt. I can't upload a huge file to USB-HDD because of kernel panics. My device is mir3p (mt7621+mt7530)
Disabling offloading in fw4 didn't help. But on fw3 it helps. @jow- any chance to improve offloading disabling?

ynezz, 2022-03-13:
@Vladdrako it's not related in any way, this is about mt7622, completely different platform, drivers etc.

0093Axis, 2022-04-07:
I got same issue in Xiaomi Redmi AX6S.
```root@ImmortalWrt:~# cat /sys/fs/pstore/*
Oops#1 Part1
<15>[   15.075610] urandom-seed: Seed file not found (/etc/urandom.seed)
<14>[   15.125279] procd: - early -
<14>[   15.128296] procd: - watchdog -
<14>[   15.682616] procd: - watchdog -
<14>[   15.688127] procd: - ubus (running as root!) -
<14>[   15.743042] procd: - init -
<14>[   16.252109] urngd: v1.0.2 started.
<14>[   16.295468] kmodloader: loading kernel modules from /etc/modules.d/*
<6>[   16.323889] NET: Registered protocol family 38
<6>[   16.332101] tun: Universal TUN/TAP device driver, 1.6
<6>[   16.341732] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
<6>[   16.373919] cryptodev: driver 1.12 loaded.
<6>[   16.379045] GACT probability on
<6>[   16.382777] Mirror/redirect action on
<6>[   16.390620] u32 classifier
<6>[   16.393335]     input device check on
<6>[   16.397038]     Actions configured
<6>[   16.405648] Loading modules backported from Linux version v5.15.8-0-g43e577d7a2cb
<6>[   16.413159] Backport generated by backports.git v5.15.8-1-0-g83f664bb
<6>[   16.477732] xt_time: kernel timezone is -0000
<7>[   16.545317] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
<7>[   16.556073] mt7915e 0000:01:00.0: assign IRQ: got 139
<7>[   16.561217] pci 0000:00:00.0: enabling bus mastering
<6>[   16.566195] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
<7>[   16.572285] mt7915e 0000:01:00.0: enabling bus mastering
<6>[   16.577701] mt7915e 0000:01:00.0: attaching wed device 0
<6>[   16.583137] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   16.590792] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   16.599145] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   16.607794] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<6>[   16.647850] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
<6>[   16.647850] 
<6>[   16.738149] mt7622-wmac 18000000.wmac: N9 Firmware Version: 2.0, Build Time: 20200131180931
<6>[   16.781859] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20201105222230a
<6>[   16.781859] 
<6>[   16.928479] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
<6>[   16.954826] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
<6>[   17.073619] PPP generic driver version 2.4.2
<6>[   17.079208] PPP MPPE Compression module registered
<6>[   17.084939] NET: Registered protocol family 24
<14>[   17.091648] kmodloader: done loading kernel modules from /etc/modules.d/*
<6>[   21.210955] xt_FULLCONENAT: RFC3489 Full Cone NAT module
<6>[   21.210955] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
<6>[   24.063176] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   24.073910] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   24.083027] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   24.095330] br-lan: port 1(eth0) entered blocking state
<6>[   24.100650] br-lan: port 1(eth0) entered disabled state
<6>[   24.106053] device eth0 entered promiscuous mode
<6>[   24.112803] br-lan: port 1(eth0) entered blocking state
<6>[   24.118084] br-lan: port 1(eth0) entered forwarding state
<6>[   24.129330] mtk_soc_eth 1b100000.ethernet eth1: configuring for fixed/rgmii link mode
<6>[   24.137456] mtk_soc_eth 1b100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   24.146014] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
<6>[   26.439831] br-lan: port 2(wlan1) entered blocking state
<6>[   26.445169] br-lan: port 2(wlan1) entered disabled state
<6>[   26.450879] device wlan1 entered promiscuous mode
<6>[   26.486971] br-lan: port 3(wlan0) entered blocking state
<6>[   26.492301] br-lan: port 3(wlan0) entered disabled state
<6>[   26.497873] device wlan0 entered promiscuous mode
<6>[   26.502690] br-lan: port 3(wlan0) entered blocking state
<6>[   26.508022] br-lan: port 3(wlan0) entered forwarding state
<6>[   27.241032] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[   27.247582] br-lan: port 2(wlan1) entered blocking state
<6>[   27.252902] br-lan: port 2(wlan1) entered forwarding state
<6>[   27.447025] br-lan: port 3(wlan0) entered disabled state
<6>[   29.487726] pppoe-wan: renamed from ppp0
[... 140 lines cut ...]
<7>[ 5897.386964] x19: dead0000000000f0 x18: 0000000000000000 
<7>[ 5897.392268] x17: 0000000000000000 x16: 0000000000000000 
<7>[ 5897.397573] x15: 0000000000000000 x14: 0000000000000000 
<7>[ 5897.402878] x13: 0000000000000000 x12: ffffffffffffffff 
<7>[ 5897.408184] x11: 0000000000000040 x10: 0000000000000810 
<7>[ 5897.413489] x9 : 0000000000000000 x8 : 0000000000000000 
<7>[ 5897.418794] x7 : ffffffffffffffff x6 : 0000000000000000 
<7>[ 5897.424099] x5 : 00000000d7099f6c x4 : 0000000000000000 
<7>[ 5897.429404] x3 : ffffff8000ac6e00 x2 : 0000000000000000 
<7>[ 5897.434708] x1 : ffffff8000ac6e00 x0 : 00000000ffffffff 
<7>[ 5897.440013] Call trace:
<7>[ 5897.442456]  nf_flow_offload_ipv6_hook+0xa08/0x1460 [nf_flow_table]
<7>[ 5897.448717]  nf_flow_table_offload_setup+0x444/0x4d8 [nf_flow_table]
<7>[ 5897.455065]  process_one_work+0x1fc/0x398
<7>[ 5897.459066]  worker_thread+0x150/0x4e0
<7>[ 5897.462810]  kthread+0x120/0x128
<7>[ 5897.466031]  ret_from_fork+0x10/0x18
<0>[ 5897.469600] Code: eb1302bf d1004273 54000540 52800014 (f9401263) 
<4>[ 5897.475685] ---[ end trace 3c60db474dc19e45 ]---
<0>[ 5897.486324] Kernel panic - not syncing: Oops: Fatal exception
<2>[ 5897.492064] SMP: stopping secondary CPUs
<0>[ 5897.495980] Kernel Offset: disabled
<0>[ 5897.499460] CPU features: 0x0000002,04002004
<0>[ 5897.503720] Memory Limit: none
```

MeIsReallyBa, 2022-04-10:
@nbd168 Could you please have a look at it? This issue still remains especially when using 2.4G wifi.