openwrt/openwrt #9442: xiaomi-4a-gigabit-edition has a new flash which is EN25QX128@44Mhz cause a endless reboot

FredyVia, 2022-03-11:
I tried [openwrt-21.02.1-squashfs-sysupgrade](https://downloads.openwrt.org/releases/21.02.1/targets/ramips/mt7621/openwrt-21.02.1-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin) and 
[openwrt-21.02.2-squashfs-sysupgrade](https://downloads.openwrt.org/releases/21.02.2/targets/ramips/mt7621/openwrt-21.02.2-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin), both cause a endless reboot,
Due to conditional restrictions, I cannot provide the system startup log.
this is my info in breed(I use this breed:[breed-mt7621-pbr-m1.bin](https://breed.hackpascal.net/breed-mt7621-pbr-m1.bin))
[image]

FredyVia, 2022-03-11:
according to [this commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=17e690017d76c49070bc99d6a376b1926259c5ff) a Optional patch is change the frequency to 44Mhz, but I don't know whether this will cause the performance loss

FredyVia, 2022-03-11:
I tried modify the frequency to 44Mhz,but doesn't work

yasamoka, 2022-03-18:
Have you had any success with this eventually? Thank you.

FredyVia, 2022-03-19:
not yet

bill888uk, 2022-03-19:
Another user reporting similar problem with EON EN25QX128 flash chip
[https://forum.openwrt.org/t/xiaomi-4a-3-gigaport-cannot-flash-openwrt-21-02/122878](url)

Probably unrelated.  But in same thread, another user with Global R4AG couldn't install OpenWrt from **Global** or incorrectly restored '**chinese**' Xiaomi firmware using openwrtinvasion github based instructions from Owrt R4AG device page for **Global** R4AG device - reason for failures unknown. Uses GigaDevice GD25Q128**E** flash chip.  Eventually installed OpenWrt after reflashing Global firmware.

hauke, 2022-03-19:
Could you or anyone else please post a boot log of OpenWrt on such a device? You can obtain it when you connect a serial console to the device. 

Does anyone have a data sheet of the EON EN25QX128 flash chip and could share a link or the document please.

bill888uk, 2022-03-20:
Datasheet ?
https://pdf1.alldatasheet.com/datasheet-pdf/view/458189/EON/EN25Q128-104YIP.html

kopijahe, 2022-03-21:
> Unclear why they would use breed bootloader, as regular openwrt images surely would not work anyway with Breed?

Well, breed offers a web ui to do various things:

 - Full backup and restore of the flash chip
 - Backup and restore of eeprom (Factory) partition
 - Changing MAC Addresses (sometimes needed for some ISPs)
 - Overclock (never tested this, as the heatsink on 4A Gigabit/100M would not be adequate for it)

For using with regular openwrt images, you need to enable environment configuration:

[image]

And add `autoboot.command` `boot flash 0x00180000`:

[image]

After that, you can use it normally, sysupgrade also work.

AbraxasDMG, 2022-03-22:
Hi guys.
Sorry for probably asking the obvious, but... Is this issue related only the Chinese models or for Global/European as well?
After going through a number of openwrt forum topics I still am not certain if this is limited only to the new Chinese devices.

I have just bought 2 new units in Europe and intend to put openwrt on them this weekend. Should I first check something (open the box and look at some labels on the chips, or similar)?

Thanks in advance!

AbraxasDMG, 2022-03-22:
> 

Thank you. I'll do that and see what's there (fingers crossed).

AbraxasDMG, 2022-03-22:
Would someone know if there is an "exhaustive" list of the flash chips expected to work without any issues in the official release (like 21.02)?

I am trying to validate the status for Winbond 25Q128JVSQ (production date 12/2020).

db260179, 2022-03-24:
EN25QX128 is a brand new SPI-NOR flash chip and will need linux kernel support in the spi driver for openwrt to be able to use it.

Bootloaders work independent - uboot or breed

The only solution for for now with boards with this spi chip is replace it with a support flash chip or wait for upstream support.

bill888uk, 2022-03-26:
fwiw, the 4C also uses same flash chip with same problem.

https://forum.openwrt.org/t/new-xiaomi-4c-cannot-install-openwrt-flash-chip-changed-to-en25qx128a/123634

EpicDogger, 2022-04-05:
i dont konw the flash chip ,but with same problem
[https://github.com/openwrt/openwrt/issues/9652](url)
[https://forum.openwrt.org/t/openwrt-on-xiaomi-4a-gigabit-will-infinite-restart/124496](url)

RadioOperator, 2022-04-06:
back to stock FW and try my snapshot build for this device:
[openwrt-snapshot-r19255-009f8afe06-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/8424329/openwrt-snapshot-r19255-009f8afe06-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.zip)

Evan-tc, 2022-04-12:
> Well, breed offers a web ui to do various things:
>
>     * Full backup and restore of the flash chip
>
>     * Backup and restore of eeprom (Factory) partition
>
>     * Changing MAC Addresses (sometimes [...]

does it work?

kopijahe, 2022-04-12:
@Evan-tc: I used it on my devices, both winbond and gigadevice-equipped works.

EON-equipped device should work too, but not openwrt (for now, at least).

Evan-tc, 2022-04-12:
> I used it on my devices, both winbond and gigadevice-equipped works.
> 
> EON-equipped device should work too, but not openwrt (for now, at least).

Thanks, that's meat the devise still can not use the openwrt.  
sad.

Evan-tc, 2022-04-24:
does this issue be solved?

jcmolero, 2022-04-26:
I’ve found one 4A with production date 05/2020, does it have the old giga Device?

bill888uk, 2022-04-27:
fwiw, this PCB with screen printed production date of 2114 (week 14 of 2021?) is fitted with a GigaDevice flash chip:
https://openwrt.org/_detail/media/xiaomi/xiaomi-mi-router-4a-gigabit-board-without-shielding-small.jpg?id=inbox%3Atoh%3Axiaomi%3Axiaomi_mi_router_4a_gigabit_edition

db260179, 2022-06-13:
I've added the new EON flash chip to my gitlab branch - https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8

Also prebuilt some images for the Xiaomi 4/4AG Gigabit models for testing - 21.02-snapshot

Any feedback welcome, im only doing Xiaomi 4AG models, nothing else, if people want to build for their mt7620 xiaomi models then just build from my source code as above address.

Download for prebuilt image - https://drive.google.com/drive/folders/1bc5dCxMRMwbkVDSNgYFkol4hZqry2WPf?usp=sharing

rn7s2, 2022-07-03:
> I've added the new EON flash chip to my gitlab branch - https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8
>
> Also prebuilt some images for the Xiaomi 4/4AG [...]

I have a Mi-4A-Gigabit, production date March, 2022

It has an `spi-nor spi0.0: en25qx128a (16384 Kbytes)` flash.

I tested your prebuilt image and it works great for me. It resolves the reboot-loop.

Thanks.

heastern, 2022-07-21:
> @db260179
> 
> > I've added the new EON flash chip to my gitlab branch - https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8
> > Also prebuilt some images for the Xiaomi 4/4AG Gigabit models for testing - 21.02-snapshot
> > Any feedback welcome, im only doing Xiaomi 4AG models, nothing else, if people want to build for their mt7620 xiaomi models then just build from my source code as above address.
> > Download for prebuilt image - https://drive.google.com/drive/folders/1bc5dCxMRMwbkVDSNgYFkol4hZqry2WPf?usp=sharing
> 
> I have a Mi-4A-Gigabit, production date March, 2022
> 
> It has an `spi-nor spi0.0: en25qx128a (16384 Kbytes)` flash.
> 
> I tested your prebuilt image and it works great for me. It resolves the reboot-loop.
> 
> Thanks.

[image]
 endless reboot

pikerr, 2022-07-25:
> @db260179
>
>
> I have a Mi-4A-Gigabit, production date March, 2022
>
> It has an `spi-nor spi0.0: en25qx128a (16384 Kbytes)` flash.
>
> I tested your prebuilt image and it works great for me. It resolves the [...]

I also managed flashing my Mi-4A-G successfully after bootloop. But now can't persist anything after reboot, eg password.

db260179, 2022-07-27:
> #### I also managed flashing my Mi-4A-G successfully after bootloop. But now can't persist anything after reboot, eg password.

Make sure to use openwrt-21.02-snapshot-r16618+25-e43d6e3f51-ramips…iaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin and not the initramfs image!

allgood, 2022-08-14:
Booted chinese 4ag with prebuilt images from @db260179 , but was unable to install some kmod packages. Is there a place with the kernel dependent packages available or do I need to build them from source?

db260179, 2022-08-15:
> [https://gitlab.com/db260179/openwrt-base/](https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8)

Yep, im not sure if these chips have been added in openwrt as of yet, so for now will have to build your own image

[https://gitlab.com/db260179/openwrt-base/](https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8)

Piradoxlanieve, 2022-08-20:
yesterday I bought in the xiaomi spain store (https://www.mi.com/es/product/mi-router-4a-gigabit-edition/), I suppose it will arrive next week, it has been more than 12 years or more since I flashed a router, my last one was a wrt-54g with dd-wrt, I suppose it has the en25qx128a chip Would I have to use this firmware openwrt-21.02-snapshot-r16618+25-e43d6e3f51-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel? and this guide https://www.radiusdesk.com/docuwiki/2021/xiaomi_flash? Thank you.  @db260179

db260179, 2022-08-23:
@Piradoxlanieve

> Make sure to use openwrt-21.02-snapshot-r16618+25-e43d6e3f51-ramips…iaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin and not the initramfs image!

Piradoxlanieve, 2022-08-23:
> @Piradoxlanieve
> 
> > ¡Asegúrese de usar openwrt-21.02-snapshot-r16618+25-e43d6e3f51-ramips…iaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin y no la imagen initramfs!

Can I use today's version of openwrt selector snapshot?

Today they give me the router.

Model:
Xiaomi Mi Router 4A Gigabit Edition
Platform:
ramips/mt7621
Version:
SNAPSHOT (r20375-d6994c53cd)
Date:
2022-08-22 17:34:33
https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin

 @db260179 thanks for your job.

Sorry my bad english.

kennymnk, 2022-08-23:
As the next stable version 22.03 is in the stage of rc6 now, should we still use the firmware openwrt-21.02-snapshot-r16618+25-e43d6e3f51-ramips…iaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin for en25qx128a flash?

db260179, 2022-08-23:
> Can I use today's version of openwrt selector snapshot?
>
> Today they give me the router.
>
> Model: Xiaomi Mi Router 4A Gigabit Edition Platform: ramips/mt7621 Version: SNAPSHOT (r20375-d6994c53cd) [...]

Dont know?
Unless openwrt have added this spi flash chip, then only my 21.02 will work. Otherwise you will need to patch the kernel to support en25qx128a.

Piradoxlanieve, 2022-08-27:
> @Piradoxlanieve
> 
> > Make sure to use openwrt-21.02-snapshot-r16618+25-e43d6e3f51-ramips…iaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin and not the initramfs image!

I have been working with this firmware and the traffic between vlans was connected to a 300mbps max spee to (iperf) tp-link sg105e, however with the openwrt snapshot it works perfectly at 1000mbps, I have disassembled the router and it has the Gigadevice 25Q128ES fash.
Sorry my bad english

yukipastelcat, 2022-09-13:
I can't seem to figure out the instructions to install 21.02 build for a router with chip id 1c7118 after I flashed it to Padavan firmware, for me it keeps booting into Breed, can someone post the instructions/show where to find them?

mark4z, 2022-09-19:
> Dont know?
> Unless openwrt have added this spi flash chip, then only my 21.02 will work. Otherwise you will need to patch the kernel to support en25qx128a.

And How to get sheep for this one? Big Lao

FredyVia, 2022-09-29:
I got the flash model after I open the case, it's "cFeon QH128A-104HIP"

FredyVia, 2022-09-29:
root@XiaoQiang:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 01000000 00010000 "ALL"
mtd1: 00030000 00010000 "Bootloader"
mtd2: 00010000 00010000 "Config"
mtd3: 00010000 00010000 "Bdata"
mtd4: 00010000 00010000 "Factory"
mtd5: 00010000 00010000 "crash"
mtd6: 00010000 00010000 "cfg_bak"
mtd7: 00100000 00010000 "overlay"
mtd8: 00d00000 00010000 "OS1"
mtd9: 00b30000 00010000 "rootfs"
mtd10: 00180000 00010000 "disk"

NorthBig, 2022-12-16:
> 

My R4A GB flash is Winbond 250128JVSQ. I have a breed in it now, But I don't know which OpenWrt is OK for me. Could you pls tell me which firm is OK?

NorthBig, 2022-12-16:
> > @db260179
> > > I've added the new EON flash chip to my gitlab branch - https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8
> > > Also prebuilt some images for the Xiaomi 4/4AG Gigabit models for testing - 21.02-snapshot
> > > Any feedback welcome, im only doing Xiaomi 4AG models, nothing else, if people want to build for their mt7620 xiaomi models then just build from my source code as above address.
> > > Download for prebuilt image - https://drive.google.com/drive/folders/1bc5dCxMRMwbkVDSNgYFkol4hZqry2WPf?usp=sharing
> > 
> > 
> > I have a Mi-4A-Gigabit, production date March, 2022
> > It has an `spi-nor spi0.0: en25qx128a (16384 Kbytes)` flash.
> > I tested your prebuilt image and it works great for me. It resolves the reboot-loop.
> > Thanks.
> 
> #### I also managed flashing my Mi-4A-G successfully after bootloop. But now can't persist anything after reboot, eg password.
> #### Another Mi 4A-G router's dmesg with healthy MTD partitioning:
> [    0.614908] spi-nor spi0.0: w25q128 (16384 Kbytes)
> [    0.619787] 8 fixed-partitions partitions found on MTD device spi0.0
> [    0.626135] Creating 8 MTD partitions on "spi0.0":
> [    0.630930] 0x000000000000-0x000000030000 : "u-boot"
> [    0.637423] 0x000000030000-0x000000040000 : "u-boot-env"
> [    0.644011] 0x000000040000-0x000000050000 : "Bdata"
> [    0.650103] 0x000000050000-0x000000060000 : "factory"
> [    0.656590] 0x000000060000-0x000000070000 : "crash"
> [    0.662716] 0x000000070000-0x000000080000 : "cfg_bak"
> [    0.669258] 0x000000080000-0x000000180000 : "overlay"
> [    0.675752] 0x000000180000-0x000001000000 : "firmware"
> [    0.682344] 2 uimage-fw partitions found on MTD device firmware
> [    0.688307] Creating 2 MTD partitions on "firmware":
> [    0.693275] 0x000000000000-0x000000242995 : "kernel"
> [    0.699653] 0x000000242995-0x000000e80000 : "rootfs"
> [    0.705881] mtd: device 9 (rootfs) set to be root filesystem
> [    0.711685] 1 squashfs-split partitions found on MTD device rootfs
> [    0.717879] 0x000000570000-0x000000e80000 : "rootfs_data"
> [    0.750964] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module
> #### Current router's dmesg with bad MTD partitioning:
> [    3.692132] spi-mt7621 1e000b00.spi: sys_freq: 220000000
> [    3.698407] spi-nor spi0.0: en25qx128a (16384 Kbytes)
> [    3.703561] 8 fixed-partitions partitions found on MTD device spi0.0
> [    3.709968] Creating 8 MTD partitions on "spi0.0":
> [    3.714774] 0x000000000000-0x000000030000 : "u-boot"
> [    3.721254] 0x000000030000-0x000000040000 : "u-boot-env"
> [    3.727835] 0x000000040000-0x000000050000 : "Bdata"
> [    3.734094] 0x000000050000-0x000000060000 : "factory"
> [    3.740546] 0x000000060000-0x000000070000 : "crash"
> [    3.746827] 0x000000070000-0x000000080000 : "cfg_bak"
> [    3.753188] 0x000000080000-0x000000180000 : "overlay"
> [    3.759664] 0x000000180000-0x000001000000 : "firmware"
> [    3.798778] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module

I have got R4A GB with winbond 250128JVSQ, and I have installed Breed console in it now. Can I use Breed to flash your bin file in Google share? Wait....

closed 2022-12-16 by openwrt-bot

vivek-gangani, 2022-12-23:
> I've added the new EON flash chip to my gitlab branch - https://gitlab.com/db260179/openwrt-base/-/commit/e43d6e3f51d453b69d658b11eea3bacd292487b8
>
> Also prebuilt some images for the Xiaomi 4/4AG [...]

Thanks !! This worked for me, now how do I upgrade openwrt to latest version 22.03 ?

commit d7876daf (2022-12-16): generic: add support for EON EN25QX128A spi nor flash

Add support for EON EN25QX128A spi nor flash with no flags as it does
support SFDP parsing.

Fixes: #9442

Tested-by: Szabolcs Hubai <szab.hu@gmail.com> [ramips/mt7621: xiaomi_mi-router-4a-gigabit]
PR #11537 (2022-12-16): generic: add support for EON EN25QX128A spi nor flash
Add support for EON EN25QX128A spi nor flash with no flags as it does support SFDP parsing.

Fixes: #9442

Tested-by: Szabolcs Hubai <szab.hu@gmail.com> [ramips/mt7621: xiaomi_mi-router-4a-gigabit]

---

Split from #10971
commit b18a0d0b (2022-12-17): generic: add support for EON EN25QX128A spi nor flash

Add support for EON EN25QX128A spi nor flash with no flags as it does
support SFDP parsing.

Fixes: #9442

Tested-by: Szabolcs Hubai <szab.hu@gmail.com> [ramips/mt7621: xiaomi_mi-router-4a-gigabit]
(cherry picked from commit d7876daf6552a9f39bd5e0bf50b554e9406ec275)
PR #11539 (2022-12-17): [22.03] generic: add support for EON EN25QX128A spi nor flash
Add support for EON EN25QX128A spi nor flash with no flags as it does support SFDP parsing.

Fixes: #9442

Tested-by: Szabolcs Hubai <szab.hu@gmail.com> [ramips/mt7621: xiaomi_mi-router-4a-gigabit] 
(cherry picked from commit d7876daf6552a9f39bd5e0bf50b554e9406ec275)