openwrt/openwrt #9448: wireguard-tools: set default gateway

mikhirev, 2022-03-11:
I want to use a wireguard interface with mwan3, but it requires a default gateway being assigned for the interface. As fair as I can see, currently it is possible only by executing `ip route add default via <address> dev <interface> metric <metric>` manually. Please add uci configuration options for setting default gateway for wg interfaces.

BTW there's already `metric` option being set by luci, but it seems unused.

Leo-PL, 2022-04-23:
Wireguard protocol handler will set default gateway if one of peers has a default prefix (0.0.0.0/0 or ::/0) assigned,

mikhirev, 2022-04-23:
My config is

```
config interface 'wg0'
        option proto 'wireguard'
        option private_key '<...>'
        list addresses '192.0.2.2/24'
        option gateway '192.0.2.1'

config wireguard_wg0
        option public_key '<...>'
        option endpoint_host 'example.com'
        list allowed_ips '0.0.0.0/0'
```

and  I need a default gateway to be set for this interface. It is not set when `wg0` interface is started. I'll check this once again after upgrade, I'm still on 21.02.1.

Leo-PL, 2022-04-23:
Once packets reach Wireguard interface targeting the default route set on the interface, the routing inside is handled according to the "AllowedIPs" option. See here: https://www.wireguard.com/#cryptokey-routing. I don't see the point why do you need to override this setting, unless "RouteAllowedIPs" option is disabled - in such case you need to add static routes manually.

mikhirev, 2022-04-23:
Maybe the problem is that mwan3 does not route packets to the interface until it has a default gateway?

champtar, 2022-04-25:
@mikhirev here my redacted config that work with mwan3, I use 0/0 route instead of gateway
```
config interface 'wg0'
	option proto 'wireguard'
	option private_key 'REDACTED'
	list addresses '10.255.254.253/32'
	option metric '201'

config wireguard_wg0
	option description 'wg0'
	option public_key 'REDACTED'
	option preshared_key 'REDACTED'
	list allowed_ips '0.0.0.0/0'
	option endpoint_host 'REDACTED'
	option endpoint_port '65432'
	option persistent_keepalive '25'

config route
	option interface 'wg0'
	option target '0.0.0.0'
	option netmask '0.0.0.0'
	option metric '201'

config route
	option interface 'wg0'
	option target '10.255.254.0'
	option netmask '255.255.255.0'
	option metric '201'
```