openwrt/openwrt #9459: Mixed up switch visualization in FritzBox 7530/7520

hollyclergyman, 2022-03-13:
Hej,

as shown in the picture appended to this report, the luci GUI mixes up the correct order of LAN ports available to a FritzBox 7530/7520. As I confirmed, this confusion only affects the luci GUI, the corresponding /etc/config/network file still contains all entries in a correct manner. 
[image]
Thanks a lot for looking at the problem.

Kind regards

Thomas

abajk, 2022-03-13:
@hollyclergyman 
Modify [thin line](https://github.com/openwrt/openwrt/blob/ec6293febc244d187e71a6e54f44920be679cde4/target/linux/ipq40xx/base-files/etc/board.d/02_network#L64) like this and let us know if it works correctly:
~~~
			"0u@eth0" "1:lan:4" "2:lan:3" "3:lan:1" "4:lan:2"
~~~

abajk, 2022-03-13:
[Here](https://forum.openwrt.org/t/openwrt-21-02-0-second-release-candidate/98026/120) it is mentioned that:
`LAN ports are not configured propley For example LAN 1 is LAN 4 and LAN4 is LAN1...`
This conflicts with your description of the ports.

hollyclergyman, 2022-03-13:
> [Here](https://forum.openwrt.org/t/openwrt-21-02-0-second-release-candidate/98026/120) it is mentioned that: `LAN ports are not configured propley For example LAN 1 is LAN 4 and LAN4 is LAN1...` This conflicts with your description of the ports.

Well, the ports work as intended and according to the definition in the /etc/config/network file, I may try your solution from this reply https://github.com/openwrt/openwrt/issues/9459#issuecomment-1066111246, which hopefully fixes the issue in my current version OpenWrt 21.02.2 r16495-bf0c965af0

maurerle, 2022-12-11:
I can confirm that there is something wrong:
I have 2 cables plugged in - into port 1 and port 3

## This is how it looks like on openwrt 19.07 (correct visualization)
[image]

## And thats how it looks like on the latest 22.03 (ports are off by one)

[image]

Djfe, 2023-02-11:
@hollyclergyman Please close this issue as it has been resolved :)
(as far as I understand it, I could be wrong)

hollyclergyman, 2023-02-11:
> @hollyclergyman Please close this issue as it has been resolved :) (as far as I understand it, I could be wrong)

Hello,
As you can see from the previous comment, dating to the 11th of December 2022, it has not been resolved recently. I can confirm from my setup, that still mixed up visualization persists. Thus, I may still leave this issue open, even though it is a minor one.

Djfe, 2023-02-11:
In that case leave it open. Sorry 😅