openwrt/openwrt #9463: packet loss with more than one wan

ezplanet, 2022-03-13:
Hardware: Linksys WRT 3200 
OpenWrt 21.02-SNAPSHOT, r16514-604274c24b
This has always been happening on Linksys wrt 3200 with OpenWrt 21.02 (all good with OpenWrt 19)

Configuration with 2 wan interfaces as follows:

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd4c:70eb:7a40::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	option mtu '1492'
	option ipv6 '0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.10.4'
	option netmask '255.255.255.0'
	option ip6assign '60'
	list dns '192.168.10.19'
	list dns '192.168.10.20'
	list dns '8.8.8.8'
	list dns_search 'ezplanet.org'
	option defaultroute '0'

config interface 'wan1'
        option proto 'pppoe'
        option device 'wan'
        option username 'username'
        option password 'password'
        option ipv6 '0'
        option delegate '0'
        option dns_metric '2'
        option metric '2'

config device
        option name 'lan1'
        option mtu '1492'
        option ipv6 '0'

config interface 'wan2'
        option proto 'static'
        option device 'lan1'
        option ipaddr 'x.x.x.92'
        option netmask '255.255.255.240'
        option gateway 'x.x.x.81'
        list dns 'x.x.x.82'
        list dns 'x.x.x.90'
        option delegate '0'
        option dns_metric '1'
        option metric '1'

Now if I ping x.x.x.81 as follows:
ping -c 10 -I lan1 x.x.x.81

most packets get lost.
Is this a bug or a configuration issue? Please note that the same configuration on a BT Home Hub 5 with the same 21.02 OpenWrt release works fine. Could it be an issue with WRT3200 switch in 21.02?

ezplanet, 2022-03-16:
I understand that there aren't many using dual or more wan configuration, however this is a blocking issue that makes OpenWRT 21.02 unusable. Is there any plan to fix it?
In the meantime I recommend everyone to stay with the more stable 19.07

sahilister, 2024-01-16:
Observing similar issue on RB750Gr3 running OpenWrt 21.02.5

Added details at https://forum.openwrt.org/t/multi-wan-packet-loss-on-21-02-all-good-on-19-07/122895/13?u=sahilister