openwrt/openwrt #9464: ath9k_htc: Firmware - ath9k_htc/htc_9271-1.4.0.fw download failed

sinancetinkaya, 2022-03-13:
I have two usb adapters plugged into the router. mt76x2u chipset one is working and ath9k one is not working
```
[    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 SMP Wed Feb 16 20:29:10 2022
[    0.000000] SoC: xRX200 rev 1.1
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019555 (MIPS 34Kc)
[    0.000000] MIPS: machine is ZyXEL P-2812HNU-F1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Detected 1 available secondary CPU(s)
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 14 pages/cpu s26256 r8192 d22896 u57344
[    0.000000] pcpu-alloc: s26256 r8192 d22896 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyLTQ0,115200
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0000a160
[    0.000000] Readback ErrCtl register=0000a160
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120648K/131072K available (6258K kernel code, 219K rwdata, 1040K rodata, 1232K init, 223K bss, 10424K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from 0x8075ca38 with crng_init=0
[    0.000000] CPU Clock: 500MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[    0.000011] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns
[    0.007906] Calibrating delay loop... 332.54 BogoMIPS (lpj=665088)
[    0.045951] pid_max: default: 32768 minimum: 301
[    0.050877] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.058068] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.069440] rcu: Hierarchical SRCU implementation.
[    0.075026] smp: Bringing up secondary CPUs ...
[    0.080730] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.080747] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.080869] CPU1 revision is: 00019555 (MIPS 34Kc)
[    0.111667] Synchronize counters for CPU 1: done.
[    0.136014] smp: Brought up 1 node, 2 CPUs
[    0.146898] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.156549] futex hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.163609] pinctrl core: initialized pinctrl subsystem
[    0.171141] NET: Registered protocol family 16
[    0.183677] dcdc-xrx200 1f106a00.dcdc: Core Voltage : 1016 mV
[    0.197124] pinctrl-xway 1e100b10.pinmux: Init done
[    0.206049] dma-xway 1e104100.dma: Init done - hw rev: 7, ports: 7, channels: 28
[    0.218625] PCI host bridge /fpi@10000000/pci@e105400 ranges:
[    0.224310]  MEM 0x0000000018000000..0x0000000019ffffff
[    0.229553]   IO 0x000000001ae00000..0x000000001affffff
[    0.276327] gpio-stp-xway 1e100bb0.stp: Init done
[    0.283358] usbcore: registered new interface driver usbfs
[    0.288971] usbcore: registered new interface driver hub
[    0.294326] usbcore: registered new device driver usb
[    0.300138] PCI host bridge to bus 0000:00
[    0.304170] pci_bus 0000:00: root bus resource [mem 0x18000000-0x19ffffff]
[    0.311068] pci_bus 0000:00: root bus resource [io  0x1ae00000-0x1affffff]
[    0.318050] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.324869] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[... 150 lines cut ...]
[   14.847048] usb 2-1: ath9k_htc: USB layer deinitialized
[   30.423026] device eth0 entered promiscuous mode
[   30.427253] br-lan: port 1(eth0.1) entered blocking state
[   30.431747] br-lan: port 1(eth0.1) entered disabled state
[   30.437691] device eth0.1 entered promiscuous mode
[   33.460845] lantiq,xrx200-net 1e108000.eth eth0: port 0 got link
[   33.468766] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.482817] br-lan: port 1(eth0.1) entered blocking state
[   33.486849] br-lan: port 1(eth0.1) entered forwarding state
[   33.496278] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   45.343769] Lantiq PTM driver, version 1.0.27, (c) 2001-2013 Lantiq Deutschland GmbH
[   45.343769]     PTM (E1) firmware version 2.16.0
[   45.354796] ifxmips_ptm: PTM init succeed
[   97.976196] enter showtime
[   97.977780] IPv6: ADDRCONF(NETDEV_CHANGE): dsl0: link becomes ready
[   97.989367] IPv6: ADDRCONF(NETDEV_CHANGE): dsl0.35: link becomes ready
[   97.999752] enter showtime
[   99.282877] pppoe-wan: renamed from ppp0
[  135.212467] br-lan: port 2(wlan0) entered blocking state
[  135.216409] br-lan: port 2(wlan0) entered disabled state
[  135.222524] device wlan0 entered promiscuous mode
[  152.560462] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  152.566016] br-lan: port 2(wlan0) entered blocking state
[  152.570732] br-lan: port 2(wlan0) entered forwarding state
```

sinancetinkaya, 2022-03-14:
Same bug has been reported here https://forum.openwrt.org/t/lantiq-xrx200-fb-3370-usb-wifi-device-not-working-kmod-ath9k-htc/17552

chunkeey, 2022-03-21:
Ok, this seems to be a combination of dwc2 + ath9k_htc + (big endian?).

I can reproduce this error with a MyBook Live DUO (NAS, powerpc apm82181, but uses dwc2 for the USB-port as well).
Thing is,  a NetGear WNDR4700 (Same PowerPC APM82181) which has a Renesas xhci-pci supported chip for the usb-ports has no problems with ath9k_htc and neither does a iBook G4 (PowerPC). 

It would be interesting to know, if a RPI1/2 (uses dwc2 too) could check out ath9k_htc.

(I've tested with other USB Wifi chip as well. carl9170, rt2x00, p54usb, rtl8192cu, rtl8xxxu they all probed successfully with the MyBook Live DUO)

sinancetinkaya, 2022-03-21:
@chunkeey What's your kernel version? This bug has been reported here https://bugzilla.kernel.org/show_bug.cgi?id=208251
It's said that it's been [fixed](https://bugzilla.kernel.org/show_bug.cgi?id=208251#c30)

>  Viktor Jägersküpper 2020-07-31 17:26:53 UTC
> 
> The original patch (which caused the problems) and the fix by Mark O'Donovan are now included in the stable and longterm kernels (5.7.11, 5.4.54, 4.19.135, 4.14.190, 4.9.232, 4.4.232 and higher).

Can you try the wifi adapter with a newer kernel?

chunkeey, 2022-03-21:
> Can you try the wifi adapter with a newer kernel?

OpenWrt uses ath9k_htc from the mac80211 backport. This is 5.15.8 with additional patches:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/kernel/mac80211/Makefile;h=715e851a13ef2f9635e68f68331a0119f2e09c3c;hb=HEAD#l13

On the MyBook Live I've tested with the OpenWrt 21.02.02, Current OpenWrt Snapshot (5.10.107+mac80211-5.15.8) and a Debian PowerPC Image with a Vanilla "5.17-rc8". All refused to work with the same error as the op.

On the WNDR4700 (same ARCH as the MyBook Live! But it uses the dwc2 internally, it's hardwired on the PCB to the internal SD-Card reader.). I tested with the same OpenWrt snapshot (5.10.107+mac80211-5.15.8). It works there just fine since it's supported by the Renesas xhci-pci usb ports.

The ath9k-htc firmware github page lists some issues related to dwc2:
https://github.com/qca/open-ath9k-htc-firmware/issues/154 :disappointed: 
https://github.com/qca/open-ath9k-htc-firmware/issues/124 <--- worth a try... But sadly outside of the scope of OpenWrt.