openwrt/openwrt #9467: IPQ806x: TP-Link Archer VR2600v, Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

botischroeder, 2022-03-13:
The TP-Link Archer VR2600v is stuck in a boot loop on written snapshot image. It's able to boot using the snapshot uimage though, but there ath10k firmware can't be found. 21.02.2 release version doesn't have either problem. 
@chunkeey: You recently made some changes for that target related to partitions. Could you please check if your changes are causing the boot loop? Thank you!

Log from openwrt-ipq806x-generic-tplink_vr2600v-squashfs-sysupgrade.bin
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.103 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r19085-289fbc5102) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Mar 10 02:43:32 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: TP-Link Archer VR2600v
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] percpu: Embedded 15 pages/cpu s30796 r8192 d22452 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line:
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8 root=mtd:rootfs rootfstype=squashfs
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 475432K/491520K available (6977K kernel code, 615K rwdata, 1688K rodata, 1024K init, 241K bss, 16088K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3a0/0x54c with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000024] Switching to timer-based delay loop, resolution 160ns
[    0.000293] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000324] pid_max: default: 32768 minimum: 301
[    0.000502] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000523] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001625] CPU: Testing write buffer coherency: ok
[    0.001916] qcom_scm: convention: smc legacy
[    0.002853] Setting up static identity map for 0x42300000 - 0x42300060
[    0.003027] rcu: Hierarchical SRCU implementation.
[    0.003311] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.003554] smp: Bringing up secondary CPUs ...
[    0.005478] smp: Brought up 1 node, 2 CPUs
[    0.005502] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005516] CPU: All CPU(s) started in SVC mode.
[    0.016351] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016526] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016559] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.016710] pinctrl core: initialized pinctrl subsystem
[    0.018000] NET: Registered protocol family 16
[    0.018332] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019698] thermal_sys: Registered thermal governor 'step_wise'
[    0.021868] cpuidle: using governor ladder
[    0.021952] cpuidle: using governor menu
[    0.066821] usbcore: registered new interface driver usbfs
[    0.066910] usbcore: registered new interface driver hub
[    0.066986] usbcore: registered new device driver usb
[    0.067050] pps_core: LinuxPPS API ver. 1 registered
[    0.067066] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.067105] PTP clock support registered
[    0.069307] clocksource: Switched to clocksource dg_timer
[    0.070412] NET: Registered protocol family 2
[    0.070583] IP idents hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.071553] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.071614] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.071658] TCP bind hash table entries: 4096 (order: 3, 32768 bytes, linear)
[... 458 lines cut ...]
[   11.427319] ath10k_pci 0000:01:00.0: could not fetch firmware files (-12)
[   11.437046] ath10k_pci 0000:01:00.0: could not probe fw (-12)
[   11.450203] kmodloader: done loading kernel modules from /etc/modules.d/*
[   27.933607] dwmac1000: Master AXI performs any burst length
[   27.933647] ipq806x-gmac-dwmac 37400000.ethernet eth1: No Safety Features support found
[   27.948004] ipq806x-gmac-dwmac 37400000.ethernet eth1: IEEE 1588-2008 Advanced Timestamp supported
[   27.948145] ipq806x-gmac-dwmac 37400000.ethernet eth1: registered PTP clock
[   27.956020] ipq806x-gmac-dwmac 37400000.ethernet eth1: configuring for fixed/sgmii link mode
[   27.963097] ipq806x-gmac-dwmac 37400000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[   27.971581] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   27.983625] br-lan: port 1(eth1.1) entered blocking state
[   27.986290] br-lan: port 1(eth1.1) entered disabled state
[   27.992121] device eth1.1 entered promiscuous mode
[   27.997175] device eth1 entered promiscuous mode
[   28.004710] br-lan: port 1(eth1.1) entered blocking state
[   28.006691] br-lan: port 1(eth1.1) entered forwarding state
[   28.022046] dwmac1000: Master AXI performs any burst length
[   28.022082] ipq806x-gmac-dwmac 37200000.ethernet eth0: No Safety Features support found
[   28.036444] ipq806x-gmac-dwmac 37200000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   28.036593] ipq806x-gmac-dwmac 37200000.ethernet eth0: registered PTP clock
[   28.047396] ipq806x-gmac-dwmac 37200000.ethernet eth0: configuring for fixed/rgmii link mode
[   28.051554] ipq806x-gmac-dwmac 37200000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   28.079968] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   28.971157] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

chunkeey, 2022-03-13:
> @chunkeey: You recently made some changes for that target related to partitions. Could you please check if your changes are causing the boot loop? Thank you!

very likely yes. The vr2600v ran out of space for its kernel partition on the flash. From you bootlog, I can see that the uImage mtdsplitter doesn't pick up on the uImage header. I didn't notice this earlier, but there's an 512-byte padding in front of the sysupgrade production rule under image/generic.mk. I "think" this fixed it?

Can you apply this patch, compile an image for your vr2600v and report back whenever this helped?

```
diff --git a/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8064-vr2600v.dts b/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8064-vr2600v.dts
index 8fa0a5d291..a3fd867e2a 100644
--- a/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8064-vr2600v.dts
+++ b/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8064-vr2600v.dts
@@ -257,7 +257,8 @@
 				partition@320000 {
 					label = "firmware";
 					reg = <0x320000 0xc60000>;
-					compatible = "denx,uimage";
+					compatible = "openwrt,uimage";
+					openwrt,offset = <512>; /* account for pad-extra 512 */
 				};
 
 				/* hole 0xf80000 - 0xfaf100 */
```

botischroeder, 2022-03-14:
Thank you for your super fast response! The patch fixes both problems.

chunkeey, 2022-03-16:
The same fix was just merged.

closed 2022-03-16 by chunkeey