openwrt/openwrt #9470: Daylight savings not working correctly

darisb, 2022-03-14:
We live in the Los Angeles, CA, USA time zone (GMT -8).  When we go to daylight savings time, our clocks "spring forward".  Two AM becomes 3 AM.  I have created some firewall traffic rules that block some devices at specific times. These rules function properly when our time is normal. When we enter daylight savings time, devices that should be blocked continue to work after the designated time.  I must adjust the rules two hours back.  So if a device should be blocked at 22:00, I must set the specified time to 20:00 in order to block the device at 10 PM. When we exit daylight savings time and return to normal time, the I must reset the specified time on my traffic rules to the precise time that I want.

I first noticed this when I upgraded to OpenWRT 21.x.x last summer or fall.  It seemed to work correctly in version 19.  I recently upgraded to 21.02.2 r16495 and it's still happening.  My router is set to use NTP, and the system time is correct.

darisb, 2022-03-19:
At some point this week, the offset seems to have decreased.  Now I am only a hour head.  If i want the rule to stop allowing connections at 22:00, I now have to set the time to 21:00.

az0, 2023-11-06:
I am having the same problem going from MDT to MST on OpenWrt 23.05 / Belkin RT3200. We lost DST this morning, and today the firewall rule for 23:30 went into effect at 22:30 instead.

az0, 2024-11-05:
This happened again yesterday (first day after DST ended) with OpenWrt 23.05.5 . I had crontab `0 0 * * * service system restart`, but I think it needs to be moved a few hours later.

Spudz76, 2024-11-05:
Initial reflex without checking is that when a rule is set the time is converted to and stored as UTC which does not have the concept of savings.  But then the interface should also shift by an hour when converted back for display.

az0, 2024-11-05:
The firewall rule that was supposed to start at 11:40pm started at 10:40pm today and yesterday.

These did not help
* `service system restart` : no output, no effect
* `hwclock -u --systz`: no output, no effect
* `date -k`: unrecognized option

Command line `date` and LuCI showed correct current MST, and the firewall rule in LuCI showed the correct start time 11:40pm.

**What helped was `fw4 reload`**

darisb, 2024-11-05:
This has been going on for years.  If you do nothing, the rule(s) will begin using the correct time about 1 to 2 weeks after the official time change.
Based on my observations, I believe what's going on is the schedule for the network firewall rules checks time against UTC and adjusts for the daylight savings settings. BUT, the day light savings dates are wrong. 
My firewall uses NTP, and the time is always correct.  It seems like it would be more accurate if the network firewall rules uses the local time or the firewall.rules.
Daris