openwrt/openwrt #9489: netfilter: Linux kernel build fails when IPv6 is disabled

c---, 2022-03-17:
Environment:
OpenWrt 21.02.2
mvebu-cortexa9-linksys_wrt3200acm

Using the default configuration from:
https://downloads.openwrt.org/releases/21.02.2/targets/mvebu/cortexa9/config.buildinfo

```shell
make menuconfig
# Disable IPv6
Global build settings ---> Enable IPv6 support in packages ---> N
```
Build fails in `nft_core` related to missing `nft_reject_ipv6` module. I looked briefly to see if there was an easy fix but could not determine what the problem is in a short time. In the CONFIG settings nft_reject_ipv6 **is** disabled as it should be so I don't know why the build script is requiring that module.
```shell
make[3]: Entering directory '/home/openwrt/package/kernel/linux'
mkdir -p /home/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/stamp

SHELL= flock /home/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/packages/.pkgdir/kernel/. /home/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/'
touch /home/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/stamp/.kernel_installed
ERROR: module '/home/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.4.182/net/ipv6/netfilter/nft_reject_ipv6.ko' is missing.
make[3]: *** [modules/netfilter.mk:1068: /home/openwrt/bin/targets/mvebu/cortexa9/packages/kmod-nft-core_5.4.182-1_arm_cortex-a9_vfpv3-d16.ipk] Error 1
make[3]: Leaving directory '/home/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#1.83#0.13#1.93
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/openwrt/'
make[1]: *** [package/Makefile:110: /home/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/'
make: *** [/home/openwrt/include/toplevel.mk:230: world] Error 2
```

M95D, 2022-04-01:
See https://github.com/openwrt/openwrt/issues/9580

c---, 2022-04-01:
Looks like it's going to remain broken until removed at some future release.
[Won't Fix]

closed 2022-04-01 by c---

ddmesh, 2023-01-06:
Droping IPv6 as a solution breaks a lot of projects depending on openwrt. If this happends, then we have 
huge problems which can not be solved and the project (freifunk) has do be end in near future.

see:  #10471 #9580