openwrt/openwrt #9491: IPROUTE2: package/network/utils/iproute2 failed to build (build variant: tcfull). LIBBPF_FORCE=on set, but couldn't find a usable libbpf

okibcn, 2022-03-17:
Maintainer: @guidosarducci @hauke

# ERROR DESCRIPTION
When building the config file below, the build **always** fails when building IPROUTE2-TCFULL package using a single thread. The full dump is attached as a txt file [IPROUTE2-TCFULL_build_dump.txt](https://github.com/openwrt/openwrt/files/8289367/IPROUTE2-TCFULL_build_dump.txt)

These are the last few lines:
```make
make[3]: Entering directory '/build/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/iproute2-tcfull/iproute2-5.15.0'
TC schedulers
 ATM    no
 IPT    using xtables
 IPSET  yes

iptables modules directory: /usr/lib/iptables
libc has setns: yes
libc has name_to_handle_at: yes
SELinux support: no
libbpf support: no
        LIBBPF_FORCE=on set, but couldn't find a usable libbpf
make[3]: *** [Makefile:84: config] Error 1
make[3]: Leaving directory '/build/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/iproute2-tcfull/iproute2-5.15.0'
make[2]: *** [Makefile:271: /build/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/iproute2-tcfull/iproute2-5.15.0/.built] Error 2
make[2]: Leaving directory '/build/openwrt/package/network/utils/iproute2'
time: package/network/utils/iproute2/tcfull/compile#0.61#0.16#0.65
    ERROR: package/network/utils/iproute2 failed to build (build variant: tcfull).
make[1]: *** [package/Makefile:119: package/network/utils/iproute2/compile] Error 1
make[1]: Leaving directory '/build/openwrt'
make: *** [/build/openwrt/include/toplevel.mk:230: package/network/utils/iproute2/compile] Error 2
```
The failure happens 100% of the time and it is easy to reproduce. I have experienced it at least when building the same config file for the Netgear R7800 (arm) and Linksys E8450 (aarch64).

Note that the error can be overcome if during the first try the make command is called with a high number of threads make -j16. Then, after the failure, any number of threads will work. This behavior suggests that maybe a missing dependency in IPROUTE2-TCFULL that is luckily compiled by another thread saves the day.

# ENVIRONMENT:
`.config` file:
```ini
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_ipq806x_generic_DEVICE_netgear_r7800=y
CONFIG_DEVEL=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUILD_LOG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVEHISTORY=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVE_ON_EXIT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_LESS_FLAGS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_LESS_REGEXP=y
CONFIG_BUSYBOX_CONFIG_FEATURE_LESS_WINCH=y
CONFIG_BUSYBOX_DEFAULT_STAT=y
# CONFIG_BUSYBOX_DEFAULT_FEATURE_STAT_FORMAT is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_STAT_FILESYSTEM is not set
CONFIG_DEBUG=y
CONFIG_DROPBEAR_ECC=y
CONFIG_IMAGEOPT=y
CONFIG_INCLUDE_CONFIG=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
# CONFIG_LUCI_JSMIN is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
[... 188 lines cut ...]
CONFIG_PACKAGE_luci-lib-ipkg=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_nano-plus=y
```

## STEPS TO REPRODUCE

1. Prepare an OpenWrt instance:
```bash
git clone -b master https://github.com/openwrt/openwrt.git openwrt
cd openwrt
```
2. copy the `.config` file listed above in the openwrt folder.
3. Try to build the image with:
```bash
make defconfig
make
```
To speedup the failure, this alternative process would bring the same issue faster:
```bash
make defconfig
make -j16 toolchain/install
make -j16 target/compile
make -j1 V=sc package/network/utils/iproute2/compile
```

okibcn, 2022-03-20:
I think I've found the issue. This change in the [Makefile ](https://github.com/openwrt/openwrt/blob/master/package/network/utils/iproute2/Makefile)of iproute2 seems to solve the problem:
```diff
diff --git a/package/network/utils/iproute2/Makefile b/package/network/utils/iproute2/Makefile
index 55c00a0d6d..465f107575 100644
--- a/package/network/utils/iproute2/Makefile
+++ b/package/network/utils/iproute2/Makefile
@@ -75,7 +75,7 @@ $(call Package/iproute2/Default)
   VARIANT:=tcfull
   PROVIDES:=tc
   ALTERNATIVES:=400:/sbin/tc:/usr/libexec/tc-full
-  DEPENDS:=+kmod-sched-core +(PACKAGE_devlink||PACKAGE_rdma):libmnl +libbpf +libxtables +tc-mod-iptables
+  DEPENDS:=+kmod-sched-core +libbpf +libxtables +tc-mod-iptables +(PACKAGE_devlink||PACKAGE_rdma):libmnl
 endef
 
 define Package/tc-mod-iptables
```

okibcn, 2022-03-21:
The problem can be fixed with this patch. I have confirmed it works in three platforms: mips, arm and aarch64.

```diff
diff --git a/package/network/utils/iproute2/Makefile b/package/network/utils/iproute2/Makefile
index 55c00a0d6d..465f107575 100644
--- a/package/network/utils/iproute2/Makefile
+++ b/package/network/utils/iproute2/Makefile
@@ -75,7 +75,7 @@ $(call Package/iproute2/Default)
   VARIANT:=tcfull
   PROVIDES:=tc
   ALTERNATIVES:=400:/sbin/tc:/usr/libexec/tc-full
-  DEPENDS:=+kmod-sched-core +(PACKAGE_devlink||PACKAGE_rdma):libmnl +libbpf +libxtables +tc-mod-iptables
+  DEPENDS:=+kmod-sched-core +libbpf +libxtables +tc-mod-iptables +(PACKAGE_devlink||PACKAGE_rdma):libmnl
 endef
 
 define Package/tc-mod-iptables
```

Anyone willing to commit it?

okibcn, 2022-03-21:
So fat, the only workaround to this issue is to call make twice, or including libbpf package even though it is not used.

closed 2022-09-10 by jow-

damnms, 2022-10-15:
got the same error today on 22.03.01

commit fa468d4b (2022-09-10): iproute2: add missing libbpf dependency

This patch adds libbpf to the dependencies of tc-mod-iptables.

The package tc-mod-iptables is missing libbpf as a dependency,
which leads to the build failure described in bug #9491

    LIBBPF_FORCE=on set, but couldn't find a usable libbpf

The build dependency is already automatically added because some other
packages from iproute2 depend on libbpf, but bpftools has multiple build
variants. With multiple build variants none gets build by default and
the build system will not build bpftools before iproute2.

Fixes: #9491
commit fea7478f (2023-04-02): iproute2: add missing libbpf dependency

This patch adds libbpf to the dependencies of tc-mod-iptables.

The package tc-mod-iptables is missing libbpf as a dependency,
which leads to the build failure described in bug #9491

    LIBBPF_FORCE=on set, but couldn't find a usable libbpf

The build dependency is already automatically added because some other
packages from iproute2 depend on libbpf, but bpftools has multiple build
variants. With multiple build variants none gets build by default and
the build system will not build bpftools before iproute2.

Fixes: #9491

(cherry picked from commit f [... cut]