openwrt/openwrt #9492: initramfs boot: ubusd running but non-functional: "procd: Connection to ubus failed"

Lanchon, 2022-03-17:
hi,

i came across this issue while porting to a new device, but there seems to be many similar reported cases floating in the tubes of the interwebs.

the `ubusd` process is running, but unreachable by `ubus`. restarting the daemon does not help, but manually starting an instance lets the boot process continue normally. this seems to be a user/permissions/jail issue, and i'm starting to think it might an issue of the platform rather than my port.

full details here: https://forum.openwrt.org/t/arris-tr4400-v2-procd-connection-to-ubus-failed/122922

any help welcome, and will be repaid in copious cat memes.

thanks!

Lanchon, 2022-03-20:
NOTE: this issue is not related to [this other issue](https://github.com/openwrt/openwrt/issues/9511).

Lanchon, 2022-03-22:
## root cause

strace runs reveal that ubusd hangs while trying to read /etc/passwd and /etc/group, and user sbus cannot read those files.
```
root@(none):/# ls -l /etc/passwd /etc/group 
-rw-------    1 root     root           211 Mar 19  2022 /etc/group
-rw-------    1 root     root           374 Mar 19  2022 /etc/passwd
```
a simple `chmod +r /etc/passwd /etc/group` is enough to make the router start normally.

how can we fix this?

note: this was tested on an initramfs build.

Lanchon, 2022-03-22:
@dangowrt this issue is unrelated to #9511. #9511 is caused -in my setup- by building on ubuntu and is solved by building on debian. whereas this issue is present when i build on debian and is solved by changing permissions of the aforementioned files.

please note that this issue happens on initramfs builds, as i am bringing up a new device. i did not test squashfs builds, nor i would until the initramfs builds work as intended. (on the contrary, issue #9511 i encounter on initramfs builds but several others report it on squashfs builds too.)

Lanchon, 2022-03-22:
@dangowrt i build these images on a LMDE VM using ext4. the VM runs on virtualbox 6.1.

hibby50, 2022-03-22:
This is happening to me on my x86 build on master. Moving my comments over from #9511 below

@dangowrt I have an x86 build already compiled with this issue (or https://github.com/openwrt/openwrt/issues/9492 frankly I'm confused as to what the difference is). The issue repros booting on my kvm hypervisor. Though the image does include quite a few additional packages and tweaks. This was built 2 days ago from master in the openwrt builder docker container. If you'd like I can try to compile one after a make distclean tomorrow.

I am running suse leap 15.3 on my server but using the openwrtorg docker container based on debian:latest

It was too large to upload here https://drive.google.com/file/d/165Ld0-GU5rXyz3pgmwkyzD7qK_wMGf4J/view?usp=sharing

___

@dangowrt Here is an x86 image I just built from master with a fresh, unedited .config file (aside from target) exhibiting the issue. It was built in https://hub.docker.com/r/openwrtorg/imagebuilder

[openwrt-x86-64-generic-squashfs-combined.img.gz](https://github.com/openwrt/openwrt/files/8324267/openwrt-x86-64-generic-squashfs-combined.img.gz)

[config.txt](https://github.com/openwrt/openwrt/files/8324281/config.txt)

```
    sudo apt install python3-distutils
    make clean
    rm .config*
    make menuconfig
    make download
    make -j 20
```

___

>all the busybox symlinks which were corrupted in the previous images you shared are fine now.

I had coreutils in the first image, when I booted it in safe mode I could see symlinks from /bin to /usr/libexec. What tool are you using to explore the image? I did a mount on the rootfs and everything looked good. I'm guessing you're looking at the initramfs but I don't know how to extract it.

>Which filesystem type are you using on the machine you use to build OpenWrt? And is the build host running in WSL or a VM or Linux on bare-metal?

Linux bare metal. Filesystem where the tree lives is ZFS. FWIW I've compiled many successful images both in the container and on the bare metal with this setup

Here is the new image and the commands I used. I did it outside the container directly on my host OS (SUSE Leap 15.3). The only thing i changed in menu config during the make clean was the target.

```
10074  git pull
10075  make distclean
10076  make clean
10077  ./scripts/feeds update -a
10078  ./scripts/feeds install -a
10079  ./scripts/feeds update -a
10080  ./scripts/feeds install -a
10081  make download
10082  make -j 20
```
[openwrt-x86-64-generic-squashfs-combined.img.gz](https://github.com/openwrt/openwrt/files/8325093/openwrt-x86-64-generic-squashfs-combined.img.gz)
[config.txt](https://github.com/openwrt/openwrt/files/8325102/config.txt)

___

dangowrt, 2022-03-22:
It looks like the rootfs has permissions all wrong, basically all *group* and *others* bits are missing for most (but not all) files. OpenWrt's buildroot should set `umask` appropriately, but apparently there is still something wrong here. Maybe aliases defined in shell, replacing `install` or giving it additional parameters?

Lanchon, 2022-03-22:
FYI, my LMDE VM on which i build and repro this issue is a clean LMDE beta 5 install.

on it:
```
$ umask 
0022
$ umask -S
u=rwx,g=rx,o=rx
$ alias
alias ls='ls --color=auto'
```

Lanchon, 2022-03-22:
good. this issue is blocking me.

is this normal on initramfs builds?
```
root@(none):/# ls -l /etc/passwd /etc/group 
-rw-------    1 root     root           211 Mar 19  2022 /etc/group
-rw-------    1 root     root           374 Mar 19  2022 /etc/passwd
```
if it is, what process is supposed to alter the permissions before ubusd is ran?

dangowrt, 2022-03-22:
No, both files are supposed to be 0644 and not 0600. Also note that in `/bin/` many executables are 0700, which effectively means that only the owner (root) can run them. Seems like maybe host `cpio` is used for Linux build on targets where initramfs is built-into the kernel, and that host `cpio` could drop those permission bits.
Which permissions do you see on the buildhost in `build_dir/target-*/root-*/etc` (and `build_dir/target-*/root-*/bin`)?

hibby50, 2022-03-22:
>when you boot an initramfs image non-failsafe with log level 4 you should get lone, repeated "procd: Connection to ubus failed" messages as depicted [here](https://forum.openwrt.org/t/arris-tr4400-v2-procd-connection-to-ubus-failed/122922).

Yes
>when you boot an initramfs image failsafe and type ls -l /etc/passwd /etc/group you should see permissions restricted to root as depicted https://github.com/openwrt/openwrt/issues/9492#issuecomment-1074710551.

No mine are 0640 and chmodding to 0644 and continuing the boot works. Though at the moment this is a speculative workaround and I don't think precludes these issues from being the same.

>after 2, typing chmod +r /etc/passwd /etc/group and https://github.com/openwrt/openwrt/issues/9493#issue-1172839615 from failsafe, ubusd should now work.

Yes

>note that initramfs run from RAM. they are (typically) not flashed on the router. you cannot use sysupgrade to flash an initramfs image. initramfs on real hardware are typically loaded to RAM by TFTP transfer to the bootloader, then invoked.

This seems irrelevant because the boot process is the same after grub loads vmlinuz kernel into memory and /sbin/init is called

Lanchon, 2022-03-22:
> This seems irrelevant ...

it is relevant in that my /etc/passwd /etc/group files come directly from the build. i have no overlay and my files are rw on tmpfs. you might have a persistent overlay which could confound issues because your /etc/passwd /etc/group files might come from there. IDK when in the boot process the overlay is mounted. so yes, it could be related... or not, IDK.

dangowrt, 2022-03-22:
It's not irrelevant if you are using initramfs or squashfs to boot.
First of all, the filesystems or archives are created by different tools (cpio vs. mksquashfs) which may show different behavior when in comes to permissions in some corner cases.
Second, because when booting initramfs, instead of `/sbin/init` there is `/init` being executed setting of tmpfs and calling `/sbin/init`, leaving a mark in the environment which will lead to procd knowing that it's running off initramfs.
Both could contribute somehow to this issue.

Lanchon, 2022-03-22:
@dangowrt so disregard [this build](https://github.com/openwrt/openwrt/issues/9492#issuecomment-1075350873) then, because it is not initramfs.

regarding you own build and test [here](https://github.com/openwrt/openwrt/issues/9511#issuecomment-1075489340) it is not an initramfs either. this issue 9492 involves initramfs booting. could you try booting your build via initramfs and see if you finally repro?

dangowrt, 2022-03-22:
https://user-images.githubusercontent.com/9948313/159554455-27a2ab11-3b84-4ce8-83c8-9bfaf426774a.mp4

Also initramfs image works and boots perfectly when build on all distributions I've tried for now, including LXDE 5 as you can see here.

Lanchon, 2022-03-22:
> Which permissions do you see on the buildhost in `build_dir/target-*/root-*/etc` (and `build_dir/target-*/root-*/bin`)?

```
rod@lmde-openwrt:~/openwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$ cat /etc/os-release 
PRETTY_NAME="LMDE 5 (elsie)"
NAME="LMDE"
VERSION_ID="5"
VERSION="5 (elsie)"
ID=linuxmint
ID_LIKE=debian
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=elsie
DEBIAN_CODENAME=bullseye
rod@lmde-openwrt:~/openwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$ ls -l root
root-ipq806x/      root.orig-ipq806x/ 
rod@lmde-openwrt:~/openwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$ ls -l root-ipq806x/bin
total 596
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ash -> busybox
-rwxr-xr-x 1 rod rod    205 Mar 19 11:13 board_detect
-rwxr-xr-x 1 rod rod 413713 Mar 19 11:13 busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 cat -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 chgrp -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 chmod -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 chown -> busybox
-rwx------ 1 rod rod  11778 Mar 19 11:13 config_generate
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 cp -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 date -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 dd -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 df -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 dmesg -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 echo -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 egrep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 false -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 fgrep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 fsync -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 grep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 gunzip -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 gzip -> busybox
-rwx------ 1 rod rod   1550 Mar 19 11:13 ipcalc.sh
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 kill -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ln -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 lock -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 login -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ls -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mkdir -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mknod -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mktemp -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mount -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mv -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 netmsg -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 netstat -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 nice -> busybox
-rwxr-xr-x 1 rod rod 132843 Mar 19 11:13 opkg
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 passwd -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 pidof -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ping -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ping6 -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ps -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 pwd -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 rm -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 rmdir -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sed -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sh -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sleep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sync -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 tar -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 touch -> busybox
[... 162 lines cut ...]
drwxr-xr-x  2 rod rod 4096 Mar 19 19:59 ppp
-rwx------  1 rod rod  609 Mar 16 13:50 preinit
-rw-r--r--  1 rod rod 1046 Mar 16 13:50 profile
-rw-------  1 rod rod 2541 Mar 16 13:50 protocols
drwxr-xr-x  2 rod rod 4096 Mar 19 19:59 rc.button
-rwxr-xr-x  1 rod rod 3359 Mar 16 13:50 rc.common
drwxr-xr-x  2 rod rod 4096 Mar 16 13:50 rc.d
-rw-------  1 rod rod  132 Mar 16 13:50 rc.local
lrwxrwxrwx  1 rod rod   16 Mar 19 19:59 resolv.conf -> /tmp/resolv.conf
-rw-------  1 rod rod 3073 Mar 16 13:50 services
-rw-------  1 rod rod  114 Mar 16 13:50 shadow
-rw-------  1 rod rod    9 Mar 16 13:50 shells
-rw-------  1 rod rod 1120 Mar 16 13:50 shinit
drwxr-xr-x  3 rod rod 4096 Mar 19 19:59 ssl
-rw-------  1 rod rod   80 Mar 16 13:50 sysctl.conf
drwxr-xr-x  2 rod rod 4096 Mar 19 19:59 sysctl.d
-rw-r--r--  1 rod rod    0 Jan  7 20:49 syslog.conf
-rw-------  1 rod rod  128 Mar 16 13:50 sysupgrade.conf
lrwxrwxrwx  1 rod rod    7 Mar 19 19:59 TZ -> /tmp/TZ
drwxr-xr-x  2 rod rod 4096 Oct 22 18:51 uci-defaults
-rw-------  1 rod rod   70 Feb 20 06:52 udhcpc.user
drwxr-xr-x  2 rod rod 4096 Feb 20 06:52 udhcpc.user.d
rod@lmde-openwrt:~/openwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$ ls -l root.orig-ipq806x/etc/passwd 
-rw------- 1 rod rod 190 Mar 16 13:50 root.orig-ipq806x/etc/passwd
```

Lanchon, 2022-03-22:
> Seems like maybe host `cpio` is used for Linux build on targets where initramfs is built-into the kernel, and that host `cpio` could drop those permission bits.

FYI
```
rod@lmde-openwrt:~/openwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$ which cpio
/usr/bin/cpio
rod@lmde-openwrt:~/openwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$ cpio --version 
cpio (GNU cpio) 2.13
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Phil Nelson, David MacKenzie, John Oleynick,
and Sergey Poznyakoff.
```

Lanchon, 2022-03-22:
i suppose root.orig- is for initramfs and root- for squashfs.

> Are you using filesystem ACLs? Any MAC (SELinux, AppArmor, ...) or just normal UNIX DAC?

i simply did a clean LMDE 5 beta install on ext4 on virtualbox. i installed LMDE solely to build, only installed prerequisites as per the openwrt page. oh... and i installed virtualbox guest additions. i did not change anything else on the guest. the VM has 3D acceleration enabled.

dangowrt, 2022-03-22:
> i suppose root.orig- is for initramfs and root- for squashfs.

Exactly.

> i simply did a clean LMDE 5 beta install on ext4 on virtualbox. i installed LMDE solely to build, only installed prerequisites as per the openwrt page. oh... and i installed virtualbox guest additions. i did not change anything else on the guest. the VM has 3D acceleration enabled.

Same here, only that only found LMDE 5 release image (didn't find beta version). And I used libvirt with Gnome boxes, no guest additions what-so-ever, only installed `build-essential`, `git`, `libncurses-dev` and `cmake` to make OpenWrt buildsystem happy.

Lanchon, 2022-03-22:
this seems like it is not a low-level, kernel related thing. i don't think the hypervisor or VM settings should be an issue. this seems to be totally userland.

- you could install virtualbox guest additions (and reboot)
- i could update LMDE (it self-updates to the release version, which i'll do now)

anything else crosses your mind?

Lanchon, 2022-03-22:
i was already on latest LMDE.

dangowrt, 2022-03-22:
Can you check if permissions before opkg are also already broken?
Ie.
```
ls -l build_dir/target-*/linux-*/base-files/ipkg-*/base-files/bin
```

Lanchon, 2022-03-22:
> Ah ok, I didn't run any updates after installing LMDE 5 from installation medium. Any other changes made after installation? Like additional packages apart from OpenWrt build-requirements installed?

your medium should be fairly up to date, it has just been released.

> Any other changes made after installation?

no, just the build prep from the openwrt page and guest additions.

- i could start from scratch again... new VM?
- or better, what if you build my platform on my commit [here](https://github.com/Lanchon/openwrt/commits/arris-tr4400-v2)?

Lanchon, 2022-03-22:
> Can you check if permissions before opkg are also already broken?

please hold on. i am currently `make -j9 clean world`

Lanchon, 2022-03-22:
> Can you check if permissions before opkg are also already broken?

no changes after make clean world:
```
rod@lmde-openwrt:~/openwrt/openwrt$ ls -l build_dir/target-*/linux-*/base-files/ipkg-*/base-files/bin
total 20
-rwxr-xr-x 1 rod rod   205 Mar 15 19:17 board_detect
-rwx------ 1 rod rod 11778 Mar 12 21:23 config_generate
-rwx------ 1 rod rod  1550 Mar 12 21:23 ipcalc.sh
rod@lmde-openwrt:~/openwrt/openwrt$ 
rod@lmde-openwrt:~/openwrt/openwrt$ 
rod@lmde-openwrt:~/openwrt/openwrt$ 
rod@lmde-openwrt:~/openwrt/openwrt$ 
rod@lmde-openwrt:~/openwrt/openwrt$ ls -l build_dir/target-*/root*/{etc,bin}
build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/bin:
total 596
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ash -> busybox
-rwxr-xr-x 1 rod rod    205 Mar 19 11:13 board_detect
-rwxr-xr-x 1 rod rod 413713 Mar 19 11:13 busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 cat -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 chgrp -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 chmod -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 chown -> busybox
-rwx------ 1 rod rod  11778 Mar 19 11:13 config_generate
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 cp -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 date -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 dd -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 df -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 dmesg -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 echo -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 egrep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 false -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 fgrep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 fsync -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 grep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 gunzip -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 gzip -> busybox
-rwx------ 1 rod rod   1550 Mar 19 11:13 ipcalc.sh
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 kill -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ln -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 lock -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 login -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ls -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mkdir -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mknod -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mktemp -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mount -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 mv -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 netmsg -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 netstat -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 nice -> busybox
-rwxr-xr-x 1 rod rod 132843 Mar 19 11:13 opkg
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 passwd -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 pidof -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ping -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ping6 -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 ps -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 pwd -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 rm -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 rmdir -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sed -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sh -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sleep -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 sync -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 tar -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 touch -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 traceroute -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 traceroute6 -> busybox
lrwxrwxrwx 1 rod rod      7 Mar 19 11:13 true -> busybox
-rwxr-xr-x 1 rod rod  12300 Mar 19 11:13 ubus
[... 159 lines cut ...]
lrwxrwxrwx  1 rod rod   21 Mar 22 16:37 os-release -> ../usr/lib/os-release
-rw-------  1 rod rod  190 Mar 16 13:50 passwd
drwxr-xr-x  2 rod rod 4096 Mar 22 16:37 ppp
-rwx------  1 rod rod  609 Mar 16 13:50 preinit
-rw-r--r--  1 rod rod 1046 Mar 16 13:50 profile
-rw-------  1 rod rod 2541 Mar 16 13:50 protocols
drwxr-xr-x  2 rod rod 4096 Mar 22 16:37 rc.button
-rwxr-xr-x  1 rod rod 3359 Mar 16 13:50 rc.common
drwxr-xr-x  2 rod rod 4096 Mar 16 13:50 rc.d
-rw-------  1 rod rod  132 Mar 16 13:50 rc.local
lrwxrwxrwx  1 rod rod   16 Mar 22 16:37 resolv.conf -> /tmp/resolv.conf
-rw-------  1 rod rod 3073 Mar 16 13:50 services
-rw-------  1 rod rod  114 Mar 16 13:50 shadow
-rw-------  1 rod rod    9 Mar 16 13:50 shells
-rw-------  1 rod rod 1120 Mar 16 13:50 shinit
drwxr-xr-x  3 rod rod 4096 Mar 22 16:37 ssl
-rw-------  1 rod rod   80 Mar 16 13:50 sysctl.conf
drwxr-xr-x  2 rod rod 4096 Mar 22 16:37 sysctl.d
-rw-r--r--  1 rod rod    0 Jan  7 20:49 syslog.conf
-rw-------  1 rod rod  128 Mar 16 13:50 sysupgrade.conf
lrwxrwxrwx  1 rod rod    7 Mar 22 16:37 TZ -> /tmp/TZ
drwxr-xr-x  2 rod rod 4096 Oct 22 18:51 uci-defaults
-rw-------  1 rod rod   70 Feb 20 06:52 udhcpc.user
drwxr-xr-x  2 rod rod 4096 Feb 20 06:52 udhcpc.user.d
```

Lanchon, 2022-03-22:
> * you could install virtualbox guest additions (and reboot)

> * i could start from scratch again... new VM?
> 
> * or better, what if you build my platform on my commit [here](https://github.com/Lanchon/openwrt/commits/arris-tr4400-v2)? EDIT: since we know where to look you don't need to run it. EDIT: my platform is ipq806x / arris / tr4400-v2
> 
> * or give you ssh access to my VM?

Lanchon, 2022-03-22:
is there a build log you can contrast against yours?

dangowrt, 2022-03-22:
> rod@lmde-openwrt:~/openwrt/openwrt$ ls -l build_dir/target-*/linux-*/base-files/ipkg-*/base-files/bin
> total 20
> -rwxr-xr-x 1 rod rod   205 Mar 15 19:17 board_detect
> -rwx------ 1 rod rod 11778 Mar 12 21:23 config_generate
> -rwx------ 1 rod rod  1550 Mar 12 21:23 ipcalc.sh

How odd. Should be 0755 for all of them.
What you see in `package/base-files/files/bin` as the files are copied there using
```
        $(CP) ./files/* $(1)/
```
while
```
CP:=cp -fpR
```
which sould obvious preserve all attributes. Are you working on a git checkout or did you use another way to populate the buildroot?

> you could install virtualbox guest additions (and reboot)

Where to find (download) them? And will I have to also install and use VirtualBox then? (I avoided that because it means running non-open-source components in the kernel, which is something I strongly dislike doing).

> i could start from scratch again... new VM?

I strongly believe it will work, as it did work for me. But most likely there is still some detail we do different.

> or give you ssh access to my VM?

Sure. Contact me by email pr via openwrt forums and I can send you a public key for that purpose.

Lanchon, 2022-03-22:
> What you see in `package/base-files/files/bin`
```
rod@lmde-openwrt:~/openwrt/openwrt$ ls -l package/base-files/files/bin
total 20
-rwxr-xr-x 1 rod rod   205 Mar 15 19:17 board_detect
-rwx------ 1 rod rod 11778 Mar 12 21:23 config_generate
-rwx------ 1 rod rod  1550 Mar 12 21:23 ipcalc.sh
```

> Are you working on a git checkout

yes

dangowrt, 2022-03-22:
Ok, so your git checkout already has broken permissions.
Here is what it should look like:
```
[daniel@box openwrt]$ ls -l package/base-files/files/bin/
total 15
-rwxr-xr-x 1 daniel daniel   205 Mar 21 01:12 board_detect
-rwxr-xr-x 1 daniel daniel 11778 Mar 21 01:12 config_generate
-rwxr-xr-x 1 daniel daniel  1550 Mar 21 01:12 ipcalc.sh
```
Did you change local git configuration (ie. `~/.gitconfig`) in any way?

Lanchon, 2022-03-22:
> Where to find (download) them?

https://download.virtualbox.org/virtualbox/6.1.32/VBoxGuestAdditions_6.1.32.iso
you mount that on the guest and install.

> And will I have to also install and use VirtualBox then?

no. the additions just won't activate i suppose.

>  (I avoided that because it means running non-open-source components in the kernel, which is something I strongly dislike doing)

FYI: AFAIK, virtualbox is open source. there is a non-free licensed (still OS i think) extension pack that will give guests access to host's USB devices and a few other add-ons. the host kernel module is for sure open source as it is not recompiled after installing the extension pack (but you don't even have to install it). on the guest: the guest additions, IDK the license.

Lanchon, 2022-03-22:
> Did you change local git configuration (ie. ~/.gitconfig) in any way?

it is a checkout from my github:
```
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[remote "origin"]
	#url = git@github.com:Lanchon/openwrt.git
	url = https://github.com/Lanchon/openwrt.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
[gui]
	wmstate = normal
	geometry = 1918x860+0+25 206 212
[branch "openwrt-21.02"]
	remote = origin
	merge = refs/heads/openwrt-21.02
[branch "arris-tr4400-v2"]
	remote = origin
	merge = refs/heads/arris-tr4400-v2
```

Lanchon, 2022-03-22:
wait a minute... i see this line:
```
#url = git@github.com:Lanchon/openwrt.git
```

i don't have my priv key in the VM! i'm thinking i might have tarballed (with preserve) my openwrt checkout from the ubuntu-based machine instead of checking out again.

it the git checkout from ubuntu is flawed, this would have ported over to LMDE.

Lanchon, 2022-03-22:
this REALLY sounds like why you can't repro on LMDE...

sorry to have wasted your time trying :(((

Lanchon, 2022-03-22:
so if git checkouts include permissions, why is my git status clean???

```
rod@lmde-openwrt:~/openwrt/openwrt$ git status
On branch arris-tr4400-v2
Your branch is up to date with 'origin/arris-tr4400-v2'.

nothing to commit, working tree clean
```

Lanchon, 2022-03-22:
```
rod@lmde-openwrt:~$ cd openwrt/
rod@lmde-openwrt:~/openwrt$ mv openwrt openwrt-problem
rod@lmde-openwrt:~/openwrt$ git clone https://github.com/Lanchon/openwrt.git -b arris-tr4400-v2
Cloning into 'openwrt'...
remote: Enumerating objects: 593727, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 593727 (delta 0), reused 0 (delta 0), pack-reused 593726
Receiving objects: 100% (593727/593727), 215.19 MiB | 4.92 MiB/s, done.
Resolving deltas: 100% (401133/401133), done.
rod@lmde-openwrt:~/openwrt$ ls -l openwrt/package/base-files/files/bin/
total 20
-rwxr-xr-x 1 rod rod   205 Mar 22 18:15 board_detect
-rwxr-xr-x 1 rod rod 11778 Mar 22 18:15 config_generate
-rwxr-xr-x 1 rod rod  1550 Mar 22 18:15 ipcalc.sh
rod@lmde-openwrt:~/openwrt$ ls -l openwrt-problem/package/base-files/files/bin/
total 20
-rwxr-xr-x 1 rod rod   205 Mar 15 19:17 board_detect
-rwx------ 1 rod rod 11778 Mar 12 21:23 config_generate
-rwx------ 1 rod rod  1550 Mar 12 21:23 ipcalc.sh
rod@lmde-openwrt:~/openwrt/openwrt$ #goddammit!!!!!
```

Lanchon, 2022-03-22:
BTW, on my ubuntu-based mothership:
```
rod@rod-latitude:~/openwrt/openwrt$ ls -l package/base-files/files/bin/
total 20
-rwx------ 1 rod rod   205 Mar 12 21:23 board_detect
-rwx------ 1 rod rod 11778 Mar 16 21:26 config_generate
-rwx------ 1 rod rod  1550 Mar 12 21:23 ipcalc.sh
rod@rod-latitude:~/openwrt/openwrt$ git status
On branch arris-tr4400-v2
Your branch is up to date with 'origin/arris-tr4400-v2'.

nothing to commit, working tree clean
rod@rod-latitude:~/openwrt/openwrt$ git --version
git version 2.17.1
```

while back on LMDE:
```
rod@lmde-openwrt:~/openwrt$ git --version
git version 2.30.2
```

Lanchon, 2022-03-22:
this is not caused by umask during checkout. this was ran on my problem ubuntu-based box:
```
rod@rod-latitude:~/openwrt/test$ umask
0002
rod@rod-latitude:~/openwrt/test$ git clone https://github.com/Lanchon/openwrt.git -b arris-tr4400-v2
Cloning into 'openwrt'...
remote: Enumerating objects: 593727, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 593727 (delta 0), reused 0 (delta 0), pack-reused 593726
Receiving objects: 100% (593727/593727), 215.13 MiB | 5.46 MiB/s, done.
Resolving deltas: 100% (401111/401111), done.
rod@rod-latitude:~/openwrt/test$ ls -l openwrt/package/base-files/files/bin/
total 20
-rwx------ 1 rod rod   205 Mar 22 18:29 board_detect
-rwx------ 1 rod rod 11778 Mar 22 18:29 config_generate
-rwx------ 1 rod rod  1550 Mar 22 18:29 ipcalc.sh
rod@rod-latitude:~/openwrt/test$ rm -rf openwrt
rod@rod-latitude:~/openwrt/test$ umask 022
rod@rod-latitude:~/openwrt/test$ umask 
0022
rod@rod-latitude:~/openwrt/test$ git clone https://github.com/Lanchon/openwrt.git -b arris-tr4400-v2
Cloning into 'openwrt'...
remote: Enumerating objects: 593727, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 593727 (delta 0), reused 0 (delta 0), pack-reused 593726
Receiving objects: 100% (593727/593727), 215.19 MiB | 5.25 MiB/s, done.
Resolving deltas: 100% (401133/401133), done.
rod@rod-latitude:~/openwrt/test$ ls -l openwrt/package/base-files/files/bin/
total 20
-rwx------ 1 rod rod   205 Mar 22 18:32 board_detect
-rwx------ 1 rod rod 11778 Mar 22 18:32 config_generate
-rwx------ 1 rod rod  1550 Mar 22 18:32 ipcalc.sh
```

so why its git checkout giving me these unexpected permissions?

Lanchon, 2022-03-22:
ok, git **does not** store permissions. that's why my status show clean; because it is. effectively it only stores a single bit: the executable bit.

openwrt depends on specific permissions being set on the tree -which are not managed by git, and this is sort of a bug.

openwrt depends on git checkout setting permissions to 0644, but that is not a function of git, nor git will detect if they are not set as expected. so something must be done to openwrt:
- preferred but difficult: make openwrt build not depend on tree permissions.
- otherwise: 
  - document a way to force git checkout to produce the right 0644 permissions.
  - use a method to fix the permissions (git hook, metastore, etc):
    - https://hyperprog.com/howto/etc-git.html

Lanchon, 2022-03-22:
i still don't know why my checkout on ubuntu comes out with the unexpected permissions even after setting umask.

dangowrt, 2022-03-22:
At least we found the root cause of the issue :100: 
I've seen weird things happening on Ubuntu inside ecryptfs homedirs, so in case you have encrypted homedir switched on, you can't really rely on UNIX file permissions not being force into something very restrictive (I don't remember exactly, can be that this has long been fixed though)

Lanchon, 2022-03-22:
> in case you have encrypted homedir switched on

no i don't. this might be my own doing many years ago. i'm checking the correct umask and acl to use and the post answer in forum and add to wiki

Lanchon, 2022-03-22:
will probably do it later during the night.

Lanchon, 2022-03-24:
root cause found: see #9545
closing.

closed 2022-03-24 by Lanchon

Lanchon, 2022-03-24:
@dangowrt thanks for all your help Daniel, i really appreciate it

Lanchon, 2022-03-25:
FYI, i added a wiki warning here: https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem