openwrt/openwrt #9497: musl 1.2

kskog, 2022-03-18:
We are trying to run a .NET 6 app on OpenWrt. Problem is when we build for `linux-musl-arm` it compiles for musl 1.2.

From the release history of musl, I notice that musl 1.1.24 in the latest version of OpenWrt is quite old (released October 13, 2019).
Are there any plans to upgrade this to 1.2.x in the near future?

Is there any other way we can run this app on the latest version of OpenWrt? By installing some compatibility package?

openwrtfirmwarecostum, 2022-03-18:
Master musl update

kskog, 2022-03-21:
What do you mean by this comment?

PRIMETSS, 2023-04-08:
Just setup the latest stable version of OpenWRT 22.03 and in both latest versions dotnet SDK 7.0.202 & 8.0.100-preview2 has same issue. There are multiple discussion on what this version upgrade was done, affecting other older version of musl distros than they thought.

[https://github.com/dotnet/runtime/issues/69361](https://github.com/dotnet/runtime/issues/69361)
[https://github.com/dotnet/core/pull/7437](https://github.com/dotnet/core/pull/7437)

There has been discussion plan around what to do for some time (over year) with word it was going to be fixed soon, so I have a project in R&D at the moment that assumes this will one day be the case.

Anyone know of any definite signs of progression?

ynezz, 2023-04-08:
> Anyone know of any definite signs of progression?

ynezz, 2023-04-08:
In other words, if you want .NET working in OpenWrt 22.03 release, you need to create a custom build with c0b4303d2e2f4a9e1d4684fd584e6b6548666f0f fix backported.

openwrtfirmwarecustom, 2023-04-08:
@PRIMETSS 

it should be easy for you to make changes in openwrt 22.03.x

PRIMETSS, 2023-04-09:
Have just tried it with the the latest SNAPSHOT today `Firmware Version | OpenWrt SNAPSHOT r22514-c8934099bf / LuCI Master git-23.093.56957-2145121`

And have a working runtime dotnet and SDK for `dotnet-sdk-7.0.202-linux-musl-arm64`

PRIMETSS, 2023-10-09:
Still seems to be an issue in v22.03.5 but have confirmed `--enable-libstdcxx-dual-abi \` has been set in `openwrt-23.05/toolchain/gcc/common.mk`

[image]

But can install and use dotnet on most/all snapshots over last 8 months? From my install notes they seem to use package ICU72, 73+.
Seems ICU71 which is max for 22.03.5 ? Is this the problem?

anonhostpi, 2024-03-29:
OpenWRT 23.05.3 runs on musl 1.2.4:

[image]

hauke, 2024-04-01:
OpenWrt is using musl 1.2 since OpenWrt v22.03.0, see aaec2ad13b18086e2aaf9ca9152827c50e578bc0
OpenWrt is using `--enable-libstdcxx-dual-abi` since OpenWrt v23.05.0, see 3c06a344e9c7c03c49c9153342e68a5390651323

closed 2024-04-01 by hauke