openwrt/openwrt #9498: Openwrt not writing to logfile

Rachel2jcb, 2022-03-18:
**firmware :** r16279-5cc0535800)) #0 SMP Tue Aug 31 22:20:08 2021

Linux version 5.4.143 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 

**device :** pc-engines APU4D4

**Problem :** 
OpenWRT does no longer write to a log file. Neither to the default /var/log/system.log or to my preferred /mnt/sda3/log/system.log.
The problem occurred while testing rsyslog. 

Initially, everything worked but I was not happy with the rsyslog configuration and modified it a couple of times. Suddenly, there was no logging at all.
With rsyslog disabled, I changed the pad and the filename in luci_system_logging and/or in /etc/config/system and restarted (/etc/init.d) log and system to no avail.
I did several reboots and a single poweroff from within a ssh-session, but the problem remains.
/mnt/sda3 is one of the partitions on a mSATA-storage device. OpenWRT is installed on another partition on that same device. There is plenty of space available on both partitions and there is also plenty of RAM available.

M95D, 2022-03-29:
Looks like a configuration problem. By default there is no log file. Can you see the log in Luci web UI? How about logread in console? Is there any logger still running? The default is /sbin/logd.
Try resetting the router to defaults and follow each step again until the problem appears again.

Rachel2jcb, 2022-04-01:
Thank you for your advice.

Logd was running. There were no logs, not in luci and not as a result of 
logread, both with config set to circular or writing to file. Resetting 
the router didn't solve the problem.

I re-installed openwrt with the latest image (21.02.2) and the log 
problem was solved. However, due to incompatibilities between kernel and 
several previously installed packages (kmod-...) I downgraded from 
21.02.2 tot 21.02.0. Everything is OK now.

Thanks and greetings
Jan

buggsdummy, 2024-12-14:
This isn't working completely on current build, I created a FS and tried writing onto there for a short time, but it just randomly stops until I have to remove the setting and then re-add it and save it and then it works. Not sure what the issue is.

geekycow, 2026-01-01:
I'm also seeing this on 24.10.5 on the OpenWRT, although it's very confusing and hard to work out what makes it happen and what stops it