openwrt/openwrt #9501: qmi: read sim status problem

obsy, 2022-03-19:
https://github.com/openwrt/openwrt/commit/81832b38a0833290ad794065f4bde07df271beb1

Reading pin status in this way is wrong, this code return empty values. See debug script:

```
root@OpenWrt:~# uqmi -s -d /dev/cdc-wdm0 --get-pin-statu
{"pin1_status":"disabled","pin1_verify_tries":3,"pin1_unblock_tries":10,"pin2_status":"not_verified","pin2_verify_tries":3,"pin2_unblock_tries":10}
```

```
root@OpenWrt:~# uqmi -s -d /dev/cdc-wdm0  --uim-get-sim-state
{"pin1_status":"disabled","pin1_verify_tries":3,"pin1_unlock_tries":10,"pin2_status":"not_verified","pin2_verify_tries":3,"pin2_unlock_tries":10}
```

```
root@OpenWrt:~# cat ./a.sh 
#!/bin/sh
device=/dev/cdc-wdm0

. /usr/share/libubox/jshn.sh

json_load "$(uqmi -s -d "$device" --get-pin-status)" 2>&1 | grep -q Failed &&
	json_load "$(uqmi -s -d "$device" --uim-get-sim-state)"
json_get_var pin1_status pin1_status
json_get_var pin1_verify_tries pin1_verify_tries

echo "pin1_status: $pin1_status"
echo "pin1_verify_tries: $pin1_verify_tries"
```

```
root@OpenWrt:~#  ./a.sh 
pin1_status: 
pin1_verify_tries: 
```

```
root@OpenWrt:~# sh -x  ./a.sh 
+ device=/dev/cdc-wdm0
+ . /usr/share/libubox/jshn.sh
+ grep -q Failed
+ uqmi -s -d /dev/cdc-wdm0 --get-pin-status
+ json_load '{"pin1_status":"disabled","pin1_verify_tries":3,"pin1_unblock_tries":10,"pin2_status":"not_verified","pin2_verify_tries":3,"pin2_unblock_tries":10}'
+ json_get_var pin1_status pin1_status
+ local '__dest=pin1_status'
+ local __cur
+ _json_get_var __cur JSON_CUR
+ eval '__cur="$JSON_CUR"'
+ __cur=
+ local '__var=_pin1_status'
+ eval 'export -- "pin1_status=${_pin1_status:-}"; [ -n "${_pin1_status+x}${3+x}" ]'
+ export -- 'pin1_status='
+ '[' -n  ]
+ json_get_var pin1_verify_tries pin1_verify_tries
+ local '__dest=pin1_verify_tries'
+ local __cur
+ _json_get_var __cur JSON_CUR
+ eval '__cur="$JSON_CUR"'
+ __cur=
+ local '__var=_pin1_verify_tries'
+ eval 'export -- "pin1_verify_tries=${_pin1_verify_tries:-}"; [ -n "${_pin1_verify_tries+x}${3+x}" ]'
+ export -- 'pin1_verify_tries='
+ '[' -n  ]
+ echo 'pin1_status: '
pin1_status: 
+ echo 'pin1_verify_tries: '
pin1_verify_tries: 
root@OpenWrt:~# 
```

Tested with Quectel EC20

CC: @dangowrt

closed 2022-03-23 by obsy

dangowrt, 2022-03-23:
@obsy I still had this on my to-do list. What was your reason for closing the issue now?

obsy, 2022-03-23:
Because I don't know what the issue is. This code working for Askey modem, working for BroadMobo BM806U and not working for my Quectel EC20. I don't know if there is anything wrong with him. If you can, check it with some qmi modem.

dangowrt, 2022-03-23:
Does the output of `--get-pin-status` and `--uim-get-sim-state` @obsy has provided [earlier](https://github.com/openwrt/openwrt/issues/9501#issue-1174237321) belong to that Quectel EC20 modem?

obsy, 2022-03-23:
Yes, from this EC20.

obsy, 2022-03-24:
See:
```
#  ./a.sh 
{"pin1_status":"disabled","pin1_verify_tries":3,"pin1_unblock_tries":10,"pin2_status":"not_verified","pin2_verify_tries":3,"pin2_unblock_tries":10}
{ }
{ "pin1_status": "disabled", "pin1_verify_tries": 3, "pin1_unblock_tries": 10, "pin2_status": "not_verified", "pin2_verify_tries": 3, "pin2_unblock_tries": 10 }
```

```
# cat a.sh 
#!/bin/sh
device=/dev/cdc-wdm0

. /usr/share/libubox/jshn.sh

O="$(uqmi -s -d "$device" --get-pin-status)" 
echo "$O"
json_load "$O" | grep Failed
json_dump
json_load "$O"
json_dump

exit 0

```

grep Failed is the problem?

obsy, 2022-03-24:
Ok. it works. But looks very bad. Maybe like this:
```
pin_json="$(uqmi -s -d "$device" --get-pin-status)"
pin=$(echo "$pin_json" | jsonfilter -q -e '@.pin1_status')
[ -z "$pin" ] && pin_json="$(uqmi -s -d "$device" --uim-get-sim-state)"
 json_load "$pin_json"
```
Duplicate, but tests if it's really json.

dangowrt, 2022-03-24:
What about
```diff
diff --git a/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh b/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh
index ad577ea317..c2c5fc1eca 100755
--- a/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh
+++ b/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh
@@ -105,10 +105,12 @@ proto_qmi_setup() {
                        }
                }
        else
-               . /usr/share/libubox/jshn.sh
-               json_load "$(uqmi -s -d "$device" --get-pin-status)" 2>&1 | grep -q Failed &&
-                       json_load "$(uqmi -s -d "$device" --uim-get-sim-state)"
+               json_load "$(uqmi -s -d "$device" --get-pin-status)"
                json_get_var pin1_status pin1_status
+               if [ -z "$pin1_status" ]; then
+                       json_load "$(uqmi -s -d "$device" --uim-get-sim-state)"
+                       json_get_var pin1_status pin1_status
+               fi
                json_get_var pin1_verify_tries pin1_verify_tries
 
                case "$pin1_status" in
```
That would avoid duplicate parsing and also avoid wildly checking for the first character of the replied string...

obsy, 2022-03-24:
Ok.

commit d7354297 (2022-03-27): uqmi: fix acquiring PIN status

Evaluating the return value of 'json_load' didn't work in the
intended way resulting in PIN status no longer being read on modems
where --get-pin-status doesn't fail.
Fix this by trying --get-pin-status first and checking if pin1_status
field exists in JSON, and if it doesn't try again with
--uim-get-sim-state.

Fixes: #9501

(cherry picked from commit ee7cb5e885118b78fb5f692d8ed6c93bb7e35853)