openwrt/openwrt #9507: settings lost during sysupgrade with flash chip mx25l12805d (kernel 5.4 to 5.10)

mpratt14, 2022-03-19:
https://github.com/torvalds/linux/commit/02892d405358288e0057cbdfe6476d13ea62fd2f

This new commit in kernel 5.10 causes settings to not be saved during sysupgrade for devices with flash chip is mx25l12805d

blocksize changes from 64k to 4k, but the sysupgrade.tar with the settings is still placed on the next 64k block instead of the next 4k block

effects ramips but not ath79/generic because 4k blocksizes are not enabled in every subtarget

I would make a commit now to handle padding in image building, but that only saves space, and not certain of exactly how many devices affected.

at the very least, a warning in the next release to backup settings, possibly ask all testers for kernel log / whether blocksize changed.

zorun, 2022-07-15:
On which device(s) did you see this issue? Is it with a standard snapshot, 22.03 RC release, or a custom image?

mpratt14, 2022-07-28:
any subtarget with kernel config that has `CONFIG_MTD_SPI_NOR_USE_4K_SECTORS` 

sysupgrade from 21.02 or older to any newer branch with kernel >= 5.10