openwrt/openwrt #9511: boot failure: "procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds"

Lanchon, 2022-03-20:
ubusd cannot start and device is bricked, but failsafe works fine.

### how to reproduce:
build OpenWrt on a ubuntu-based system (tested on: linux mint 19.3 (ubuntu bionic -based)).
others report same issue on arch linux.

### workaround:
build OpenWrt on a debian-based system (tested on: LMDE 5 beta (debian bullseye -based)).

### more info
this happens at least on arm and x86_64 archs. many reports of this issue. other ubusd startup bricking boot issues exist, but this entry specifically refers to the "exit with error code 32512 after 0 seconds" issue. this specific issue can be worked-around by building on debian.

besides me, other had this issue. eg:
https://forum.openwrt.org/t/x86-64-build-of-21-02-doesnt-boot-because-ubus-exits-with-error-code-32512/115222
https://forum.openwrt.org/t/init-routine-goes-into-loop-when-trying-to-execute-ubus-with-custom-image-for-deco-m4r/104339

### bootlogs

same build (r19142-02d8dbf9f2) on ubuntu:
```
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
4
[    8.006343] ipq806x-gmac-dwmac 37200000.ethernet eth1: Link is Down
[    8.010995] procd: Change state 0 -> 1
[    8.011496] procd: - early -
sendto(): Network unreachable
[    8.015303] procd: Watchdog handover: fd=3
[    8.020985] procd: - watchdog -
[    8.024979] procd: Ping
[    8.028037] procd: Opened watchdog with timeout 30s
[    8.030481] procd: Watchdog did not previously reset the system
[    8.077722] procd: {{"ACTION":"add","DEVPATH":"/devices/virtual/mem/full","SUBSYSTEM":"mem","SYNTH_UUID":"0","MAJOR":"1","MINOR":"7","DEVNAME":"full","DEVMODE":"0666","SEQNUM":"963"}}
[    8.077810] procd: Command: makedev
[    8.093056] procd:  /dev/full
[    8.096357] procd:  0666
[    8.099481] procd: Message:
[    8.103286] procd:  ACTION=add
[    8.104606] procd:  DEVPATH=/devices/virtual/mem/full
[    8.107731] procd:  SUBSYSTEM=mem
[    8.113047] procd:  SYNTH_UUID=0
[    8.116184] procd:  MAJOR=1
[    8.119476] procd:  MINOR=7
[    8.122084] procd:  DEVNAME=full
[    8.124772] procd:  DEVMODE=0666
[    8.128244] procd:  SEQNUM=963
>>>>>>>>>>>>>>>> [    8.131705] procd: Finished udevtrigger <<<<<<<<<<<<<<<<<
[    8.134419] procd: {{"ACTION":"add","DEVPATH":"/devices/virtual/mem/urandom","SUBSYSTEM":"mem","SYNTH_UUID":"0","MAJOR":"1","MINOR":"9","DEVNAME":"urandom","DEVMODE":"0666","SEQNUM":"964"}}

...

[   12.261964] procd: Change state 1 -> 2
[   12.264565] procd: - watchdog -
[   12.268136] procd: Ping
[   12.271304] procd: Opened watchdog with timeout 30s
[   12.273599] procd: Watchdog did not previously reset the system
[   12.279042] procd: - ubus -
[   12.284803] procd: Create service ubus
[   12.287594] procd: Start instance ubus::instance1
[   12.292111] procd: Started instance ubus::instance1[214]
[   12.295824] procd: running /etc/init.d/ubus running
[   12.301136] procd: glob failed on /etc/init.d/ubus
[   12.305638] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   12.335568] procd: Connection to ubus failed
[   12.435579] procd: Connection to ubus failed
[   12.639512] procd: Connection to ubus failed
[   13.049273] procd: Connection to ubus failed
[   13.317266] procd: Started instance ubus::instance1[215]
[   13.321406] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   13.859758] procd: Connection to ubus failed
[   14.328161] procd: Started instance ubus::instance1[216]
[   14.332200] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   14.870824] procd: Connection to ubus failed
[   15.344551] procd: Started instance ubus::instance1[217]
[   15.348487] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   15.881199] procd: Connection to ubus failed
[   16.361056] procd: Started instance ubus::instance1[218]
[   16.365052] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   16.891225] procd: Connection to ubus failed
[   17.280200] procd: Ping
[   17.367622] procd: Started instance ubus::instance1[219]
[   17.371696] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   17.901821] procd: Connection to ubus failed
[   18.380701] procd: Started instance ubus::instance1[220]
[   18.384622] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   18.912141] procd: Connection to ubus failed
[   19.397159] procd: Started instance ubus::instance1[221]
[   19.401304] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   19.921878] procd: Connection to ubus failed
[   20.414102] procd: Started instance ubus::instance1[222]
[   20.418012] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   20.932285] procd: Connection to ubus failed
[   21.431665] procd: Started instance ubus::instance1[223]
[   21.435586] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   21.942617] procd: Connection to ubus failed
[   22.288602] procd: Ping
[   22.439883] procd: Started instance ubus::instance1[224]
[   22.443787] procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds
[   22.952172] procd: Connection to ubus failed
```

same build (r19142-02d8dbf9f2) on debian:
```
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
4
[    7.420070] ipq806x-gmac-dwmac 37200000.ethernet eth1: Link is Down
[    7.423941] procd: Change state 0 -> 1
[    7.425204] procd: - early -
[    7.429006] procd: Watchdog handover: fd=3
[    7.432184] procd: - watchdog -
[    7.435963] procd: Ping
[    7.438993] procd: Opened watchdog with timeout 30s
[    7.441554] procd: Watchdog did not previously reset the system
sendto(): Network unreachable
[    7.506060] procd: {{"ACTION":"add","DEVPATH":"/devices/virtual/mem/full","SUBSYSTEM":"mem","SYNTH_UUID":"0","MAJOR":"1","MINOR":"7","DEVNAME":"full","DEVMODE":"0666","SEQNUM":"963"}}
[    7.506156] procd: Command: makedev
[    7.521918] procd:  /dev/full
[    7.524783] procd:  0666
[    7.527817] procd: Message:
[    7.530661] procd:  ACTION=add
[    7.532944] procd:  DEVPATH=/devices/virtual/mem/full
[    7.536074] procd:  SUBSYSTEM=mem
[    7.541253] procd:  SYNTH_UUID=0
[    7.544500] procd:  MAJOR=1
[    7.547793] procd:  MINOR=7
[    7.550466] procd:  DEVNAME=full
[    7.553126] procd:  DEVMODE=0666
[    7.556584] procd:  SEQNUM=963
[    7.560022] procd: {{"ACTION":"add","DEVPATH":"/devices/virtual/mem/urandom","SUBSYSTEM":"mem","SYNTH_UUID":"0","MAJOR":"1","MINOR":"9","DEVNAME":"urandom","DEVMODE":"0666","SEQNUM":"964"}}

...

[   11.695804] procd: Change state 1 -> 2
[   11.698396] procd: - watchdog -
[   11.702127] procd: Ping
[   11.705008] procd: Opened watchdog with timeout 30s
[   11.707433] procd: Watchdog did not previously reset the system
[   11.712961] procd: - ubus -
[   11.718373] procd: Create service ubus
[   11.721618] procd: Start instance ubus::instance1
[   11.725829] procd: Started instance ubus::instance1[213]
[   11.729950] procd: running /etc/init.d/ubus running
[   11.735043] procd: glob failed on /etc/init.d/ubus
[   11.768173] procd: Connection to ubus failed
[   11.870176] procd: Connection to ubus failed
[   12.075125] procd: Connection to ubus failed
[   12.483695] procd: Connection to ubus failed
[   13.303996] procd: Connection to ubus failed
[   14.329226] procd: Connection to ubus failed
[   15.354575] procd: Connection to ubus failed
[   16.379990] procd: Connection to ubus failed
[   16.711325] procd: Ping
[   17.397342] procd: Connection to ubus failed
[   18.422648] procd: Connection to ubus failed
[   19.447797] procd: Connection to ubus failed
[   20.473061] procd: Connection to ubus failed
[   21.498229] procd: Connection to ubus failed
[   21.716274] procd: Ping
[   22.518162] procd: Connection to ubus failed
```

NOTE: the "procd: Connection to ubus failed" messages are an unrelated ubusd boot issue.

Lanchon, 2022-03-20:
### some progress

i built a pair of images on ubuntu and debian to strace ubusd and diff the traces.

surprising result: `/sbin/ubusd` doesn't even get invoked in the ubuntu build, AFAICT.

Lanchon, 2022-03-21:
next is an strace of `procd` on a system built on ubuntu.

it loops around:
```
connect(7, {sa_family=AF_UNIX, sun_path="/var/run/ubus/ubus.sock"}, 110) = -1 ENOENT (No such file or directory)
```

```
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
f
- failsafe -
Generating 1024 bit rsa key, this may take a while...
Public key portion is:
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAAAgQCQuLg9KESCo3mNDd0N2uZ5xXqcUS7IWphDeSf8hDGC8wC8qPZvAsYhUIeoXCuhP9CC/sT1STPp8d5e1pN8+WiRMk5O9FGaph5LXb1V2z+9KiuBdvaulqmwdcPJy1EIc6ORt7NckTriODAGw8AHQQt1eZuqie++8T+/f2DgF56CtQ== root@(none)
Fingerprint: sha1!! 7e:b0:e3:70:7e:98:b1:61:13:18:f2:06:bb:5d:0a:10:95:b8:15:18

BusyBox v1.35.0 (2022-03-19 14:13:31 UTC) built-in shell (ash)

ash: can't access tty; job control turned off
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r19142-02d8dbf9f2
 -----------------------------------------------------
================= FAILSAFE MODE active ================
special commands:
* firstboot	     reset settings to factory defaults
* mount_root	 mount root-partition with config files

after mount_root:
* passwd			 change root's password
* /etc/config		    directory with config files

for more help see:
https://openwrt.org/docs/guide-user/troubleshooting/
- failsafe_and_factory_reset
- root_password_reset
=======================================================

root@(none):/# cd /sbin
root@(none):/sbin# mv procd procd-bin
root@(none):/sbin# cat >procd <<EOF
> #!/bin/sh
> exec strace -o /tmp/procd-strace -- procd-bin "\$@"
> EOF
root@(none):/sbin# chmod +x procd
root@(none):/sbin# 

root@(none):/sbin# touch /tmp/sysupgrade
root@(none):/sbin# lock -u /tmp/.failsafe
root@(none):/sbin# rm /tmp/sysupgrade
root@(none):/sbin# strace: Exit of unknown pid 128 ignored
strace: Exit of unknown pid 234 ignored

root@(none):/sbin# 

root@(none):/sbin# cat /tmp/procd-strace 
execve("/sbin/procd-bin", ["procd-bin"], 0xbe876e8c /* 7 vars */) = 0
set_tls(0xb6f40408)                     = 0
set_tid_address(0xb6f3f804)             = 238
brk(NULL)                               = 0x1cb8000
brk(0x1cba000)                          = 0x1cba000
mmap2(0x1cb8000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1cb8000
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libubox.so.20211120", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
fcntl64(4, F_SETFD, FD_CLOEXEC)         = 0
[... 197 lines cut ...]
rt_sigaction(SIGPIPE, NULL, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6f029e0}, 8) = 0
mmap2(NULL, 69632, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e1c000
socket(AF_UNIX, SOCK_STREAM, 0)         = 7
fcntl64(7, F_GETFD)                     = 0
fcntl64(7, F_SETFD, FD_CLOEXEC)         = 0
connect(7, {sa_family=AF_UNIX, sun_path="/var/run/ubus/ubus.sock"}, 110) = -1 ENOENT (No such file or directory)
close(7)                                = 0
munmap(0xb6e1c000, 69632)               = 0

clock_gettime64(CLOCK_MONOTONIC, {tv_sec=105, tv_nsec=280373280}) = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=105, tv_nsec=280828800}) = 0
epoll_pwait(4, [], 10, 1000, NULL, 8)   = 0
clock_gettime64(CLOCK_MONOTONIC, {tv_sec=106, tv_nsec=296281120}) = 0
rt_sigaction(SIGINT, NULL, {sa_handler=0xb6ea720c, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f029e0}, 8) = 0
rt_sigaction(SIGTERM, NULL, {sa_handler=0xb6ea720c, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f029e0}, 8) = 0
rt_sigaction(SIGCHLD, NULL, {sa_handler=0xb6ea71f4, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6f029e0}, 8) = 0
rt_sigaction(SIGPIPE, NULL, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0xb6f029e0}, 8) = 0
mmap2(NULL, 69632, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e1c000
socket(AF_UNIX, SOCK_STREAM, 0)         = 7
fcntl64(7, F_GETFD)                     = 0
fcntl64(7, F_SETFD, FD_CLOEXEC)         = 0
connect(7, {sa_family=AF_UNIX, sun_path="/var/run/ubus/ubus.sock"}, 110) = -1 ENOENT (No such file or directory)
close(7)                                = 0
munmap(0xb6e1c000, 69632)               = 0
```

ynezz, 2022-03-21:
>  OpenWrt SNAPSHOT, r19142-02d8dbf9f2

`02d8dbf9f2`  doesn't exist in our Git tree, are you able to reproduce it with latest [development snapshots](https://downloads.openwrt.org/snapshots/targets/ipq806x/generic/)?

> this happens at least on arm

ARM in this case is ipq806x target or it happens on some other ARM targets as well?

> this happens on x86_64 archs

Ok, how can one reproduce it on `x86_64`? If one downloads latest [development snapshot](https://downloads.openwrt.org/snapshots/targets/x86/64/) and run it, what do I need to do in order to see that issue?

Lanchon, 2022-03-21:
hi, thanks!

> > OpenWrt SNAPSHOT, r19142-02d8dbf9f2
> 
> `02d8dbf9f2` doesn't exist in our Git tree, are you able to reproduce it with latest [development snapshots](https://downloads.openwrt.org/snapshots/targets/ipq806x/generic/)?

you are right, i'm porting to a new device. here is [02d8dbf9f2](https://github.com/Lanchon/openwrt/commits/02d8dbf9f2d51b8ec503837b752fdfe76b2808ea), it's basically master from 2 days ago. the exact version is not critical according to various reports:
- [master from 2 days ago](https://github.com/openwrt/openwrt/commits/54aab4e7198b07c7323b5aad4cfedef3793ed1ee): affected
- 21.02.1 and 21.02: affected
- 19.07.8: not affected

> 
> > this happens at least on arm
> 
> ARM in this case is ipq806x target or it happens on some other ARM targets as well?
> 

i tested ipq806x. other people tested others, you can search for their reports. but given the variety i've seen, i wouldn't be surprised if all targets were affected.

> > this happens on x86_64 archs
> 
> Ok, how can one reproduce it on `x86_64`? If one downloads latest [development snapshot](https://downloads.openwrt.org/snapshots/targets/x86/64/) and run it, what do I need to do in order to see that issue?

of course not! please re-read the OP. to repro you need to BUILD FROM SOURCES on some specific linux distros (ubuntu bionic -based distro verified by myself, others reported ubuntu focal and arch linux (unspecified version) being affected too). debian-based distros are not affected (verified by me).

your devel snapshots are build on debian, so none of those images are affected.

if not fixed, this should be added to the wiki ASAP. this has been going on for years wasting developer's time, because AFAICT it is not clearly stated in the build instructions that only debian is supported. on the contrary, specific build instructions for ubuntu are provided... which produce non-working builds.

if you want to repro on `x86_64`, look [here](https://forum.openwrt.org/t/x86-64-build-of-21-02-doesnt-boot-because-ubus-exits-with-error-code-32512/115222).

thanks!

Lanchon, 2022-03-21:
i went ahead and added a warning to the wiki:
https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem

Lanchon, 2022-03-21:
BTW, i think the maintainers of `procd` should be brought in the loop.

dangowrt, 2022-03-21:
This looks very much like ubus user in `/etc/passwd` may be missing it's homedir...? Because most likely the `ENOENT` is a result of the homedir not being created by procd in [state.c#l147](https://git.openwrt.org/?p=project/procd.git;a=blob;f=state.c;h=fb81248fd7e7bc7b588a82a5e667424af188cbab;hb=HEAD#l147). Hence please check `/etc/passwd` (which is created during build) as the problem may come from there...

Lanchon, 2022-03-21:
i'm sorry but i'm working on other stuff now, i'm debugging my port.

i think passwd was there, but permissions were set to r/w only for owner (root), so user 'ubus' would not have been able to read it. which does not matter, because -as i said- ubusd is never invoked.

also, there you have procd's strace. it is not accessing passwd at all. who would be accessing it?

i think whoever is maintaining procd should pick up here; anyone can debug an x86 build.

dangowrt, 2022-03-21:
I'm currently co-maintaining procd and I'm asking because I cannot reproduce the issue locally (neither in Ubuntu, nor Archlinux, nor Debian). I vaguely remember someone had similar issues with permissions on the resulting rootfs caused by the build running fakeroot and the user was using WSL, but that's most likely a bug in WSL then.
So in order to allow me to debug the issue, I will need to know more about your build environment.

Lanchon, 2022-03-21:
@dangowrt, here is the rootfs of my ubuntu-ran build so you can check:

[root.squashfs.zip](https://github.com/openwrt/openwrt/files/8319764/root.squashfs.zip)

(for whatever reason, my system is not mounting it. old kernel version?)

Lanchon, 2022-03-21:
my build environment is Linux Mint 19.3, up to date, and stuff installed on it. whatever you need to know about my build environment i can tell you.

Lanchon, 2022-03-21:
> @dangowrt, here is the rootfs of my ubuntu-ran build so you can check:
> 
> [root.squashfs.zip](https://github.com/openwrt/openwrt/files/8319764/root.squashfs.zip)
> 
> (for whatever reason, my system is not mounting it. old kernel version?)

/etc/passwd in my ubuntu-built rootfs has this content:
```
root:x:0:0:root:/root:/bin/ash
daemon:*:1:1:daemon:/var:/bin/false
ftp:*:55:55:ftp:/home/ftp:/bin/false
network:*:101:101:network:/var:/bin/false
nobody:*:65534:65534:nobody:/var:/bin/false
ntp:x:123:123:ntp:/var/run/ntp:/bin/false
dnsmasq:x:453:453:dnsmasq:/var/run/dnsmasq:/bin/false
logd:x:514:514:logd:/var/run/logd:/bin/false
ubus:x:81:81:ubus:/var/run/ubus:/bin/false
```
i could not mount the fs so i can't tell you the owner/group/permissions.

dangowrt, 2022-03-22:
The reason it can't be mounted are possibly incompatible compressor settings. I've extracted your file using `binwalk`.

While `/etc/passwd` and `/etc/group` look all normal, what does look strange is the presence of [`/init`](https://github.com/openwrt/openwrt/blob/master/target/linux/generic/other-files/init) which is intended to only be present in initramfs images. If this then used as PID 1, it changes some behaviors of procd and also remounts the rootfs.
As the binaries in the filesystem are for some ipq806x 32-bit ARM platform I could not easily test-run them and see any further because I don't have any such board.
It'd hence be great if you can provide an image for debugging built for QEMU, ie. either x86, armvirt or malta target **with read/write squashfs rootfs**.

Lanchon, 2022-03-22:
> /init which is intended to only be present in initramfs images

but it is an initramfs! i'm porting to a new device so i'm building initramfs for now.

btw, i just found the root cause of another issue i'm debugging that causes ubusd not to start. i'll post results in that issue's thread and then point you there.

> It'd hence be great if you can provide ...

sure! but i'm completely new to OpenWrt. i see if i can figure out how to setup to build such an image.

dangowrt, 2022-03-22:
> but it is an initramfs! i'm porting to a new device so i'm building initramfs for now.

What you uploaded is `squashfs`, so I was surprised. initramfs is kernel with built-in cpio archive extracted into RAM at boot.

> btw, i just found the root cause of another issue i'm debugging that causes ubusd not to start. i'll post results in that issue's thread and then point you there.

Now I'm curios :drum: 

> sure! but i'm completely new to OpenWrt. i see if i can figure out how to setup to build such an image.

I've installed Mint 19 into a x86/64 VM by now, it's currently building an x86/64 generic OpenWrt image for starters, so I'll see what's going on. Now that you said the problem is with initramfs, I'll try armvirt 32-bit initramfs next.

hibby50, 2022-03-22:
@dangowrt I have an x86 build already compiled with this issue (or #9492 frankly I'm confused as to what the difference is). The issue repros booting on my kvm hypervisor. Though the image does include quite a few additional packages and tweaks. This was built 2 days ago from master in the openwrt builder docker container. If you'd like I can try to compile one after a `make distclean` tomorrow.

I am running suse leap 15.3 on my server but using the openwrtorg docker container based on debian:latest

It was too large to upload here https://drive.google.com/file/d/165Ld0-GU5rXyz3pgmwkyzD7qK_wMGf4J/view?usp=sharing

dangowrt, 2022-03-22:
https://user-images.githubusercontent.com/9948313/159390937-c06998a8-3a14-4735-93ac-3a92d1ef24fc.mp4

Even when using Mint 19.3 I can't reproduce the issue, neither with initramfs nor with squashfs+f2fs overlay image.

In your image things don't work at all, and that's because many things which are usually busybox applets (ie. symlinks to `busybox`) are in your case symlinks to `/dev/null`. Hence it's very logical that init scripts and such don't work if things like cat, `chmod`, `chown`, `cp`, `dd`, `echo`, ... point to `/dev/null`.

Lanchon, 2022-03-22:
> In your image things don't work at all, and that's because many things which are usually busybox applets (ie. symlinks to `busybox`) are in your case symlinks to `/dev/null`. Hence it's very logical that scripts like `/etc/preinit` and such don't work if things like cat, `chmod`, `chown`, `cp`, `dd`, `echo`, ... point to `/dev/null`.

well believe me, i followed the build instructions. i'll try to build an image like you asked ASAP.

> I have an x86 build already compiled with this issue (or https://github.com/openwrt/openwrt/issues/9492 frankly I'm confused as to what the difference is).

no, #9492 is a completely different issue and is not solved by building on debian. i'll post about that one soon.

dangowrt, 2022-03-22:
Can you share `bin/targets/ipq8*/*/config.buildinfo`? That will allow me to build exactly what you are building, because maybe it depends on your build options.

Lanchon, 2022-03-22:
im building a pretty standard build:
```
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_ipq806x_generic_DEVICE_arris_tr4400-v2=y
CONFIG_PACKAGE_strace=y
CONFIG_PACKAGE_trace-cmd=y
CONFIG_PACKAGE_trace-cmd-extra=y
```

Lanchon, 2022-03-22:
> It'd hence be great if you can provide an image for debugging built for QEMU, ie. either x86, armvirt or malta target with read/write squashfs rootfs.

i can't find it. what is the exact target system/subtarget/target profile/target image you want?

Lanchon, 2022-03-22:
> Now I'm curios

here... https://github.com/openwrt/openwrt/issues/9492#issuecomment-1074710551

hibby50, 2022-03-22:
@dangowrt Here is an x86 image I just built from master with a fresh, unedited .config file (aside from target) exhibiting the issue. It was built in https://hub.docker.com/r/openwrtorg/imagebuilder

[openwrt-x86-64-generic-squashfs-combined.img.gz](https://github.com/openwrt/openwrt/files/8324267/openwrt-x86-64-generic-squashfs-combined.img.gz)

[config.txt](https://github.com/openwrt/openwrt/files/8324281/config.txt)

dangowrt, 2022-03-22:
I have never used that Docker container and don't have the available disk space to deal with Docker stuff. I know that it's intended to be used for the ImageBuilder, and not for the buildroot to run.

When analyzing the resulting image you shared, I've noticed a few things:
 * things get killed early because something fires SIGILL
 * again, some very weird things present in the filesystem, like symlink `/etc/mtab -> /proc/5576/mounts`
 * all the busybox symlinks which were corrupted in the previous images you shared are fine now.

Maybe something is odd within your build tree? Did you try `make distclean` to also build the toolchain from scratch?
Also interesting: Which filesystem type are you using on the machine you use to build OpenWrt?

hibby50, 2022-03-22:
>all the busybox symlinks which were corrupted in the previous images you shared are fine now.

I had coreutils in the first image, when I booted it in safe mode I could see symlinks from /bin to /usr/libexec. What tool are you using to explore the image? I did a mount on the rootfs and everything looked good. I'm guessing you're looking at the initramfs but I don't know how to extract it.

>Which filesystem type are you using on the machine you use to build OpenWrt? And is the build host running in WSL or a VM or Linux on bare-metal?

Linux bare metal. Filesystem where the tree lives is ZFS. FWIW I've compiled many successful images both in the container and on the bare metal with this setup

Here is the new image and the commands I used. I did it outside the container directly on my host OS (SUSE Leap 15.3). The only thing i changed in menu config during the make clean was the target.

```
10074  git pull
10075  make distclean
10076  make clean
10077  ./scripts/feeds update -a
10078  ./scripts/feeds install -a
10079  ./scripts/feeds update -a
10080  ./scripts/feeds install -a
10081  make download
10082  make -j 20
```
[openwrt-x86-64-generic-squashfs-combined.img.gz](https://github.com/openwrt/openwrt/files/8325093/openwrt-x86-64-generic-squashfs-combined.img.gz)
[config.txt](https://github.com/openwrt/openwrt/files/8325102/config.txt)

hibby50, 2022-03-22:
> which issue are you talking about? if you don't get "error code 32512 after 0 seconds" then it is not this issue. i think your issue might be https://github.com/openwrt/openwrt/issues/9492. please confirm.

I'm unable to confirm. Booting into failsafe there is not ubus process running. and /etc/passwd has g+r bit set. but I do not see that exit code in the log. So I don't know that what I'm seeing strictly fits into either of the issues you created. I can open a new issue, but it seems silly to have 3 issues that essentially boil down to ubus is not working on the latest builds. We may be looking at the same issue with different failure scenarios.

If others have a different opinion I'm more than happy to continue the discussion in #9492 or open a new issue

Lanchon, 2022-03-22:
@hibby50, thanks! as per the OP:
> this entry specifically refers to the "exit with error code 32512 after 0 seconds" issue

unless you can verify this is your issue, please don't post here as you will just mislead investigations. there are other issues that cause ubusd not to work on boot. they've always been reported clumped together, and that is possibly why they were never fixed. i've created two entries here for two different issues i can reproduce, with two different workarounds i tested to work.

> but I do not see that exit code in the log

boot up your build. at the "Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level" select level 4. if you don't see repeated "procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds" messages, then **this is not your issue** (more details in the OP). if this is not your issue, please cleanup/delete your messages to restrict the conversation to relevant topics. thanks again!

dangowrt, 2022-03-22:
@Lanchon `bin/targets/armvirt/32/*` would be needed to test-run and analyze. Also I would really like to reproduce this problem locally on my system, but I still fail to generate those broken images with my setup, even when using the config seed you have provided on Linux Mint 19.3 (only difference is that my build tree lives on ext4). I'm now trying to setup SuSE Linux in a VM, but I don't see it offering ZFS as an option for the installation rootfs. So I have to install some things on SuSE to support ZFS in additional volume, if I understand correctly...

@hibby50 Permissions of `/etc/passwd` itself don't matter, it's procd running root user resolving the UID and home directory using C standard library tools. And even if that would somehow fail, ubus is then just run as `root` user instead (which doesn't happen)

dangowrt, 2022-03-22:
@Lanchon Ok, I'll go ahead and try on btrfs first then, also never did that before ;)

dangowrt, 2022-03-22:
Tried building on both Linux Mint 19 and OpenSUSE LEAP 15.3, freshly installed into a VM, tried both, xfs and btrfs as filesystem and still can't reproduce the issues.

Lanchon, 2022-03-22:
ok :(

1. this issue i sidestep by building on a debian VM. i built an armvirt image for you on the bare metal linux mint/btrfs box i use to repro this issue. did you try it? does it repro? on this bare metal box:
```
$ alias
$ umask
0002
$ umask -S
u=rwx,g=rwx,o=rx
```

2. anything else regarding this issue i can help you with? otherwise i will go ahead and remove the warning [here](https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem), because you cannot repro.

dangowrt, 2022-03-22:
I'm pretty sure that you didn't just make it up and there is a real issue hiding somewhere in front of me. LDME 5 Cinnamon VM finshed installing and is now building OpenWrt. I'm hoping to encounter this problem now in this way.

Lanchon, 2022-03-22:
you won't find *this* issue there, but hopefully you'll find #9492

dangowrt, 2022-03-22:
So how would you try to reproduce this one? I've now tried Linux Mint, SuSE, LDME (and Archlinux and Debian, which I'm using myself every day to build), building on ext4, xfs and btrfs filesystem, encountering neither #9511 nor #9492.

dangowrt, 2022-03-22:
https://user-images.githubusercontent.com/9948313/159551752-db3d851c-60e6-4445-a0c9-1e4c00113245.mp4

Fresh build on newly installed LMDE 5 VM, also works without any issues.

Lanchon, 2022-03-22:
> Fresh build on newly installed LMDE 5 VM, also works without any issues.

ok this actually applies to #9492.

regarding this issue, did you try my armvirt build? does it repro?

dangowrt, 2022-03-22:
> regarding this issue, did you try my armvirt build? does it repro?
The filesystem of this build is broken, permissions are all borked. I can see that, but I cannot reproduce how you managed to end up with that. My armvirt/32 builds work perfectly, no matter which OS I'm using to build them, also independent from the filesystem type (ok, I didn't test ZFS because installing it turned out to be impossible in any of the OS recommended by @Lanchon and @hibby50 )

Lanchon, 2022-03-22:
> The filesystem of this build is broken

i assume you are talking about my build [here](https://github.com/openwrt/openwrt/issues/9511#issuecomment-1075352190)

ok then. my ubuntu-based build system doesn't work and we don't know why. i'm out of ideas. i will remove the wiki warning now.

for now, can't fix.

lets move on entirely to LMDE and the other issue... #9492

Lanchon, 2022-03-24:
root cause found: see #9545
closing.

closed 2022-03-24 by Lanchon

Its-Just-Nans, 2025-08-25:
Note for future reader

On my side, the issue was that ubusd wasn't working because `/var/run` wasn't existing

So I added

```sh
mkdir -p /var/run
```

Just before the ubusd start