openwrt/openwrt #9512: snapshot dnsmasq-full 2.86-11 failed to start on NanoPi-R4S when /etc/dnsmasq.conf has conf-dir

UMU618, 2022-03-20:
I am sure /etc/dnsmasq.d exists, I did this successfully with R2S/release 21.02.2.
I tried [dnsmasq-full_2.85-8_aarch64_generic.ipk](https://mirrors.ustc.edu.cn/openwrt/releases/21.02.2/packages/aarch64_generic/base/dnsmasq-full_2.85-8_aarch64_generic.ipk), failed again.

Log:

```
Sun Mar 20 11:49:19 2022 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Sun Mar 20 11:49:19 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Mar 20 11:49:24 2022 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Sun Mar 20 11:49:24 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Mar 20 11:49:29 2022 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Sun Mar 20 11:49:29 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Mar 20 11:49:35 2022 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Sun Mar 20 11:49:35 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Mar 20 11:49:40 2022 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Sun Mar 20 11:49:40 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Mar 20 11:49:45 2022 daemon.crit dnsmasq[1]: cannot access directory /etc/dnsmasq.d: No such file or directory
Sun Mar 20 11:49:45 2022 daemon.crit dnsmasq[1]: FAILED to start up
Sun Mar 20 11:49:45 2022 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
```

UMU618, 2022-03-20:
[image]

imebeh, 2022-07-29:
remove `conf-dir=/etc/dnsmasq.d` from `/etc/dnsmasq.conf`
add `option confdir '/etc/dnsmasq.d'` in `/etc/config/dhcp` under `config dnsmasq`

tested works