openwrt/openwrt #9514: Fast Ethernet devices connected to a Gigabit Ethernet switch are limiting upload speeds of Gigabit Ethernet devices to 100 Mbit/s

Matthaiks, 2022-03-20:
Netgear R6220 with OpenWrt SNAPSHOT (r19158-1eb5d50fd3) gets ~940/940 Mbit/s only if there's no slower devices connected to its switch. When a Fast Ethernet device is connected, upload speeds drop to 100 Mbit/s for all devices, even for those that can handle Gigabit Ethernet. Download speeds are not affected.

Vladdrako, 2022-03-20:
Duplicate of https://github.com/openwrt/openwrt/issues/9420

rossiniscarface, 2022-03-20:
Same Problem for me on an Netgear R6260 with OpenWrt SNAPSHOT r19090-34fd5e325a. 
Even worse for me, as a Client PC switches to 10Mbps if it goes to sleep. 
Wan Upload drops to 10Mbps if the pc goes into sleep mode and goes up to regular 50Mbps if it wakes up and switches to 1Gbps

Ansuel, 2022-03-20:
> as a Client PC switches to 10Mbps if it goes to sleep.
That is really how pc works... not a bug honestly.

rossiniscarface, 2022-03-20:
Of course its no bug that a pc goes to 10Mbps Link speed if the pc goes to sleep. But its a bug that for all other devices the wan speed drops to 10Mbps.

sppmasterspp, 2022-03-20:
If this bug is similar to this one https://github.com/openwrt/openwrt/issues/9202
Can any of you try the WAN down/up speeds as described here https://github.com/openwrt/openwrt/issues/9202#issuecomment-1070960720

rossiniscarface, 2022-03-20:
For me it only affects Upload Speed. Never Download. So this is different to the #9202 

Lan Client A connected to Router with 1G
Lan Client B connected to Router with 1G

on both Clients speedtest.net results in 200Mb down/50Mb up
on Router itself speedtest-netperf.sh -H netperf-eu.bufferbloat.net results in 200Mb down/50Mb up
-----
Lan Client B goes into sleep mode and in kernel log it states: lan2: Link is Up - 10Mbps/Full - flow control off

Resulting in
Lan Client A still connected to Router with 1G
on Client A speedtest.net results in 200Mb down/9.5Mb up
on Router itself speedtest-netperf.sh -H netperf-eu.bufferbloat.net still results in 200Mb down/50Mb up.
---
As soon as Client B returns from sleep mode and kernel log states: lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Client A speedtest results in 200Mb down/50Mb Up

sppmasterspp, 2022-03-20:
@rossiniscarface I agree that you're limited in upload speed only and that is a lot better than to have WAN download speed affected or even both.
But maybe the cause of this negative effect is at least of similar nature.
I don't even notice a problem with WAN down/up speeds normally. But once there is LAN traffic between the LAN clients (and one of them is 100Mbps device) I see this massive drop in WAN performance.

justinkb, 2022-03-21:
vladdrako already pointed out this is a dupe of Duplicate of #9420

Matthaiks, 2022-03-23:
Closing in favor of https://github.com/openwrt/openwrt/issues/9420.

closed 2022-03-23 by Matthaiks