openwrt/openwrt #9517: [ OpenWRT 21.02.2 on WRT3200ACM V1] All Wi-Fi interfaces get deactivated when adding a related LED configuration.

gtrevi, 2022-03-20:
I've setup OpenWRT (`OpenWrt 21.02.2 r16495-bf0c965af0 / LuCI openwrt-21.02 branch git-22.046.85957-59c3392`) on my WRT3200ACM V1, and all perfectly worked until I added LED configurations for the Wi-Fi interfaces: as soon as I saved and rebooted, all the Wi-Fi interfaces went down, and could revive them no matter what I did (LAN and internet connectivity through WAN were still working).

Thanks to @anomeome in the OpenWRT forum, an old discussion on LEDs causing issues came up, so I deleted the LED configurations for the Wi-Fi interfaces and that fixed the issue!

Therefore, I believe this may be a bug, and the easy repro is to just add the following LED configurations, reboot and observe the Wi-Fi radios go off:

```
config led
	option name 'WLAN 2.4GHz'
	option sysfs 'rango:white:wlan_2g'
	option trigger 'netdev'
	option dev 'wlan1'
	list mode 'tx'
	list mode 'rx'

config led
	option name 'WLAN 5GHz'
	option sysfs 'rango:white:wlan_5g'
	option trigger 'netdev'
	option dev 'wlan0'
	list mode 'tx'
	list mode 'rx'
```

cybdn, 2022-12-20:
I can confirm the bug is still on 22.03.

Tried to create a led actions on 'rango:white:wlan_5g' broke my wifi.

Linksys WRT3200ACM
Firmware Version | OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd
Kernel Version | 5.10.146

cybdn, 2022-12-20:
I tested some more. It doesn't bug if i set that led to hearbeat. If set to heartbeat, it actually does it.
But if i use network activity, the wifi will drops.

mikysal78, 2026-10-23:
I think I found the cause: these are not LEDs. On the WRT3200ACM, `rango:white:wlan_2g` (gpio1 13, mpp45) and `rango:white:wlan_5g` (gpio1 14, mpp46) are the PERST# lines of the two wifi cards. "LED on" drives the line low and holds the card in reset.

How I checked it, on OpenWrt 25.12.6:

- With `wlan_2g` set to the `netdev` trigger on the 2.4GHz AP interface, phy1 never received a frame (rx_packets 0) and logged `buffer is NULL for tx done ring` / `adapter does not exist` about 17 times per second, until `can't get hw specs`.
- With no LED trigger, I unbound `leds-gpio` and drove gpio 45 low by hand: the 2.4GHz card (02:00.0, behind `pcie@2,0`) disappears from the bus and its config space reads `ffffffff`. Releasing the line brings the device back, but mwlwifi keeps logging `adapter does not exist` until reboot. gpio 46 does the same to the 5GHz card (01:00.0, `pcie@1,0`).
- The radio firmware never drives these lines, even while transmitting.

A heartbeat trigger only keeps the line low for a few milliseconds, which is probably why it looked harmless.

Fix: drop the two LEDs from `armada-385-linksys-rango.dts` and describe the lines as `reset-gpios` of `pcie1` and `pcie2`. Kernel patch: https://lore.kernel.org/r/20261023175302.2102555-1-mikysal78@gmail.com. OpenWrt PR: https://github.com/openwrt/openwrt/pull/26138.

closed 2026-10-24 by openwrt-bot