openwrt/openwrt #9523: firewall4: fw4 -q zone lan cannot get correct ifname

ptpt52, 2022-03-21:
/etc/config/network
```
config interface 'lan'
        option proto 'none'
        option device 'ppp+'
```
```
fw4 -q zone lan
```
this could not get the correct `ppp+`
and the nft firewall could not be set correctly

jow-, 2022-03-21:
And what would be the expected result? Does the same configuration work with fw3? If yes, what is the result there?

ptpt52, 2022-03-21:
hi, it should output `ppp+` on fw3

ptpt52, 2022-03-21:
but now I know I could do this
change config in firewall
```
config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        option mtu_fix '1'
        list network 'lan'
        list device 'ppp+'
```