openwrt/openwrt #9524: [OpenWRT 21.02.2 on WRT3200ACM V1] Wireless network SSID sometimes not discoverable on Windows 10 and IOS.

aryatavakoli, 2022-03-21:
After upgrading from 21.02.1 to 21.02.2, sometimes Windows 10 and IOS devices disassociate from the wireless network and the SSID of the wireless network is no longer discoverable even after rebooting the clients. Router needs to be rebooted to resolve issue temporarily. Problem does not happen with android or Linux clients. Problem does not occur on OpenWRT 19.07.9 or OpenWRT 21.02.1.

5Ghz Marvell 88W8964 Config:
AC Mode. Channel Auto. 80 Mhz
WPA2-CCMP (AES) auth
KRACK Counter Measures enabled
Everything else is default.

2.4Ghz Marvell 88W8964 config:
disabled

Marvell 88W8887 Config:
disabled

There is no indication of any problem in the kernel log.

Hardware:
Linksys WRT3200ACM
ARMv7 Processor rev 1 (v7l)

Software:
21.02.1 (r16325-88151b8303) upgraded to 21.02.2 (r16495-bf0c965af0)

aryatavakoli, 2022-03-21:
This fixes the issue.

Forum post with solution: https://forum.openwrt.org/t/after-8h-from-first-setup-all-wi-fi-interfaces-marvell-88w8964-deactivated/123192/17

Incredibly bizarre bug.

gtrevi, 2022-03-26:
Indeed, may be a related issue (#9517).

aryatavakoli, 2022-03-26:
I really wish Marvell would open source the firmware for the WRT3200ACM. If wireless wasn't such a pain point, this would be such an amazing router. This is a dumb question,but has anyone reached out to their reps and tried asking politely?

aryatavakoli, 2022-04-22:
The latest 21.02.3 update resolves all wireless issues- at least for me. This OpenWRT version is incredibly stable. I've finally upgraded from 19.07.9.

closed 2022-04-22 by aryatavakoli