openwrt/openwrt #9531: MT7622 hardware flow offload not work in pppoe

MeIsReallyBa, 2022-03-22:
I'm using latest master source Code  with redmi ax6s(mt7622+mt7915). I find that hw flow offload works in dhcp environment but failed in pppoe. I add a printk like this: 

https://github.com/MeIsReallyBa/openwrt/blob/5c22d69606110ce8f23b00e54dc84ebeecdc7bbd/target/linux/generic/hack-5.10/650-netfilter-add-xt_FLOWOFFLOAD-target.patch#L360

dmesg keeps shows:test 512

I make a search and found that 512 means ARPHRD_PPP .

Is this a bug or it is caused by my setting ?

MeIsReallyBa, 2022-03-22:
After I deleted several decision statements like lines 357-360, hw offload under pppoe started to work. Really weird.

namidairo, 2022-03-24:
> After I deleted several decision statements like lines 357-360

That's probably less than ideal, instead of just changing the type condition to include ARPHRD_PPP as well. Perhaps something along the lines of:

```C
	if ((dev->flags & IFF_LOOPBACK) ||
	    (dev->type != ARPHRD_ETHER && dev->type != ARPHRD_PPP) ||
 	    dev->addr_len != ETH_ALEN ||
	    !is_valid_ether_addr(dev->dev_addr))
		return;
```

Mentioning @nbd168 to check if this was correct behaviour or not.

MeIsReallyBa, 2022-03-24:
> ```c
> is_valid_ether_addr
> ```

https://elixir.bootlin.com/linux/v5.10.107/source/drivers/net/ppp/ppp_generic.c#L1481

Just adding && dev->type != ARPHRD_PPP isn't enough, dev->addr_len is 0 in ppp instead of ETH_ALEN, and is_valid_ether_addr doesn't seem to pass ppp either. Don't know if anyone else has had the same problem.

nbd168, 2022-03-25:
Does this patch work for you? https://termbin.com/ar1h

MeIsReallyBa, 2022-03-25:
> Does this patch work for you? https://termbin.com/ar1h

I couldn't connect to the internet when I enable hw offload after applying this patch.

MeIsReallyBa, 2022-04-17:
> Does this patch work for you? https://termbin.com/ar1h

After I upgrade to linux 5.15, I found that this patch works. But there is some cpu usage when running speediest compared with dhcp environment.

closed 2022-04-18 by MeIsReallyBa