openwrt/openwrt #9534: realtek/rtl838x: always rebooting during init, after rtl8231_init

casipw, 2022-03-22:
Hi, I've installed OpenWrt 21.02 on a ZyXEL GS1900-8. There is "V1.2" printed on the PCB, probably a version or revision number.

OpenWrt does always reset / reboot at the same place. Output from the serial console:

```
[    0.788872] realtek_gpio_probe probing RTL GPIO
[    0.795014] Probing RTL8231 GPIOs
[    0.798769] rtl8231_init called, MDIO bus ID: 0
II: Copied Preloader data from 0x9fc00000(15064B) to 0x9f000000.
II: Cleared Preloader BSS section at 0x9f003ad8(0B).
```

Note that `II: Copied Preloader data from 0x9fc00000(15064B) to 0x9f000000.` is the first line of the u-boot sequence. The reboot is always after `rtl8231_init called, MDIO bus ID: 0`. This issue occurred on OpenWrt 21.02.0, 21.02.1, 21.02.2 and snapshot. It occurrs with both first and backup partition, and with booting via `tftpboot` + `bootm`. The stock firmware (2.60.3, 2.60.4, 2.70.1) boots fine. It occurred with two different power adapters, so it's probably not a power issue. Both hardware reset buttons are working.

I've read that the reset buttons are controlled through rtl8231. Could it be possible that a MDIO bus scan triggers a reset? Anyway, I'm not sure if it is a hardware issue or a software issue.

Thanks for reading and for your great work. Best regards!

Hurricos, 2022-03-23:
That MDIO bus scan might trigger a reset. I know there's a reset button on the GS1900-24 v1. It also might be that there is a watch-dog that your u-boot is initializing that is not properly noted in the device tree (and thus not properly tickled by Linux).

As ZyXEL is a dual-partition device, there is probably some mechanism which checks some region of memory and sets a variable to increment the number of failed boots in order to trigger booting the fallback partition. Please check `setenv` before booting, after one reboot, and after two reboots.

dangowrt, 2022-03-25:
@bkobl @svanheule any ideas what's going on here?

bkobl, 2022-03-26:
Hmm, the MDIO bus ID might be the problem, we recently did 
https://github.com/openwrt/openwrt/commit/4b31717fb031e6b99f476e79b5e8b3d6a8b0b9ff
The board version in our ToH is the same 1.2, so I don't think the issue is that the GPIOs are now different.
Can we verify the bus ID? In u-boot, could you print out the result of
```
# md.l 0xbb00a09c
# rtk network on
# md.l 0xbb00a09c
```

svanheule, 2022-03-26:
> I've read that the reset buttons are controlled through rtl8231. Could it be possible that a MDIO bus scan triggers a reset?

Those reset buttons are plain GPIOs, the reset function is entirely software controlled on button presses.

4b31717 was applied on master, the problem here is with 21.02. In case MDIO bus ID would be wrong, the GPIO chip can't be configured anyway, so the bootloader's configuration would remain set. In case it is configured, that would mean there may be a missing gpio-restart configuration.

I don't think the "Preloader" log lines are standard. Do these come from one of the built-in test of the bootloader? My device starts by printing out some device info:
```
U-Boot Version: 2.0.0.59413 (Jul 08 2015 - 09:57:29)

CPU:   500MHz
DRAM:  128 MB
FLASH: 16 MB
Model: ZyXEL_GS1900_8
...
```
Do you also experience this problem when you run a master snapshot build?

casipw, 2023-02-03:
Please excuse the late reply. I just found some time to look at the device again.

[This build from @svanheule](https://forum.openwrt.org/t/support-for-rtl838x-based-managed-switches/57875/1998), which adds `status = "disabled";` to the gpio1 node in the device tree, works for me.

Because you asked: The `printenv` output does not change between reboots. Plus the requested output:

```
Hit any key to stop autoboot:  0
RTL838x# printenv
baudrate=115200
boardmodel=ZyXEL_GS1900_8
boardversion=V1.00(AAHH.0)
bootcmd=cst fcTest; boota
bootdelay=1
ethact=rtl8380#0
ethaddr=FC:F5:28:xx:xx:xx
ipaddr=192.168.1.1
netmask=255.255.255.0
serverip=192.168.1.111
stderr=serial
stdin=serial
stdout=serial

Environment size: 282/1020 bytes
RTL838x# md.l 0xbb00a09c
bb00a09c: 0000000c 00019000 0003e800 0007d000    ................
bb00a0ac: 00000000 00000000 00000000 00000000    ................
bb00a0bc: 00000000 00000000 00000000 00000000    ................
bb00a0cc: 00000000 00000000 00000000 00000000    ................
bb00a0dc: 00699041 00000180 00000000 00000000    .i.A............
bb00a0ec: 00000000 00000000 00000000 00000000    ................
bb00a0fc: 00000000 000480a8 00d71eec 00d71eec    ................
bb00a10c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a11c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a12c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a13c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a14c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a15c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a16c: 00d71eec 00d71eec 00d71eec 00ffffff    ................
bb00a17c: 0000ff00 00000000 00000000 00000000    ................
bb00a18c: 05000000 00000000 00220000 00000000    ........."......
RTL838x# rtk network on
Enable network
Force port28 link up 1G
Please wait for PHY init-time ...

RTL838x# md.l 0xbb00a09c
bb00a09c: 0000000c 00019000 0003e800 0007d000    ................
bb00a0ac: 00000000 00000000 00000000 00000000    ................
bb00a0bc: 00000000 00000000 00000000 00000000    ................
bb00a0cc: 00000000 00000000 00000000 00000000    ................
bb00a0dc: 00699041 00000180 00000000 00000000    .i.A............
bb00a0ec: 00000000 00000000 00000000 00000000    ................
bb00a0fc: 00000000 000480a8 00d71eec 00d71eec    ................
bb00a10c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a11c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a12c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a13c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a14c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a15c: 00d71eec 00d71eec 00d71eec 00d71eec    ................
bb00a16c: 00d71eec 00d71eec 0006192f 00ffffff    .........../....
bb00a17c: 0000ff00 00000000 00000000 00000000    ................
bb00a18c: 05000000 00020000 00220000 00000000    ........."......
```

howels, 2024-11-26:
Looks like we need a similar patch to disable GPIO1 on the smaller 8 and 10 port Zyxel switches. Seems that a likely solution may be to set `status = "disabled";` on [the device tree for RTL8380-based Zyxel switches](https://github.com/openwrt/openwrt/blob/1bdb6d84046ec1c47f28c57651358470ed212ec3/target/linux/realtek/dts/rtl8380_zyxel_gs1900.dtsi#L43)

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442

svanheule, 2024-12-26:
Would it be possible to test with the linked pull request?

This new driver should be a bit smarter about how the RTL8231's pins are configured. The DTS changes add pin 5 as a restart handler. I suspect a glitch or bad config on that pin is triggering the resets on boot.

es86de, 2024-12-28:
My GS1900-8 was also affected by the reboot/reset problem. With the linked PR it now boots:
```
[    0.000000] Linux version 6.6.67 (buildbot@0bb5d09468a1) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0+28425-7166f43de9) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Fri Dec 27 15:27:18 2024
[    0.000000] RTL838X model is 83806800
[    0.000000] SoC Type: RTL8380
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019070 (MIPS 4KEc)
[    0.000000] MIPS: machine is Zyxel GS1900-8v1/v2 Switch
[    0.000000] earlycon: ns16550a0 at MMIO 0x18002000 (options '115200n8')
[    0.000000] printk: bootconsole [ns16550a0] enabled
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Using appended Device Tree.
[    0.000000] Primary instruction cache 16kB, VIPT, 4-way, linesize 16 bytes.
[    0.000000] Primary data cache 16kB, 2-way, VIPT, cache aliases, linesize 16 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Kernel command line: earlycon
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 119276K/131072K available (6775K kernel code, 619K rwdata, 1584K rodata, 1256K init, 225K bss, 11796K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] rtl83xx-clk: initialized, CPU 500 MHz, MEM 300 MHz (8 Bit DDR2), LXB 200 MHz
[    0.000000] clocksource: realtek_otto_timer: mask: 0xfffffff max_cycles: 0xfffffff, max_idle_ns: 38225208801 ns
[    0.000006] sched_clock: 28 bits at 3125kHz, resolution 320ns, wraps every 42949672800ns
[    0.009432] Calibrating delay loop... 498.89 BogoMIPS (lpj=2494464)
[    0.136052] pid_max: default: 32768 minimum: 301
[    0.157534] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.165816] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.193618] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=1.
[    0.216412] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.227479] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.239504] pinctrl core: initialized pinctrl subsystem
[    0.251537] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.284776] rtl83xx-clk clock-controller: rate setting enabled, CPU 300-625 MHz, MEM 300-300 MHz, LXB 200-200 MHz, OVERCLOCK AT OWN RISK
[    0.304928] clocksource: Switched to clocksource realtek_otto_timer
[    0.331107] NET: Registered PF_INET protocol family
[    0.337358] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.347515] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.357119] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.365892] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.374476] TCP bind hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.382506] TCP: Hash tables configured (established 1024 bind 1024)
[    0.391536] MPTCP token hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.400266] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.407754] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.417103] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.438810] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[    0.448833] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.455517] jffs2: version 2.2 (NAND) (SUMMARY) (ZLIB) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.476862] pinctrl-single 1b001000.pinmux: 32 pins, size 4
[    0.484216] pinctrl-single 1b00a000.pinmux: 32 pins, size 4
[    0.495131] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
[    0.506302] printk: console [ttyS0] disabled
[    0.512434] 18002000.uart: ttyS0 at MMIO 0x18002000 (irq = 31, base_baud = 12500000) is a 16550A
[    0.522442] printk: console [ttyS0] enabled
[    0.531933] printk: bootconsole [early0] disabled
[    0.591588] printk: bootconsole [ns16550a0] disabled
[    0.709753] brd: module loaded
[    0.729238] spi-nor spi0.0: mx25l12805d (16384 Kbytes)
[    0.735367] 6 fixed-partitions partitions found on MTD device spi0.0
[    0.742568] OF: Bad cell count for /soc/spi@1200/flash@0/partitions
[... 170 lines cut ...]
[   43.636944] switch: port 5(lan5) entered disabled state
[   43.642911] rtl83xx-switch switch@1b000000 lan5: entered allmulticast mode
[   43.651727] rtl83xx-switch switch@1b000000 lan5: entered promiscuous mode
[   43.955442] rtl83xx-switch switch@1b000000 lan6: configuring for phy/internal link mode
[   43.975899] 8021q: adding VLAN 0 to HW filter on device lan6
[... cut ...]
[   45.005427] rtl83xx-switch switch@1b000000 lan8: configuring for phy/internal link mode
[   45.025865] 8021q: adding VLAN 0 to HW filter on device lan8
[   45.145145] switch: port 8(lan8) entered blocking state
[   45.151051] switch: port 8(lan8) entered disabled state
[   45.157144] rtl83xx-switch switch@1b000000 lan8: entered allmulticast mode
[   45.165965] rtl83xx-switch switch@1b000000 lan8: entered promiscuous mode
[   45.249135] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   51.850867] Not offloading default route for now
```
Great work, thank you!

svanheule, 2024-12-28:
Thanks for testing, @es86de !

You do seem to have a few errors in your boot log:
```
[    1.048456] rtl8231-expander rtl83xx-aux-mdio:00: failed to read READY_CODE
[    1.056411] rtl8231-expander: probe of rtl83xx-aux-mdio:00 failed with error -145
...
[    1.158505] Realtek RTL8218B (internal) mdio-bus:08: Expected internal RTL8218B, found PHY-ID 6276
[    1.168669] Realtek RTL8218B (internal): probe of mdio-bus:08 failed with error -1
...
[    1.897918] Realtek RTL8218B (internal) rtl838x slave mii-0:08: Expected internal RTL8218B, found PHY-ID 6276
[    1.909146] Realtek RTL8218B (internal): probe of rtl838x slave mii-0:08 failed with error -1
```

The first one might be solvable with an extra pinmux config for the MDIO pins. I'll have a look to push updated patches tomorrow.

svanheule, 2024-12-29:
I've tested by manully switching the pinmux to GPIO from the bootloader, which gave me the same error as you.

The extra pinmux settings were added to #17387, feel free to retry.

Neustradamus, 2025-01-01:
@es86de: Have you seen latest comment of @svanheule?

es86de, 2025-01-03:
> @es86de: Have you seen latest comment of @svanheule?

Yes, my test results are documented in [#17387](https://github.com/openwrt/openwrt/pull/17387#issuecomment-2566377202).

svanheule, 2025-01-05:
@es86de Could provide us with a dump of the bootloader partition of your device?

My device is a v1, so I suppose yours is too. Can you confirm this is the case?
v1 is the blue PCB, v2 is the green PCB: https://openwrt.org/toh/zyxel/gs1900-8#opening_the_case

es86de, 2025-01-05:
@svanheule I confirm that my device is also v1.

Unfortunately I cannot dump the bootloader at the moment. I sysupgraded to the latest snapshot and now I cannot stop the bootloader from autobooting. (Is it because of #16442?) Openwrt creates a boot loop and the bootloader is locked.

es86de, 2025-01-06:
Here is the dump of my device's u-boot partition: [mtd0.dmp](https://github.com/user-attachments/files/18312372/mtd0.dmp)

svanheule, 2025-01-06:
Thanks! I was able to flash that bootloader on my device and have it boot.

Sadly I was not able to reproduce the reboots you are seeing. I did see the `GPIO_INTF_SEL` bit was set to I2C mode instead of MDIO mode, so I needed to add an extra pincfg-single node to re-enable proper detection of the expander.

es86de, 2025-01-07:
Thanks @svanheule! I just tried booting with 09c2a4815904da129911f86cf536038514846f16, which unfortunately results in the same boot loop again.

Should I also dump my `u-boot-env` and `u-boot-env2` partitions?

svanheule, 2025-01-07:
Sure, there might be some particular bootcmd in there. Although I would expect normally most would be hardcoded in the bootloader itself.

I've unlinked the RTL8231 driver so I can merge that and we can continue with this issue separately.

es86de, 2025-01-07:
I guess nothing special in `u-boot-env`:
```
baudrate=115200
boardmodel=ZyXEL_GS1900_8
boardversion=V1.00(AAHH.0)
bootcmd=cst fcTest; boota
bootdelay=1
ethact=rtl8380#0
ethaddr=FC:F5:28:4F:E7:71
ipaddr=192.168.1.1
netmask=255.255.255.0
serverip=192.168.1.111
stderr=serial
stdin=serial
stdout=serial
```

Same for `u-boot-env2`:
```
boardid=1
resetdefault=0
dualfname0=openwrt-realtek-rtl838x-zyxel_gs1900-8-initramfs-kernel.bin
mac_end=FC:F5:28:4F:E7:79
mac_start=FC:F5:28:4F:E7:71
sn=S130H35002355
dualfname1=runtime-GS1900-8v2-V2.80(AAHH.0).bix
bootpartition=0
```

svanheule, 2025-01-08:
If I boot my device using the `bootcmd` from your dump (which unlike mine includes the `cst fcTest`), I get timeout again on the RTL8231 detection. This time it was due to the RTL8231 being held in reset.

The gs1900.dtsi aux-mdio node has a commented out definition for a MDIO bus reset line. Can you activate that one and see if that works for you?
```diff
--- a/target/linux/realtek/dts/rtl8380_zyxel_gs1900.dtsi
+++ b/target/linux/realtek/dts/rtl8380_zyxel_gs1900.dtsi
@@ -45,11 +45,9 @@
 &mdio_aux {
        status = "okay";
 
-       // Reset GPIO is <&gpio0 1 GPIO_ACTIVE_LOW>
-       // Don't specify the reset info so the mdio subsystem doesn't reset the bus
-       //reset-gpios = <&gpio0 1 GPIO_ACTIVE_LOW>;
-       //reset-delay-us = <1000>;
-       //reset-post-delay-us = <10000>;
+       reset-gpios = <&gpio0 1 GPIO_ACTIVE_LOW>;
+       reset-delay-us = <1000>;
+       reset-post-delay-us = <10000>;
 
        gpio1: expander@0 {
                compatible = "realtek,rtl8231";
```

es86de, 2025-01-09:
Unfortunately, the patch and the bootcmd change made no difference.

What could be the difference between your device, @svanheule, and my device? It doesn't seem to be the software, right?

Here are a few photos of my device:
[image]
[image]
[image]

There is an RTL8231 next to a power management section. What is it used for?

svanheule, 2025-01-09:
The top RTL8231, that you marked with "?", is the one used for the GPIO lines. The LED RTL8231 is strapped for shift register mode, so nothing can be configured for that one.

Your board is in any case labeled slightly different from mine. Where in the second picture yours has
```
LN3774252C1
...
... 1326
P2-2
```
While mine has
```
LN3774252E1
...
... 1529
P3-1
```

This might be due to different part selection, as I can tell mine has different transformer blocks behind the RJ45 ports. I don't see any other differences straight away.

es86de, 2025-01-10:
This is the results of `show board`:
```
GS1900# show board
****************************
 GS1900-8
****************************
============================
 HW Rev.: A1
============================
Device  Pin  Direction  Default  Current
------- ---- ---------- -------- --------
INT     0    OUT        0        0       
INT     1    OUT        1        1       
INT     2    OUT        0        0       
INT     3    OUT        0        0       
INT     8    OUT        1        1       
INT     9    OUT        1        1       
INT     12   IN         0        0       
INT     13   OUT        1        1       
INT     14   OUT        1        1       
INT     15   OUT        1        1       
INT     16   IN         0        0       
INT     17   OUT        1        1       
INT     18   OUT        0        0       
INT     19   IN         0        0       
INT     20   OUT        1        1       
INT     21   IN         0        0       
INT     22   IN         0        0       
INT     23   OUT        1        1       
EXT     0    IN  (IN  ) 0        0       
EXT     1    IN  (IN  ) 0        0       
EXT     2    IN  (IN  ) 0        0       
EXT     3    IN  (IN  ) 0        1       
EXT     4    IN  (IN  ) 0        0       
EXT     5    OUT (OUT ) 1        1       
EXT     6    OUT (OUT ) 1        1       
EXT     7    IN  (IN  ) 0        1       
EXT     8    OUT (OUT ) 1        1       
EXT     13   OUT (OUT ) 0        0       
EXT     14   OUT (OUT ) 0        0       
EXT     15   OUT (OUT ) 0        0       
EXT     16   OUT (OUT ) 0        0       
EXT     17   OUT (IN  ) 0        0       
EXT     18   OUT (IN  ) 0        0       
EXT     21   IN  (IN  ) 0        0       
EXT     22   IN  (IN  ) 0        0       
EXT     23   OUT (OUT ) 0        0       
EXT     24   OUT (OUT ) 0        0       
EXT     25   OUT (OUT ) 0        0       
EXT     26   IN  (IN  ) 0        0       
EXT     27   IN  (IN  ) 0        0       
EXT     28   IN  (IN  ) 0        0       
EXT     29   OUT (OUT ) 0        0       
EXT     30   OUT (OUT ) 0        0       
EXT     31   OUT (OUT ) 0        0       
EXT     32   IN  (IN  ) 0        0       
EXT     33   IN  (IN  ) 0        0       
EXT     34   OUT (OUT ) 1        1       
============================
 Board GPIO
============================
Device  Pin  Direction  Default  Current
------- ---- ---------- -------- --------
INT     0    OUT        0        0       
INT     1    OUT        1        1       
INT     2    OUT        0        0       
INT     3    OUT        0        0       
INT     8    OUT        1        1       
INT     9    OUT        1        1       
INT     12   IN         0        0       
INT     13   OUT        1        1       
INT     14   OUT        1        1       
INT     15   OUT        1        1       
INT     16   IN         0        0       
INT     17   OUT        1        1       
INT     18   OUT        0        0       
INT     19   IN         0        0       
INT     20   OUT        1        1       
INT     21   IN         0        0       
INT     22   IN         0        0       
INT     23   OUT        1        1       
EXT     0    IN  (IN  ) 0        0       
EXT     1    IN  (IN  ) 0        0       
EXT     2    IN  (IN  ) 0        0       
EXT     3    IN  (IN  ) 0        1       
EXT     4    IN  (IN  ) 0        0       
EXT     5    OUT (OUT ) 1        1       
EXT     6    OUT (OUT ) 1        1       
EXT     7    IN  (IN  ) 0        1       
EXT     8    OUT (OUT ) 1        1       
EXT     13   OUT (OUT ) 0        0       
EXT     14   OUT (OUT ) 0        0       
EXT     15   OUT (OUT ) 0        0       
EXT     16   OUT (OUT ) 0        0       
EXT     17   OUT (IN  ) 0        0       
EXT     18   OUT (IN  ) 0        0       
EXT     21   IN  (IN  ) 0        0       
EXT     22   IN  (IN  ) 0        0       
EXT     23   OUT (OUT ) 0        0       
EXT     24   OUT (OUT ) 0        0       
EXT     25   OUT (OUT ) 0        0       
EXT     26   IN  (IN  ) 0        0       
EXT     27   IN  (IN  ) 0        0       
EXT     28   IN  (IN  ) 0        0       
EXT     29   OUT (OUT ) 0        0       
EXT     30   OUT (OUT ) 0        0       
EXT     31   OUT (OUT ) 0        0       
EXT     32   IN  (IN  ) 0        0       
EXT     33   IN  (IN  ) 0        0       
EXT     34   OUT (OUT ) 1        1       

============================
 Board Configuration
============================
====== Port ==================
Type       Usr  Phy     Media       Speed          Duplex   Attr
---------- ---- ------- ----------- -------------- -------- -------
1000M      1    (0) 8   Copper      (A) ALL        Auto     0      
1000M      2    (0) 9   Copper      (A) ALL        Auto     0      
1000M      3    (0) 10  Copper      (A) ALL        Auto     0      
1000M      4    (0) 11  Copper      (A) ALL        Auto     0      
1000M      5    (0) 12  Copper      (A) ALL        Auto     0      
1000M      6    (0) 13  Copper      (A) ALL        Auto     0      
1000M      7    (0) 14  Copper      (A) ALL        Auto     0      
1000M      8    (0) 15  Copper      (A) ALL        Auto     0      

====== Button ================

------------ Reset Button
GPIO: EXT_3 
Timer: 3(sec)  Action: Reboot
Timer: 6(sec)  Action: Restore Factory

====== Led ===================
SYS (REG)
ALARM (REG)

====== Reset =================
Type: GPIO
GPIO: EXT_5 

====== WatchDog ==============
Type: REG
```
It is equivalent to [yours](https://svanheule.net/switches/gs1900-8). I have no idea how to continue at the moment.

svanheule, 2025-01-11:
Have we tried the gpio-restart node without the `open-source` property? Then the driver will actively drive the line inactive (high), instead of requiring a pull-up to  do this. The resistor layout near gpio5 look the same though, so I don't think it likely this will make the difference.

Otherwise you could try to add print statements in the rtl8231 driver, pinctrl framework, gpio-regmap driver,... to find out where things go south. Best to use the approach described here in that case: https://openwrt.org/docs/guide-developer/toolchain/use-patches-with-buildsystem#iteratively_modify_patches_without_cleaning_the_source_tree

svanheule, 2025-01-21:
From the bootloader could you give the output of the following commands to check the SoC revision you have?

```sh
mw.l bb000058 3
mw.l bb0000d8 a00000000
md.l bb0000d0 4
```

Apparently there are issues with the auxiliary MDIO controller on the A revision of the RTL8380, but I think most (if not all) devices are using the B revision.

svanheule, 2025-01-21:
According to the value of bb0000d8, you do indeed have an A revision (bits [20,16], A=1, B=2, etc). In this case the SDK uses a bitbanged MDIO bus to communicate with the RTL8231, so let's try that.

Could you give this branch a try? https://github.com/svanheule/openwrt/tree/debug/gs1900-rtl8380m-rev-a

svanheule, 2025-01-21:
FWIW my device has a C revision of the RTL8380M:
```
RTL838x# mw.l bb000058 3
RTL838x# mw.l bb0000d8 a0000000
RTL838x# md.l bb0000d0 4
bb0000d0: 00000003 83806800 a0036275 00000016    ......h...bu....
```

svanheule, 2025-01-21:
Commit 42e2786 won't work as I forgot to enable the gpio-mdio driver in the build. Running another test build...

es86de, 2025-01-21:
My ancient GS1900-8 boots with 8624c6cab5c87f30d6ee8af3a3b363d38d8b8fcc and the RESET button on the front works now 👍. Here is the complete boot log:
```
II: Copied Preloader data from 0x9fc00000(15064B) to 0x9f000000.
II: Cleared Preloader BSS section at 0x9f003ad8(0B).
II: PLL is set by SW... OK
Setting DTR
INFO: Applying ZQ result directly by plat_memctl_ZQ_force_config()
II: DRAM is set by software calibration... Start claibation test...PASSED

MCR  (0xb8001000):0x100421e0, 0x20320000, 0x54422830, 0x0404030f
DTR2 (0xb8001010):0x0630d000
DIDER(0xb8001050):0x80000000
DCDR (0xb8001060):0x9b540000
ZQPCR(0xb8001090):0x1f000000, 0x00000000, 0x00000000
PHY Registers(0xb8001500):
0xb8001500:0x80000010, 0x0000007f, 0x00000000, 0x00000000
0xb8001510:0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00
0xb8001520:0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00
0xb8001530:0x1e000000, 0x1e000000, 0x1e000000, 0x1e000000
0xb8001540:0x1e000000, 0x1e000000, 0x1e000000, 0x1e000000
0xb8001550:0x001e0f00, 0x001e0f00, 0x001e0f00, 0x001e0f00
0xb8001560:0x001e0f00, 0x001e0f00, 0x001e0f00, 0x001e0f00
0xb8001570:0x00000000, 0x00000000, 0x00000000, 0x00000000
0xb8001580:0x00000000, 0x00000000, 0x00000000, 0x00000000
0xb8001590:0x08000000, 0x66996699, 0x00000000, 0xaa55aa55
0xb80015a0:0x00000000, 0x00000000, 0x00000000, 0x00000000
II: Selected DRAM model #0.
II: Copying U-Boot from 0x9fc03ad8(235484B) to 0x83f00000... OK
II: NOR SPI Flash... searching flash parameters... supported flash ID: [c22018][c22019][c22019]... detected flash ID: [c22018]... OK

U-Boot 2011.12.39239-svn41394 (Jul 24 2013 - 09:38:34)

Board: RTL838x CPU:500MHz LXB:200MHz MEM:300MHz
DRAM:  128 MB
SPI-F: 1x16 MB
Loading 1024B env. variables from offset 0x40000
Switch Model: ZyXEL_GS1900_8 (Port Count: 8)
Switch Chip: RTL8382M
**************************************************
#### RTL8218B config - MAC ID = 8 ####
Now Internal PHY-A
REG32(0xbb00a17c) is        0
PHY[0]: disable EEE on port [0x0000ff00]
Net:   Net Initialization Skipped
rtl8380#0
Hit any key to stop autoboot:  0 
## Booting image from partition ... 0
## Booting kernel from Legacy Image at b4260000 ...
   Image Name:   MIPS OpenWrt Linux-6.6.72
   Created:      2025-01-21  18:59:07 UTC
   Image Type:   MIPS Linux Kernel Image (gzip compressed)
   Data Size:    3779743 Bytes = 3.6 MB
   Load Address: 80100000
   Entry Point:  80100000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 6.6.72 (root@060e6bb44dde) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28658-8624c6cab5) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Tue Jan 21 18:59:07 2025
[    0.000000] RTL838X model is 83806800
[    0.000000] SoC Type: RTL8380
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019070 (MIPS 4KEc)
[    0.000000] MIPS: machine is Zyxel GS1900-8v1/v2 Switch
[    0.000000] earlycon: ns16550a0 at MMIO 0x18002000 (options '115200n8')
[    0.000000] printk: bootconsole [ns16550a0] enabled
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Using appended Device Tree.
[    0.000000] Primary instruction cache 16kB, VIPT, 4-way, linesize 16 bytes.
[... 231 lines cut ...]
[   56.235260] rtl83xx-switch switch@1b000000 lan5: entered allmulticast mode
[   56.244102] rtl83xx-switch switch@1b000000 lan5: entered promiscuous mode
[   56.751261] rtl83xx-switch switch@1b000000 lan6: configuring for phy/internal link mode
[   56.781337] 8021q: adding VLAN 0 to HW filter on device lan6
[   56.971055] switch: port 6(lan6) entered blocking state
[   56.976968] switch: port 6(lan6) entered disabled state
[   56.983071] rtl83xx-switch switch@1b000000 lan6: entered allmulticast mode
[   56.991904] rtl83xx-switch switch@1b000000 lan6: entered promiscuous mode
[   57.391211] rtl83xx-switch switch@1b000000 lan7: configuring for phy/internal link mode
[   57.411079] 8021q: adding VLAN 0 to HW filter on device lan7
[   57.530817] switch: port 7(lan7) entered blocking state
[   57.536728] switch: port 7(lan7) entered disabled state
[   57.542839] rtl83xx-switch switch@1b000000 lan7: entered allmulticast mode
[   57.551659] rtl83xx-switch switch@1b000000 lan7: entered promiscuous mode
[   57.883925] rtl83xx-switch switch@1b000000 lan8: configuring for phy/internal link mode
[   57.911100] 8021q: adding VLAN 0 to HW filter on device lan8
[   58.021011] switch: port 8(lan8) entered blocking state
[   58.026924] switch: port 8(lan8) entered disabled state
[   58.033021] rtl83xx-switch switch@1b000000 lan8: entered allmulticast mode
[   58.041869] rtl83xx-switch switch@1b000000 lan8: entered promiscuous mode
[   58.112888] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported
[   87.971467] done.
[   87.973700] jffs2: notice: (1519) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   88.143637] overlayfs: upper fs does not support tmpfile
```

svanheule, 2025-01-22:
@es86de thanks for testing and good to hear things are now working as intended!

I will have a look at how to translate into some patches, so people can still use the HW MDIO bus if their device supports it.

svanheule, 2025-01-22:
Created a pull request with the complete changes: https://github.com/openwrt/openwrt/pull/17692

If you're installing a v1 image from that PR over your own build, you will have to force the sysupgrade due to the changed device compatible, but you can safely keep any existing config.

es86de, 2025-01-22:
It works! Thank you very much for your efforts! Just out of curiosity, how did you find out that the A revision has this problem?

svanheule, 2025-01-23:
I had come across it before, but I was going through some Realtek SDK code when I noticed it again. We never knew initially (years ago) how severe the issues of the rev. A silicon were, so it's interesting to see one in the wild.

svanheule, 2025-01-25:
Fixes merged to main, so this issue can be closed.

pkgadd, 2025-01-29:
Is there a(n easy) way to query the SOC revision from the running system (apart from just trying the gs1900-8-v2 image and watching if it fails), without having to resort to the bootloader? 

Admittedly, for my blue-PCB [gs1900-8](https://user-images.githubusercontent.com/6666853/107698368-bb774100-6cb4-11eb-88aa-6f7b8298f3ea.jpg) and all-sg8208m connecting to the serial console isn't much of a trouble, for some other switches it wouldn't be that trivial.

pkgadd, 2025-01-29:
That is interesting, if I read the bootloader output correctly, my blue-PCB seems to be rev. B:
```
II: Copied Preloader data from 0x9fc00000(15064B) to 0x9f000000.
II: Cleared Preloader BSS section at 0x9f003ad8(0B).
II: PLL is set by SW... OK
Setting DTR
INFO: Applying ZQ result directly by plat_memctl_ZQ_force_config()
II: DRAM is set by software calibration... Start claibation test...PASSED

MCR  (0xb8001000):0x100421e0, 0x20320000, 0x54422830, 0x0404030f
DTR2 (0xb8001010):0x0630d000
DIDER(0xb8001050):0x80000000
DCDR (0xb8001060):0x9b540000
ZQPCR(0xb8001090):0x1f000000, 0x00000000, 0x00000000
PHY Registers(0xb8001500):
0xb8001500:0x80000010, 0x0000007f, 0x00000000, 0x00000000
0xb8001510:0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00
0xb8001520:0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00, 0x0a1e0f00
0xb8001530:0x1e000000, 0x1e000000, 0x1e000000, 0x1e000000
0xb8001540:0x1e000000, 0x1e000000, 0x1e000000, 0x1e000000
0xb8001550:0x001e0f00, 0x001e0f00, 0x001e0f00, 0x001e0f00
0xb8001560:0x001e0f00, 0x001e0f00, 0x001e0f00, 0x001e0f00
0xb8001570:0x00000000, 0x00000000, 0x00000000, 0x00000000
0xb8001580:0x00000000, 0x00000000, 0x00000000, 0x00000000
0xb8001590:0x08000000, 0x66996699, 0x00000000, 0xaa55aa55
0xb80015a0:0x00000000, 0x00000000, 0x00000000, 0x00000000
II: Selected DRAM model #0.
II: Copying U-Boot from 0x9fc03ad8(235484B) to 0x83f00000... OK
II: NOR SPI Flash... searching flash parameters... supported flash ID: [c22018][c22019][c22019]... detected flash ID: [c22018]... OK

U-Boot 2011.12.39239-svn41394 (Jul 24 2013 - 09:38:34)

Board: RTL838x CPU:500MHz LXB:200MHz MEM:300MHz
DRAM:  128 MB
SPI-F: 1x16 MB
Loading 1024B env. variables from offset 0x40000
Switch Model: ZyXEL_GS1900_8 (Port Count: 8)
Switch Chip: RTL8382M
**************************************************
#### RTL8218B config - MAC ID = 8 ####
Now Internal PHY
REG32(0xbb00a17c) is        0
PHY[0]: disable EEE on port [0x0000ff00]
Net:   Net Initialization Skipped
rtl8380#0
Hit any key to stop autoboot:  0
Hit any key to stop autoboot:  0
RTL838x# mw.l bb000058 3
RTL838x# mw.l bb0000d8 a0000000
RTL838x# md.l bb0000d0 4
bb0000d0: 00000002 83806800 a0026275 00000016    ......h...bu....
```
…and `openwrt-realtek-rtl838x-zyxel_gs1900-8-v2-initramfs-kernel.bin` boots up fine.

<hr>

My Allnet ALL-SG8208m even seems to be rev. C:
```
U-Boot Version 1.0.0.48161 (Nov 28 2014 - 19:10:20)

Board: RTL838x CPU:500MHz LXB:200MHz MEM:300MHz
DRAM:  128 MB
SPI-F: 1x16 MB
Loading 65536B env. variables from offset 0x80000
[DBG] ### Board Probe ###
[DBG] ### Mac Driver Init ###
[DBG] ### Chip Config ###
Switch Model: NETRONIX_DEMO (Port Count: 8)
Switch Chip: RTL8382M
[DBG]   PHY[0] RTL8218B : macid =  8, phy_max = 8
[DBG] ### Platform Config ###
[DBG] ### MAC Config ###
[DBG] ### Intra-Serdes Config ###
[DBG] ### PHY Config (RTL82XX) ###
**************************************************
#### RTL8218B config - MAC ID = 8 ####
Now Internal PHY
[DBG] ### Misc Config ###
Net:   Net Initialization Skipped
rtl8380#0
Hit any key to stop autoboot:  3  0 
RTL838x# # 
RTL838x# # mw.l bb000058 3
RTL838x# # mw.l bb0000d8 a0000000
RTL838x# # md.l bb0000d0 4
bb0000d0: 00000003 83806800 a0036275 00000016    ......h...bu....
```
…and the `openwrt-realtek-rtl838x-zyxel_gs1900-8-v2-initramfs-kernel.bin` initramfs image boots up fine as well (it's the same PCB, just different flash partitioning).

[image]