openwrt/openwrt #9538: ODHCP IPV4 Default Router (Option 3) not working

clutch2sft, 2022-03-23:
Setup is ODHCP as maindhcp server and I want to use a different router on the network as the clients default gateway.

I used :

list dhcp_option '3,172.16.5.2'

But the darn dhcp server continues to hand out the routers interface as the gateway. Pulling my hair out.... reaching out.

Thanks!

jow-, 2022-03-23:
The odhcpd server does not implement `dhcp_option` at all. The readme [mentions a `router` option](https://github.com/openwrt/odhcpd/blob/master/README#L104-L105), try `list router 172.16.5.2`

clutch2sft, 2022-03-29:
Thank you - I was thinking about this too as the issue.  I did also try the "router" option with no positive result.

closed 2022-03-31 by clutch2sft