openwrt/openwrt #9539: Kernel Panic - Belkin rt3200 UBI - Oops: Fatal exception in interrupt

t3dium, 2022-03-23:
i've disabled hardware and software offloading as a temporary workaround for [9940](https://github.com/openwrt/openwrt/issues/9440), but i'm not quite sure what caused this kernel panic, I don't think it's related to the previous issue so I opened a new one.

pasted cat /sys/fs/pstore* logs below:

```
Oops#1 Part1
<6>[    8.872847] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20201105222230a
<6>[    8.872847]
<6>[    8.946812] mt7622-wmac 18000000.wmac: N9 Firmware Version: 2.0, Build Time: 20200131180931
<6>[    9.006374] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
<6>[    9.030784] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
<6>[    9.155048] PPP generic driver version 2.4.2
<6>[    9.160284] NET: Registered protocol family 24
<14>[    9.172422] kmodloader: done loading kernel modules from /etc/modules.d/*
<6>[   12.189343] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   12.193441] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   12.193556] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   12.193589] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<6>[   12.194803] mt7530 mdio-bus:00 lan1: configuring for phy/gmii link mode
<6>[   12.195517] 8021q: adding VLAN 0 to HW filter on device lan1
<6>[   12.200137] br-lan: port 1(lan1) entered blocking state
<6>[   12.200150] br-lan: port 1(lan1) entered disabled state
<6>[   12.201958] device lan1 entered promiscuous mode
<6>[   12.201968] device eth0 entered promiscuous mode
<6>[   12.220264] mt7530 mdio-bus:00 lan2: configuring for phy/gmii link mode
<6>[   12.220820] 8021q: adding VLAN 0 to HW filter on device lan2
<6>[   12.225115] br-lan: port 2(lan2) entered blocking state
<6>[   12.225126] br-lan: port 2(lan2) entered disabled state
<6>[   12.235581] device lan2 entered promiscuous mode
<6>[   12.243973] mt7530 mdio-bus:00 lan3: configuring for phy/gmii link mode
<6>[   12.244512] 8021q: adding VLAN 0 to HW filter on device lan3
<6>[   12.249154] br-lan: port 3(lan3) entered blocking state
<6>[   12.249166] br-lan: port 3(lan3) entered disabled state
<6>[   12.250522] device lan3 entered promiscuous mode
<6>[   12.258876] mt7530 mdio-bus:00 lan4: configuring for phy/gmii link mode
<6>[   12.259420] 8021q: adding VLAN 0 to HW filter on device lan4
<6>[   12.263478] br-lan: port 4(lan4) entered blocking state
<6>[   12.263491] br-lan: port 4(lan4) entered disabled state
<6>[   12.270111] device lan4 entered promiscuous mode
<6>[   12.273043] mt7530 mdio-bus:00 lan4: Link is Up - 100Mbps/Full - flow control rx/tx
<6>[   12.278676] mt7530 mdio-bus:00 lan3: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.282725] br-lan: port 4(lan4) entered blocking state
<6>[   12.282739] br-lan: port 4(lan4) entered forwarding state
<6>[   12.283078] br-lan: port 3(lan3) entered blocking state
<6>[   12.283089] br-lan: port 3(lan3) entered forwarding state
<6>[   12.284241] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
<6>[   12.286919] mt7530 mdio-bus:00 wan: configuring for phy/gmii link mode
<6>[   12.287490] 8021q: adding VLAN 0 to HW filter on device wan
<6>[   12.294437] mt7530 mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.294475] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
<6>[   14.866546] br-lan: port 5(wlan0) entered blocking state
<6>[   14.866559] br-lan: port 5(wlan0) entered disabled state
<6>[   14.866788] device wlan0 entered promiscuous mode
<6>[   14.964710] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   15.357750] br-lan: port 6(wlan1) entered blocking state
<6>[   15.357763] br-lan: port 6(wlan1) entered disabled state
<6>[   15.357970] device wlan1 entered promiscuous mode
<6>[   15.358050] br-lan: port 6(wlan1) entered blocking state
<6>[   15.358057] br-lan: port 6(wlan1) entered forwarding state
<6>[   15.359669] br-lan: port 1(lan1) entered blocking state
<6>[   15.359685] br-lan: port 1(lan1) entered forwarding state
<6>[   15.865738] br-lan: port 6(wlan1) entered disabled state
<6>[   22.719295] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<6>[   22.719425] br-lan: port 5(wlan0) entered blocking state
<6>[   22.719434] br-lan: port 5(wlan0) entered forwarding state
<6>[   63.183950] br-lan: port 3(lan3) entered disabled state
<6>[   63.184594] mt7530 mdio-bus:00 lan3: Link is Down
<6>[   95.709466] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[   95.709602] br-lan: port 6(wlan1) entered blocking state
<6>[   95.709610] br-lan: port 6(wlan1) entered forwarding state
[... 382 lines cut ...]
<7>[1180958.831617] x11: 0000000000000000 x10: 0000000000000820
<7>[1180958.837095] x9 : ffffffc0127c3d70 x8 : ffffff80008f6b80
<7>[1180958.842572] x7 : 0000000000000004 x6 : ffffff80029d5bd0
<7>[1180958.848050] x5 : 0000000000000000 x4 : 0000000000000000
<7>[1180958.853527] x3 : 0000000000000008 x2 : ffffff80029d5648
<7>[1180958.859004] x1 : ffffff80029d5bd8 x0 : 0000000000000038
<7>[1180958.864482] Call trace:
<7>[1180958.867104]  ieee80211_ctstoself_duration+0x494/0x594 [mac80211]
<7>[1180958.873284]  ieee80211_wake_txqs+0x30/0x50 [mac80211]
<7>[1180958.878505]  tasklet_action_common.constprop.0+0x134/0x17c
<7>[1180958.884156]  tasklet_action+0x24/0x30
<7>[1180958.887984]  _stext+0x124/0x28c
<7>[1180958.891290]  do_softirq+0x74/0x80
<7>[1180958.894770]  __local_bh_enable_ip+0x88/0x90
<7>[1180958.899120]  napi_threaded_poll+0x94/0xf0
<7>[1180958.903296]  kthread+0x120/0x124
<7>[1180958.906690]  ret_from_fork+0x10/0x18
<0>[1180958.910434] Code: d5033bbf 36180123 f9403fe0 b4000060 (b9411800)
<4>[1180958.916692] ---[ end trace e2e9109b2c1dda43 ]---
<0>[1180958.928156] Kernel panic - not syncing: Oops: Fatal exception in interrupt
<2>[1180958.935199] SMP: stopping secondary CPUs
<0>[1180958.939290] Kernel Offset: disabled
<0>[1180958.942945] CPU features: 0x0000002,04002004
<0>[1180958.947378] Memory Limit: none
```

mrkiko, 2022-03-23:
I think I was affected by a similar issue - the "ctstoself" function at least. I was not able to draw any useful conclusion.

mrkiko, 2022-03-24:
Can you try to disable the 2.4 ghz interface and let me know if this happens again? How much time does it take to happen in your scenario?

t3dium, 2022-03-26:
It happened again...

```
Oops#1 Part1
<6>[    8.872847] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20201105222230a
<6>[    8.872847]
<6>[    8.946812] mt7622-wmac 18000000.wmac: N9 Firmware Version: 2.0, Build Time: 20200131180931
<6>[    9.006374] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
<6>[    9.030784] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
<6>[    9.155048] PPP generic driver version 2.4.2
<6>[    9.160284] NET: Registered protocol family 24
<14>[    9.172422] kmodloader: done loading kernel modules from /etc/modules.d/*
<6>[   12.189343] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   12.193441] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   12.193556] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   12.193589] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<6>[   12.194803] mt7530 mdio-bus:00 lan1: configuring for phy/gmii link mode
<6>[   12.195517] 8021q: adding VLAN 0 to HW filter on device lan1
<6>[   12.200137] br-lan: port 1(lan1) entered blocking state
<6>[   12.200150] br-lan: port 1(lan1) entered disabled state
<6>[   12.201958] device lan1 entered promiscuous mode
<6>[   12.201968] device eth0 entered promiscuous mode
<6>[   12.220264] mt7530 mdio-bus:00 lan2: configuring for phy/gmii link mode
<6>[   12.220820] 8021q: adding VLAN 0 to HW filter on device lan2
<6>[   12.225115] br-lan: port 2(lan2) entered blocking state
<6>[   12.225126] br-lan: port 2(lan2) entered disabled state
<6>[   12.235581] device lan2 entered promiscuous mode
<6>[   12.243973] mt7530 mdio-bus:00 lan3: configuring for phy/gmii link mode
<6>[   12.244512] 8021q: adding VLAN 0 to HW filter on device lan3
<6>[   12.249154] br-lan: port 3(lan3) entered blocking state
<6>[   12.249166] br-lan: port 3(lan3) entered disabled state
<6>[   12.250522] device lan3 entered promiscuous mode
<6>[   12.258876] mt7530 mdio-bus:00 lan4: configuring for phy/gmii link mode
<6>[   12.259420] 8021q: adding VLAN 0 to HW filter on device lan4
<6>[   12.263478] br-lan: port 4(lan4) entered blocking state
<6>[   12.263491] br-lan: port 4(lan4) entered disabled state
<6>[   12.270111] device lan4 entered promiscuous mode
<6>[   12.273043] mt7530 mdio-bus:00 lan4: Link is Up - 100Mbps/Full - flow control rx/tx
<6>[   12.278676] mt7530 mdio-bus:00 lan3: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.282725] br-lan: port 4(lan4) entered blocking state
<6>[   12.282739] br-lan: port 4(lan4) entered forwarding state
<6>[   12.283078] br-lan: port 3(lan3) entered blocking state
<6>[   12.283089] br-lan: port 3(lan3) entered forwarding state
<6>[   12.284241] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
<6>[   12.286919] mt7530 mdio-bus:00 wan: configuring for phy/gmii link mode
<6>[   12.287490] 8021q: adding VLAN 0 to HW filter on device wan
<6>[   12.294437] mt7530 mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.294475] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
<6>[   14.866546] br-lan: port 5(wlan0) entered blocking state
<6>[   14.866559] br-lan: port 5(wlan0) entered disabled state
<6>[   14.866788] device wlan0 entered promiscuous mode
<6>[   14.964710] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   15.357750] br-lan: port 6(wlan1) entered blocking state
<6>[   15.357763] br-lan: port 6(wlan1) entered disabled state
<6>[   15.357970] device wlan1 entered promiscuous mode
<6>[   15.358050] br-lan: port 6(wlan1) entered blocking state
<6>[   15.358057] br-lan: port 6(wlan1) entered forwarding state
<6>[   15.359669] br-lan: port 1(lan1) entered blocking state
<6>[   15.359685] br-lan: port 1(lan1) entered forwarding state
<6>[   15.865738] br-lan: port 6(wlan1) entered disabled state
<6>[   22.719295] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<6>[   22.719425] br-lan: port 5(wlan0) entered blocking state
<6>[   22.719434] br-lan: port 5(wlan0) entered forwarding state
<6>[   63.183950] br-lan: port 3(lan3) entered disabled state
<6>[   63.184594] mt7530 mdio-bus:00 lan3: Link is Down
<6>[   95.709466] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[   95.709602] br-lan: port 6(wlan1) entered blocking state
<6>[   95.709610] br-lan: port 6(wlan1) entered forwarding state
<6>[1180958.175469] device wlan0 left promiscuous mode
<6>[1180958.175567] br-lan: port 5(wlan0) entered disabled state
[... 623 lines cut ...]
<7>[303344.257162] x9 : ffffffc010e73d70 x8 : ffffff8000190880
<7>[303344.262554] x7 : 0000000000000003 x6 : ffffff8002889bd0
<7>[303344.267944] x5 : 0000000000000000 x4 : 0000000000000000
<7>[303344.273335] x3 : 0000000000000008 x2 : ffffff8002889648
<7>[303344.278726] x1 : ffffff8002889bd8 x0 : 0000000000000038
<7>[303344.284117] Call trace:
<7>[303344.286653]  ieee80211_ctstoself_duration+0x494/0x594 [mac80211]
<7>[303344.292747]  ieee80211_wake_txqs+0x30/0x50 [mac80211]
<7>[303344.297881]  tasklet_action_common.constprop.0+0x134/0x17c
<7>[303344.303444]  tasklet_action+0x24/0x30
<7>[303344.307184]  _stext+0x124/0x28c
<7>[303344.310405]  do_softirq+0x74/0x80
<7>[303344.313797]  __local_bh_enable_ip+0x88/0x90
<7>[303344.318061]  napi_threaded_poll+0x94/0xf0
<7>[303344.322151]  kthread+0x120/0x124
<7>[303344.325458]  ret_from_fork+0x10/0x18
<0>[303344.329116] Code: d5033bbf 36180123 f9403fe0 b4000060 (b9411800)
<4>[303344.335286] ---[ end trace 4a127c8d772fa9fa ]---
<0>[303344.349113] Kernel panic - not syncing: Oops: Fatal exception in interrupt
<2>[303344.356071] SMP: stopping secondary CPUs
<0>[303344.360076] Kernel Offset: disabled
<0>[303344.363644] CPU features: 0x0000002,04002004
<0>[303344.367989] Memory Limit: none
```
any way to fix this?

dangowrt, 2022-03-26:
@nbd168 @ryderlee1110 @LorenzoBianconi any ideas?

nbd168, 2022-03-26:
This commit might help with the crash: https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=e6ae25ca4de843df4e866fe5fa5af7ddec56db5c

mrkiko, 2022-04-06:
Hi @t3dium
Any update on the matter? Have you tried the proposed commit?

t3dium, 2022-05-31:
Apologies for the late reply, I can confirm nbd’s commit did fix the issue, haven’t gotten a kernel panic since. @nbd168 @mrkiko

closed 2022-06-05 by t3dium