openwrt/openwrt #9540: firewall4: add dependency iptables-nft and ip6tables-nft?

feckert, 2022-03-23:
version: reboot-19180-g73c6d8fd04
I have installed a fresh system. I have additional installed the mwan3 package.
I now have a problem with the firewall.

The mwa3 works with both firewall3(iptables) and firewall4(nft). For this to work, the rules for iptables must be created with iptables-legacy or for nft with iptables-nft.
The firewall4(nft) is installed by default and only iptables or ip6tables is in the dependency of mwan3.
This means that the package is satisfied when the dependency are resolved (PROVIDE:iptables or PROVIDE:ip6tables).

On my system, however, iptables-legacy(IPv4) and iptables-nft(IPv6) are now installed.
I would have expected that if the firewall4(nft) is installed then iptables-nfg(IPv4) and ip6tables(IPv6) is installed as a transitional solution.

As it is now, it is definitely bad. I think if we use firewall4(nft) we have to install iptables-nft and ip6tables-nft

champtar, 2022-03-23:
ip(6)tables is not needed with fw4, so we should not bloat the images by default
What we could do is add a conflict between ip(6)tables-legacy and fw4 to make sure people use ip(6)tables-nft with fw4

feckert, 2022-03-23:
yes that would not be bad.
Because currently I have the situation that with mwan3 the IPv4 rules of mwan3 are installed with iptables-legacy (iptables points to iptables-legacy) and the IPv6 rules are installed with ip6tables-nft (ip6tables command points to ip6tables-nft).

I can only work around this if I change the dependency in mwan3.
But then I have to decide between `legacy` or `nft`.
Or add a mwan3-legacy package or mwan3-nft package

```
root@VR2-106149 /usr/sbin # ls -la | grep iptables
lrwxrwxrwx    1 root     root            30 Mar 23 09:26 iptables -> /usr/sbin/xtables-legacy-multi
lrwxrwxrwx    1 root     root            20 Mar 23 09:26 iptables-legacy -> xtables-legacy-multi
lrwxrwxrwx    1 root     root            20 Mar 23 09:26 iptables-legacy-restore -> xtables-legacy-multi
lrwxrwxrwx    1 root     root            20 Mar 23 09:26 iptables-legacy-save -> xtables-legacy-multi
lrwxrwxrwx    1 root     root            30 Mar 23 09:26 iptables-restore -> /usr/sbin/xtables-legacy-multi
lrwxrwxrwx    1 root     root            30 Mar 23 09:26 iptables-save -> /usr/sbin/xtables-legacy-multi
```

```
root@VR2-106149 /usr/sbin # ls -la | grep ip6tables
lrwxrwxrwx    1 root     root            27 Mar 23 09:26 ip6tables -> /usr/sbin/xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Mar 23 09:26 ip6tables-nft -> xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Mar 23 09:26 ip6tables-nft-restore -> xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Mar 23 09:26 ip6tables-nft-save -> xtables-nft-multi
lrwxrwxrwx    1 root     root            27 Mar 23 09:26 ip6tables-restore -> /usr/sbin/xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Mar 23 09:26 ip6tables-restore-translate -> xtables-nft-multi
lrwxrwxrwx    1 root     root            27 Mar 23 09:26 ip6tables-save -> /usr/sbin/xtables-nft-multi
lrwxrwxrwx    1 root     root            17 Mar 23 09:26 ip6tables-translate -> xtables-nft-multi
```

bluewavenet, 2022-03-26:
@feckert @champtar 

> On my system, however, iptables-legacy(IPv4) and iptables-nft(IPv6) are now installed.

It is worse than that, you can install **_both_** iptables-legacy and iptables-nft.
Which ever one was installed last is the one that takes precedence so the results can be rather unpredictable.

If you do `opkg install iptables` you get iptables-legacy. Post 21.02 (ie master and 22.03) this should default to iptables-nft.

The problem is in the iptables makefile. It has:
```
define Package/iptables-legacy
$(call Package/iptables/Default)
  TITLE:=IP firewall administration tool
  DEPENDS+= +xtables-legacy
  PROVIDES:=iptables
  ALTERNATIVES:=\
    200:/usr/sbin/iptables:/usr/sbin/xtables-legacy-multi \
    200:/usr/sbin/iptables-restore:/usr/sbin/xtables-legacy-multi \
    200:/usr/sbin/iptables-save:/usr/sbin/xtables-legacy-multi
endef
```
and it also has:
```
define Package/iptables-nft
$(call Package/iptables/Default)
  TITLE:=IP firewall administration tool nft
  DEPENDS:=+kmod-ipt-core +xtables-nft
  PROVIDES:=iptables
  ALTERNATIVES:=\
    300:/usr/sbin/iptables:/usr/sbin/xtables-nft-multi \
    300:/usr/sbin/iptables-restore:/usr/sbin/xtables-nft-multi \
    300:/usr/sbin/iptables-save:/usr/sbin/xtables-nft-multi
endef
```

So two `PROVIDES:=iptables` entries.
**Should there only be one PROVIDES entry, and only in the `define Package/iptables-nft` section?**

Having two Provides entries seems to have two undesirable effects:
1. As iptables-legacy comes first in the makefile, iptables-legacy becomes the default
2. It is possible to install both iptables-nft and iptables-legacy with unpredictable results.

**Should iptables-legacy have iptables-nft as a CONFLICTS entry (and vice-versa)?**

Iptables does not need to be a dependency of firewall4 but for those packages that do have such a dependency, defaulting to iptables-nft is what is required. if someone finds some extreme edge case where iptables-nft does not work for them then they can choose to install iptables-legacy.

hnyman, 2022-03-27:
> It is worse than that, you can install both iptables-legacy and iptables-nft.
> Which ever one was installed last is the one that takes precedence so the results can be rather unpredictable.

The "alternatives" statements in iptables should actually take care that iptables-nft binaries are preferred for the ip(6)tables symlinks, if both iptables-legacy and iptables-nft are installed.  But that does nto take care of the dependency evaluation by opkg.

> Having two Provides entries seems to have two undesirable effects:
>
>   1. As iptables-legacy comes first in the makefile, iptables-legacy becomes the default
>   2. It is possible to install both iptables-nft and iptables-legacy with unpredictable results.

Not sure if we could declare iptables-nft as the DEFAULT_VARIANT ?
The Makefile defines so many packages that I wonder is it is possible to declare (iptables-nft vs. iptables-legacy) and ip6tables-nft vs. ip6tables-legacy) at the same time.
Or would the Makefile need to be split into several Makefiles?

```
I can only work around this if I change the dependency in mwan3.
But then I have to decide between legacy or nft.
Or add a mwan3-legacy package or mwan3-nft package
```

In general, I feel that we should aim for a quick fw4, nft  and iptables-nft transition, instead of serving "I want to use fw3 forever" crowd. E.g. mwan should directly depend on nft ot iptables-nft (and possibly have a legacy variant for -legacy ?)

hnyman, 2022-03-27:
> What we could do is add a conflict between ip(6)tables-legacy and fw4 to make sure people use ip(6)tables-nft with fw4

Might be a quick solution, fo the time being.  (if multiple DEFAULT_VARIANTs can't be declared in the same Makefile)

bluewavenet, 2022-03-27:
@hnyman 

> In general, I feel that we should aim for a quick fw4, nft and iptables-nft transition

Absolutely yes!
**That is why we need the default install of "iptables" to be iptables-nft.**
Currently the _default install of "iptables" is iptables-legacy and that causes all sorts of problems_.

Borromini, 2022-08-26:
Is this issue still valid, or can it be closed?

bluewavenet, 2022-08-26:
@Borromini 
There is currently only a workaround/ugly hack:
https://github.com/openwrt/openwrt/commit/35fec487e30f05c81bd135326a993dad7f861812

The whole thing is still a mess. Even now, unless a package specifically has a dependency of iptables-nft, it will still default to iptables-zz-legacy (aka iptables-legacy)

champtar, 2022-08-28:
@bluewavenet this should be fixed with https://github.com/openwrt/openwrt/pull/10483

alexeys85, 2022-09-24:
I have a custom build from up to date master branch and don't have packages that depend on iptables, but libipxtc, libxtables and  xtables-nft-multi are still built. However, `.config` claims `# CONFIG_PACKAGE_ip6tables-zz-legacy is not set`,  `# CONFIG_PACKAGE_iptables-zz-legacy is not set` and any other `*iptables*` symbol is not set. So, is it possible to not build iptables* packages?

hnyman, 2022-09-24:
> don't have packages that depend on iptables, but libipxtc, libxtables and xtables-nft-multi are still built. However, .config claims # CONFIG_PACKAGE_ip6tables-zz-legacy is not set, # CONFIG_PACKAGE_iptables-zz-legacy is not set and any other *iptables* symbol is not set.

What is your full build config???? 
(The diffconfig output)

alexeys85, 2022-09-24:
@hnyman, <details><summary>diffconfig</summary>
<pre>
CONFIG_TARGET_omap=y
CONFIG_TARGET_omap_DEVICE_ti_am335x-bone-black=y
CONFIG_TARGET_BOARD="omap"
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
# CONFIG_BINUTILS_USE_VERSION_2_37 is not set
CONFIG_BINUTILS_USE_VERSION_2_38=y
CONFIG_BINUTILS_VERSION="2.38"
CONFIG_BINUTILS_VERSION_2_38=y
CONFIG_DROPBEAR_ECC=y
CONFIG_DROPBEAR_ECC_FULL=y
CONFIG_EXTRA_BINUTILS_CONFIG_OPTIONS="--enable-gold"
# CONFIG_GCC_USE_VERSION_11 is not set
CONFIG_GCC_USE_VERSION_12=y
CONFIG_GCC_VERSION="12.2.0"
CONFIG_GCC_VERSION_12=y
CONFIG_GDB_PYTHON=y
CONFIG_IMAGEOPT=y
CONFIG_IPERF_ENABLE_MULTICAST=y
CONFIG_LINUX_5_15=y
CONFIG_LUA_ECO_DEFAULT_OPENSSL=y
CONFIG_LUA_ECO_OPENSSL=y
# CONFIG_LUCI_CSSTIDY is not set
# CONFIG_LUCI_JSMIN is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_BLAKE2=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_arp-scan=y
CONFIG_PACKAGE_arp-scan-database=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-container=y
CONFIG_PACKAGE_boost-json=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_collectd=y
CONFIG_PACKAGE_collectd-mod-cpu=y
CONFIG_PACKAGE_collectd-mod-cpufreq=y
CONFIG_PACKAGE_collectd-mod-ethstat=y
CONFIG_PACKAGE_collectd-mod-exec=y
CONFIG_PACKAGE_collectd-mod-interface=y
CONFIG_PACKAGE_collectd-mod-iwinfo=y
CONFIG_PACKAGE_collectd-mod-load=y
CONFIG_PACKAGE_collectd-mod-memory=y
CONFIG_PACKAGE_collectd-mod-network=y
CONFIG_PACKAGE_collectd-mod-nftables=y
CONFIG_PACKAGE_collectd-mod-processes=y
CONFIG_PACKAGE_collectd-mod-rrdtool=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_gdbserver=y
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_ifstat=y
CONFIG_PACKAGE_iftop=y
[... 129 lines cut ...]
CONFIG_USE_STRIP=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT="TI AM335x BeagleBone Black"
CONFIG_VERSION_REPO=""
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_ZLIB_OPTIMIZE_SPEED=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-shared-libs=y
CONFIG_boost-variant-release=y
# CONFIG_IFSTAT_SNMP is not set
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
# CONFIG_PACKAGE_libnetsnmp is not set
# CONFIG_PACKAGE_libpci is not set
# CONFIG_PACKAGE_libpcre is not set
# CONFIG_VERSION_CODE_FILENAMES is not set
# CONFIG_VERSION_FILENAMES is not set
</pre>
</details>

champtar, 2022-09-24:
@alexeys85 can't reproduce your issue
I've put your diffconfig in `.config`, ran `make defconfig`, and this is what I get:
```
$ grep tables .config | grep -v '^#'
CONFIG_DEFAULT_nftables=y
CONFIG_PACKAGE_nftables-json=y
```

alexeys85, 2022-09-24:
@hnyman, I have the same `.config` content but `xtables-nft-multi`, `libip4tc` and `libip6tc` are still built, but don't copy to the staging dir.