openwrt/openwrt #9542: Several packages still call fw3 (old firewall) from their scripts

hnyman, 2022-03-23:
@nbd168 @dedeckeh @stintel 

There are still core packages calling the old firewall via "fw3" in the init scripts & hotplug scripts.

The packages feed PR  https://github.com/openwrt/packages/pull/18133 that aims to fix miniupnpd for firewall4 made me to look for "fw3" also in the core OpenWrt source repo, and I found out that also some core apps still call fw3.

Also netifd does that (for 6rd):
https://github.com/openwrt/openwrt/blob/master/package/network/config/netifd/files/lib/netifd/dhcp.script#L78

Other apps found with  https://github.com/openwrt/openwrt/search?q=fw3&type=code

* odhcp6c
  https://github.com/openwrt/openwrt/blob/master/package/network/ipv6/odhcp6c/files/dhcpv6.script#L166

*  ppp
  https://github.com/openwrt/openwrt/blob/master/package/network/services/ppp/files/lib/netifd/ppp6-up#L20

* omcproxy  (which curiously calls fw3 without declaring firewall as dependency)
  https://github.com/openwrt/openwrt/blob/master/package/network/services/omcproxy/files/omcproxy.init#L73
https://github.com/openwrt/openwrt/blob/master/package/network/services/omcproxy/files/omcproxy.init#L100

* 464xlat
  https://github.com/openwrt/openwrt/blob/master/package/network/ipv6/464xlat/files/464xlat.sh#L59

* comgt
  https://github.com/openwrt/openwrt/blob/master/package/network/utils/comgt/files/ncm.sh#L154

* map 
  https://github.com/openwrt/openwrt/blob/master/package/network/ipv6/map/files/map.sh#L158

* uqmi
  https://github.com/openwrt/openwrt/blob/master/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh#L349

Plus in packages feed:
*  netsnmp
https://github.com/openwrt/packages/blob/master/net/net-snmp/files/snmpd.init#L266

obsy, 2022-03-24:
How about making /sbin/fw as a symlink to /sbin/ fw4 or /sbin/fw3, depending on the package installed? And using /sbin/fw in packages.

schuettecarsten, 2022-04-16:
That might be the root cause why forwarding packages from LAN to an QMI device does not work, while accessing the QMI device from the router itself works fine. I this this needs to be fixed before 22.03 release, so I link https://github.com/openwrt/openwrt/issues/9248 here.

Updated link for uqmi:
https://github.com/openwrt/openwrt/blob/master/package/network/utils/uqmi/files/lib/netifd/proto/qmi.sh#L353

Ansuel, 2022-06-04:
@hnyman should we pin this and give it priority? it's becoming problematic and it's causing tons of new issue with user complaining that things doesn't work.

jow-, 2022-06-04:
The `firewall4` package already ships an `fw3 -> fw4` symlink, so I don't see a problem with this. The `fw4` cli syntax is fully compatible with the `fw3` one.

schuettecarsten, 2022-06-12:
@jow- Well, it does not work for at least uqmi. After upgrading to any 22.03 build (rc0, snapshot), the router does not forward to the uqmi connection, while accessing internet from the router itself works fine. Downgrading to latest stable build fixes the issue without changing any configuration settings, and then upgrading to a 22.03 build breaks it again. This is also reproducteable when starting with a complete new Openwrt config.

jow-, 2022-06-12:
@schuettecarsten - well, then it should be fixed. But from the problem description, it does not sound as if `s/fw3/fw4/` will fix it either since with fw4, both executable names will behave identically. Since I don't have access to any hardware using an uqmi connection, someone else has to debug it.

systemcrash, 2022-12-12:
Could this be why my system logs flood with e.g.:

```
Mon Dec 12 20:48:50 2022 daemon.notice miniupnpd[5217]: remove port mapping 19549 TCP because it has expired
Mon Dec 12 20:48:50 2022 daemon.err miniupnpd[5217]: send_batch: mnl_cb_run returned -1
Mon Dec 12 20:48:50 2022 daemon.err miniupnpd[5217]: nft_send_rule(0x77b5a5f0, 8, 2) send_batch failed -4
Mon Dec 12 20:48:50 2022 daemon.notice miniupnpd[5217]: remove port mapping 34035 TCP because it has expired
Mon Dec 12 20:48:50 2022 daemon.err miniupnpd[5217]: send_batch: mnl_cb_run returned -1
Mon Dec 12 20:48:50 2022 daemon.err miniupnpd[5217]: nft_send_rule(0x77b5a5f0, 8, 2) send_batch failed -4
Mon Dec 12 20:48:50 2022 daemon.notice miniupnpd[5217]: remove port mapping 19549 TCP because it has expired
Mon Dec 12 20:48:50 2022 daemon.err miniupnpd[5217]: send_batch: mnl_cb_run returned -1
Mon Dec 12 20:48:50 2022 daemon.err miniupnpd[5217]: nft_send_rule(0x77b5a5f0, 8, 2) send_batch failed -4
```