openwrt/openwrt #9543: [ubus][procd]The system failed to sbootup because Ubus Connect fail

yangqingshi, 2022-03-24:
I have an OpenWRT system, but after I upgraded the system through Sysupgrade for many times, I found that the system could not be started.What are the possible causes of uBUS connection failure?
Because the system cannot bootup, it cannot check the condition of the file system

[   17.327706] <2>.(2)[1:procd]procd: Coldplug complete
[   17.328373] <2>.(2)[1:procd]procd: Change state 1 -> 2
[   17.329046] <2>.(2)[1:procd]procd: - watchdog -
[   17.329635] <2>.(2)[1:procd][wdtk] kick watchdog
[   17.330225] <2>.(2)[1:procd][wdtk] kick watchdog
[   17.330826] <2>.(2)[1:procd]procd: Ping
[   17.331326] <2>.(2)[1:procd][wdtk] kick watchdog
[   17.331928] <2>.(2)[1:procd]procd: Opened watchdog with timeout 30s
[   17.332835] <2>.(2)[1:procd]procd: - ubus -
[   17.333404] <2>.(2)[1:procd]procd: Create service ubus
[   17.334095] <2>.(2)[1:procd]procd: Start instance ubus::instance1
[   17.335220] <2>.(2)[1:procd]procd: Started instance ubus::instance1[3016]
[   17.336134] <2>.(2)[1:procd]procd: running /etc/init.d/ubus running
[   17.347091] <2>.(2)[1:procd]procd: glob failed on /etc/init.d/ubus
[ql_ubus][ballon] enter ubus_connect,line:353.
[ql_ubus][ballon[   17.383557] <2>.(2)[1:procd]procd: Connection to ubus failed
] ubus_reconnect(),line:391.
[ql_ubus][ballon] ubus_reconnect(),line:432,ret:9.
[ql_ubus][ballon] ubus_reconnect fail,line:302.
[ql_ubus][ballon] ubus_connect_ctx error,line:361.
[   17.408212] <3>.(3)[36:kworker/3:1][quectel_phy] blink no Ok 1602, nTimeout = 72
[ql_ubus][ballon] enter ubus_con[   17.483840] <2>.(2)[1:procd]procd: Connection to ubus failed

jow-, 2022-04-13:
Duplicate of #9545

closed 2022-04-13 by jow-

yangqingshi, 2022-05-25:
hi  @jow- The problem has not been solved，
**First of all**, the probability of this problem happening is not 100%. But once it happens,  will always happening even reboot。
**next**，I noticed that it's related to  with the user of the file "/var/run/ubus.sock".
[2022-05-24 08:24:45.852954] 'srw-rw-rw-    1 **0**        root             0 Jan  1 00:00 /var/run/ubus.sock\r\n'
when the problem happening, the file user is  0, not root