openwrt/openwrt #9545: all platforms: build system permission handling is brittle and can build images than hang on boot

Lanchon, 2022-03-24:
this has been going on for a long time: it affects at least v21.02 all the way to master right now. there are many reports for many platforms that can all be traced back to the same root cause.

### symptoms when running builds affected by this bug

initramfs boots and boots after sysupgrade are both affected. in normal boots, ubus is non-functional and the system freezes right after `procd: Change state 1 -> 2` and `procd: - ubus -` messages. increasing log level shows repeated `procd: Connection to ubus failed` or `procd: Instance ubus::instance1 exit with error code 32512 after 0 seconds` messages. there is no console access at this point. on the other hand, failsafe mode works.

### root cause

OpenWrt's build system depends on un*x-style permissions in the source tree an build dirs being set in a specific way for the build to be functional. if this condition is not met, the build silently succeeds but is non-functional if booted.

the problem is exacerbated by OpenWrt's choice of source code management: git. **git does not store permissions in the repo.** it only stores a single bit per file: either executable or not.

at checkout time, git will create files and directories using the permission defaulting mechanism of the platform it is running on. at build time, new files will also be created with default permissions. but OpenWrt's build system depends on specific permission in its trees for correctness. **hence the issue:** the correctness of successful OpenWrt builds depends on permission defaulting settings at source tree creation time (typically git checkout) and build time. this is neither documented (AFAICT) nor detected during build.

### the build requirements

files must be readable and writable by owner and readable but not writable by other. they must also be readable by group, but whether the group writable permission is acceptable or is a security risk is unknown to me. (it probably is acceptable, but security issues should not be taken lightly. i believe on official builds files are not group writable.)

### proposed fix

the build system packages files at various stages (cpio, squashfs, opkg packages, etc), and at least some of those packages encode permissions of the enclosed files, and at least some of the enclosed files require specific permissions. **the robust way of fixing this issue is to set the required permissions at build time on all files that are to be packaged.**

### band-aids

- document this brittleness.
- add checks during build time:
  - check permissions of some known file(s) in the source tree (eg `/etc/passwd`) as a proxy for checking all source tree files.
  - check permissions of some file(s) created during build as a proxy for checking all built files.

# workaround

before checking out the source tree, `cd` to the directory where you will checkout and run these commands:

1. run `umask`

if the result is `0022` or `0002` you are ok. if not:
- **either:** change your umask now to `0022` before git checkout, and also later every time before modifying or building the sources.
- **or:** change your default umask according to your platform. (hint: see `UMASK` and `USERGROUPS_ENAB` in `/etc/login.defs`.)

2. run `getfacl -d .`

if you see this output, the default ACL is fine and you are ok:
```
# file: .
# owner: [...]
# group: [...]
user::rwx
group::r-x
other::r-x
```
else, if you see this output (no default ACL) and your umask is `0022`, you are also ok:
```
# file: .
# owner: [...]
# group: [...]
```
else, you need to set the default ACL for the current directory with this command:
```
setfacl --set u::rwx,g::rx,o::rx .
```
(and you can recheck the ACL with `getfacl -d .`)

Lanchon, 2022-03-25:
added [a warning](https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem?do=diff&rev2%5B0%5D=1648167925&rev2%5B1%5D=1648171859&difftype=inline) to the wiki. please revert that change if this issue is fixed.

daniel-santos, 2023-08-20:
Has a root cause been identified with this issue yet. I have a ***strange*** set of boards that are having this problem even when flashing older firmware that had worked just fine prior. I'm wondering if the factory partition was accidentally altered (by some bug in our code) and is somehow triggering this strange failure.

Are there any issues tracking this? Unfortunately this firmware is based off of 21.02.0. I've merged on a different branch, but not for this branch I'm working on.

daniel-santos, 2023-08-20:
> I have a _**strange**_ set of boards

After comparing debug output, one board was spamming
```
[   83.327764] procd: {{"ACTION":"change","DEVPATH":"/devices/platform/101c0000.ehci/usb1/1-1/1-1:1.0","SUBSYSTEM":"usb","OVER_CURRENT_PORT":"/devices/platform/101c0000.ehci/usb1/1-1/1-1:1.0/1-1-port4","OVER_CURRENT_COUNT":"290","DEVTYPE":"usb_interface","DRIVER":"h}
Failed to connect to ubus
```

The other two had their UARTs fried, so weren't spamming anything. I'm guessing that the "Failed to connect to ubus" message here is because procd is in the "early" state when ubusd hasn't started yet, but some things are trying to talk to it anyway.

I have patches with a bunch of boot diagnostics, but they aren't yet ready to submit. I'm hoping I'll get to finish them (after rebasing) and bring some improvement to this area of OpenWRT. It's actually a whole set of "non-release" mods enabled by a single .config entry so that there's no chance any of them accidentally end up in production (security). I have most of an interactive boot process working.

Lanchon, 2023-08-22:
i am sorry about your boards. as you figured out, this issue is not related to your issue. although some symptoms might be similar (ubus not functional), the root cause of this issue is incorrectly set file permissions during build time. thanks!

daniel-santos, 2023-08-22:
AH HAH! Then it can be helpful to know that it can also be triggered dynamically as well! I was working on a luci-app and decided to quickly copy my files over via `tar c .|ssh root@myboard 'tar xC /'. These were the files copied.

tmp/myapp/myapp.pl
usr/lib/lua/luci/controller/myapp.lua
usr/lib/lua/luci/model/cbi/myapp.lua
usr/share/luci/menu.d/luci-app-myapp.json
usr/share/rpcd/acl.d/luci-app-myapp.json

Of course, `--same-owner --preserve-permissions` are enabled by default when run as root, so I ended up with these files owned by 1000:1000 and BOOM!

The odd thing is that even after I removed the files, and did a safe reboot via sysrq from the serial console (i.e., `e` to `killall -s SIGTERM`, wait a few seconds, `k` to SIGKILL, `s` to sync, `u` to remount all read-only, and `b` to reboot), it still didn't reboot on a non-fried board -- I had to reflash from tftp via uBoot.

I would love to figure out what actually happened. I suppose I could put in an SD card, do the things that causes it to die and then `dd` the jffs2 overlay to the SD card so I can see what was different. I suppose I could also alter procd to launch ubus under gdbserver? I can always debug from the kernel level, but I've never figured out how debug a program from kgdb.

Lanchon, 2023-08-22:
i suppose there are countless ways for ubus not to start. this issue is now well known and it pertains a bug in the build system, not in openwrt itself. the way to fix this issue is to fix the build system as explained in the first post. if you think you found a bug, please open a different issue for that even if the ubus failing symptom is the same. if you want help with your setup, it's best to search then post a new thread in the openwrt forum. thanks!

daniel-santos, 2023-08-22:
OK, Thanks @Lanchon . I'll open a new bug. While I understand the build system bug, it shouldn't ever cause a core service to terminate unless there's really no way for it to continue running.

Lanchon, 2023-08-22:
thanks. automatic recovery is a double edge sword: it adds complexity and thus maintenance burden to the code base, using up resources that maybe could be best invested elsewhere. also, by not failing, automatic recovery may cover up misconfigurations making them harder to detect and track down.

regarding this issue, fixing ubus is not the solution. openwrt has always been developed and tested with some file permissions assumed. fixing ubus would allow builds with improper permissions to boot, very possibly causing stranger issues down the road. it took *years* for this issue to be reproduced and characterized, and for a workaround to be documented. previously, only "try building using this or that VM" shots in the dark were offered. i believe the same would happen again for other issues if ubus was "fixed" to work with improper permissions instead of fixing the build system.

thanks again!

daniel-santos, 2023-08-22:
Well, imho, what we need is a proper "debug" build. I've been slowly working towards this, but I keep getting pulled off onto other projects. Proper checks can be added for anomalies in a debug/checked build that aren't included in release, making development and debugging easier, and overlooking bugs harder.

Between libubox, procd and ubox, I've got some 1500 lines touched over 25-ish files and it's not even done yet. My primary task was an interactive boot to allow for better diagnostics and this is mostly working. Breaking it apart into clean reviewable patches will be yet another task once it's ready.

Lanchon, 2024-01-18:
@dwmw2 thank you! i'd guess many of those files come from different packages so everything is kinda messed up on the non-working builds.

i'd really like the build system to at least be patched with a band-aid as described in the OP. i'm willing to bet this issue has bitten many would be users that gave up on the project when stuff silently failed... i know i almost did.

Lanchon, 2024-05-09:
> My umask says 022, not 0022. Is that the same thing?

https://stackoverflow.com/questions/13268796/umask-difference-between-0022-and-022