openwrt/openwrt #9547: Omega Onion2 missing broken-flash-reset

jsantala, 2022-03-25:
It seems that broken-flash-reset flag was dropped for Onion Omega2+ from v19.08 to v21.02. The device will not reboot without this patch. See below. Thanks!

diff --git a/target/linux/ramips/dts/mt7628an_onion_omega2.dtsi b/target/linux/ramips/dts/mt7628an_onion_omega2.dtsi
index b6d2df4756..cd41be0d4a 100644
--- a/target/linux/ramips/dts/mt7628an_onion_omega2.dtsi
+++ b/target/linux/ramips/dts/mt7628an_onion_omega2.dtsi
@@ -99,6 +99,7 @@
 		compatible = "jedec,spi-nor";
 		reg = <0>;
 		spi-max-frequency = <40000000>;
+		broken-flash-reset;
 
 		partitions {
 			compatible = "fixed-partitions";

pepe2k, 2022-03-27:
@jsantala According to the manufacturer website (https://onion.io/omega2/) regular version of the Omega2 has only 16 MB of flash. DTS for the Omega2+ has this flag included, see: https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7628an_onion_omega2p.dts#L9

jsantala, 2022-03-27:
Thanks @pepe2k  for the reply and the update. I am indeed on Omega2+. How come this change isn't included in v21.02.2 or even openwrt-21.02 as of right now:

```
joker:~/rpi-4$ git status | head -1
HEAD detached at v21.02.2
joker:~/rpi-4$ cat target/linux/ramips/dts/mt7628an_onion_omega2p.dts
#include "mt7628an_onion_omega2.dtsi"

/ {
	compatible = "onion,omega2p", "onion,omega2", "mediatek,mt7628an-soc";
	model = "Onion Omega2+";
};

&firmware {
	reg = <0x50000 0x1fb0000>;
};

marvin:~/openwrt$ git status | head -1
On branch openwrt-21.02
marvin:~/openwrt$ cat target/linux/ramips/dts/mt7628an_onion_omega2p.dts
#include "mt7628an_onion_omega2.dtsi"

/ {
	compatible = "onion,omega2p", "onion,omega2", "mediatek,mt7628an-soc";
	model = "Onion Omega2+";
};

&firmware {
	reg = <0x50000 0x1fb0000>;
};
```

pepe2k, 2022-03-28:
@jsantala you are correct here, this should be backported, let me handle that!
@mpratt14 are you fine with backporting https://github.com/openwrt/openwrt/commit/74516f4357 to 21.02 (it doesn't apply cleanly so has to be manually backported)?

jsantala, 2022-03-28:
Thank you very much!

balou1974, 2022-04-06:
Bug #9636 looks like the same problem. No even u-boot will start after executing a reboot.

pepe2k, 2022-04-11:
Should be fixed in 169c9e3a88b7d4521a797864540657ef33554663.

closed 2022-04-11 by pepe2k

mpratt14, 2022-04-12:
thanks for backporting it, sorry I didn't reply earlier 😃