openwrt/openwrt #9550: [Q] partition setup: UBI, mtd-concat, U-Boot, FIT

Lanchon, 2022-03-25:
hi, thanks for the help!

i'm porting to [this device](https://openwrt.org/inbox/toh/arris/tr4400_v2) and i've got some questions regarding how to best layout the flash. 256 MB nand, u-boot, configurable persistent u-boot environment, kernel-in-UBI. the stock partitioning interleaves usable space with partitions best left alone.

1. i noticed i can use mtd-concat to concat disjoint mtd partitions into a virtual mtd and use that for UBI. but what about u-boot? it is possible to mtd-concat in u-boot so that i can boot a kernel stored in a concat UBI?
2. i'm currently using kernel,rootfs volumes in UBI. is this the recommended way? i'm thinking a combined image volume -such as FIT- might be better asi it might benefit from integrity checking (CRC) of the whole thing instead of just the kernel (let alone the simpler management). my u-boot supports FIT.
3. right now, sysupgrade creates/updates the kernel,rootfs volumes, then creates rootfs_data if needed. but it creates it using all the remaining space in the UBI device. this means that firmware upgrades will fail if the firmware grows (unless settings are wiped, i suppose). what is the correct way to limit the size of rootfs_data or otherwise reserve free UBI space? pre-creating the volume? would that strategy survive a settings wipe?
4. is there a standard way to store a recovery initramfs image to recover from a bad flash?
5. is there a standard way to do safe flashing? by that i mean space for 2 firmwares and atomic switching from one to the other being active after flashing a new firmware. UBI supports that kind of thing.

feel free to answer whatever isolated question you might be able to help with. thanks again!

dangowrt, 2022-03-25:
1. Afaik there is no implementation of mtd-concat in U-Boot, and I'm sure there is no such support in vendor U-Boot v2012.07.
2. Unfortunately U-Boot v2012.07 will not support having the rootfs included in the uImage.FIT, because that requires support for *`external data`* FIT images only supported since v2018.01. Hence the best you can do is to at least store them both in UBI and make the best of what the vendor loader has to offer (loading kernel from UBI is already great for U-Boot v2012.07!). Apart from the rootfs you can still use uImage.FIT for kernel and device-tree blob.
3. The current behavior is not a problem, as (just like for any other type of rootfs_data filesystem) the rootfs_overlay when you do sysupgrade, and only specific files are stored in RAM and then put into the newly created overlay volume. If you still want to limit the size of the rootfs_data volume you may do so by setting a U-Boot env variable (which is ignored by U-Boot in that case but used by OpenWrt sysupgrade): e.g. `fw_setenv rootfs_data_max 0x2000000`
4. There is a multitude of ways to create dual-boot setups, all of them depend on support from the bootloader. So either you go with what the vendor stock bootloader supports or you have to build U-Boot from source. In case of your target board, it looks like there is some kind of whole-flash A/B dual-boot scheme, with all the backup volumes having a `_1` suffix. Maybe try to make use of that somehow? If you are up to build U-Boot from source your options are of course less limited, then you can do it all by [scripting](https://github.com/openwrt/openwrt/blob/master/package/boot/uboot-mediatek/patches/410-add-linksys-e8450.patch#L380).
5. You may want to take a look at [luci-app-advanced-reboot](https://docs.openwrt.melmac.net/luci-app-advanced-reboot/). This can very well work with two UBI devices defined in two large MTD partitions on the same flash chip.
6. I guess at this point we don't have a standard way for having only rootfs_data volume in a UBI device of it's own, but I think this is a valid use-case and patches to add this to `/lib/upgrade/nand.sh`, fstools, ... are very welcome, I will review and help if needed.

Lanchon, 2022-03-25:
@dangowrt thank you so much for taking the time to write all that! and wow, you do know everything OpenWrt!

these bullets sort of criss-cross each other, so maybe it is best to read them all first.

---

3) rootfs_data during sysupgrade:

i'm shocked by this! i assumed rootfs_data was left alone during sysupgrade. this is no doubt a fallout from when rootfs_data was always jffs2 right after firmware. i thought there was some padding inbetween those two not shown in the docs, lol!

is there no way around this? can openwrt be configured not to do this? this is not only inefficient (wastes flash life) but dangerous: you can loose the most valuable data during an upgrade... yours. this also cripples my plans for safe upgrades: my planned safe upgrades need all updated partitions to live in the same UBI. but the kernel needs updating, so the UBI cannot be concat'ed... so it has to be a small UBI (80 MB max in my case)... and then only use half of it for the 3 partitions... :(

i wanted to use a 64 MB UBI for dual kernel and dual rootfs, and a 144 MB concat UBI for a single rootfs_data.

now i don't see the point of a large rootfs_data if it can't be trusted to survive upgrades.

---

4) recovery initramfs:

i don't have time for JTAG and u-boot, i have to work with the stock BL. as i understand it:

in my device the A/B scheme is in action all the way up to uboot. i works in case of flash corruption, not upgrades (me thinks); i wont touch that. then there is a single uboot env. but the stock env is not programmed, defaults are used on every boot, so the env is not a single point of failure in stock fw.

then comes qualcomm's (proprietary?) boot. maybe it counts failed boots somewhere and flips back and forth among XXX and XXX_1 partitions. but i won't use that command, i will use regular bootm instead.

i was thinking of a small mtd partition (UBI volume works too, but i rather have the recovery outside) having a recovery initramfs partition. if the normal chain of commands leading to bootm fails, then the recovery chain will be executed. (and if that fails, then TFTP boot will be executed.)

to fail, bootm needs to fail the CRC check (or the UBI volume is missing). so a specific sequence needs to be in place during sysupgrade to account for rootfs not being CRC'd: truncate/delete kernel volume, sync, write rootfs volume, sync, write kernel.

the recovery image is already on my router. i wanted to see if there was a standard way of doing this; seems not.

---

5) safe flashing:

i briefly looked at luci-app-advanced-reboot. it is not what i meant. it is interesting, but in my case it would require to let an 80 MB partition go and use 2 x 64 MB partitions for the system plus rootfs_data. (maybe i could mount the remaining 80 MB partition as UBIFS on, say, /opt.) also it seems to me that the fw_env needs to be written on every upgrade (boot switch). i discarded this idea when i was thinking of safe flashing because it is dangerous: not only flashing ceases to be safe if the env is written, but also there are not a lot of erase blocks allocated to it (only 4, and you could have a factory defect there or 2 already) so it's creepy.

i was thinking of a mechanism that would be stock and automatic and safe. maybe openwrt does not have it already, IDK, but it seems very simple to implement: write kernel_new and rootfs_new (and rootfs_data_new if that can't be avoided), then rename the 2(3) UBI volumes in a single atomic operation. UBI supports this precisely for firmware updates. it does it so well that you can even rename in-use volumes. you can atomically rename and delete the old ones, or atomically rename and rename the old ones (XXX_old).

you can also have XXX_A/XXX_B partitions and swap-rename them on upgrade, and have uboot first try A then B. but this complicates matters because now you have to communicate to userland which slot was actively booted and have sysupgrade write on the inactive... and it all seems completely pointless.

anyway, my idea was to have a non-concat 64 MB UBI with safe flashes of kernel/rootfs (32 MB max) using UBI rename, and a concat'ed 144 MB UBI for rootfs_data. this all looks grim now that i know that rootfs_data participates in sysupgrade :(

---

6) separate UBI for rootfs_data:

i could do it, but seems pretty pointless to have a large rootfs_data if there is no way not to rewrite it on every sysupgrade. i rather have a safe sysupgrade using UBI rename for the 3 partitions in a single 80 MB UBI (40 MB max for the 3), than a large rootfs_data that could be lost on any upgrade. atomic UBI renames cannot span UBI devices, and uboot cannot boot from concat UBIs... so we are SoL.

dangowrt, 2022-03-26:
> 3. rootfs_data during sysupgrade:
>
> i'm shocked by this! i assumed rootfs_data was left alone during sysupgrade. this is no doubt a fallout from when rootfs_data was always jffs2 right after firmware. i thought there was some padding inbetween those two not shown in the docs, lol!

The size of rootfs data is impossible to predict, as users may choose to include different packages using custom builds or the ImageBuilder. Hence the dynamic allocation of the free space, not just on NOR flash / JFFS2.
Re-creating `rootfs_data` after sysupgrade is deeply embedded into many places in OpenWrt, as `rootfs_data` is really just intended to store configuration (ie. UCI config plus packages may declare files to be saved accross sysupgrade). Users may add additional paths by using `/etc/sysupgrade.conf`. This is, obviously, limited by the size of `tmpfs` and ultimately by the amount of RAM available. Replacing kernel+rootfs while whiping rootfs_data on sysupgrade while only keeping selected files also has several advantages -- while it may have initially been due to space limitations it turned out to be useful in many ways. Sure, it might have unforeseen consequences for users used to normal r/w-rootfs Linux rolling-release distributions, but I guess that's a matter of improving OpenWrt documentation (I also fail to find a comprehensive summary introducing the basic concepts of OpenWrt as a distribution anywhere on our web resources...)

For any larger persistent storage needs you should use additional volumes -- I wrote a tool for that called [`uvol`](https://github.com/openwrt/packages/tree/master/utils/uvol) which allows you to do that easily (and it works transparent on top of LVM2 or UBI). In case of devices with NAND flash, you should set `rootfs_data_max` in the U-Boot environment to make sure there is enough free space available to initially create the additional volume(s).
In case of your device which may end up with several UBI devices, I may need to improve the logic regarding which UBI device to choose for `uvol` (the largest :smiley:)...

> i was thinking of a small mtd partition (UBI volume works too, but i rather have the recovery outside) having a recovery initramfs partition. if the normal chain of commands leading to bootm fails, then the recovery chain will be executed. (and if that fails, then TFTP boot will be executed.)

You may try to script that with the vendor loader. Relying only on the success of `bootm` without checking the integrity of the rootfs as well may easily lead to a boot loop which you have to prevent somehow using what ever the stock loader has to offer...

> the recovery image is already on my router. 

Sure, this is how I've implemented it in U-Boot for the Linksys E8450/Belkin RT3200:
Have initramfs image for recovery stored in UBI and resort to TFTP if that's broken.

> i wanted to see if there was a standard way of doing this; seems not.

Well, I was hoping to open a discussion and implement this for targets where we are building up-to-date U-Boot from source (ie. done for mediatek; next could be using the same semantics for rockchip, sunxi, ...). But it's there (in U-Boot) where things have to be implemented -- as almost none of the vendors get it right, I thought it may be a good idea to at least provide some examples of how to do things nicely (ie. with zero flash writes during boot, relying on pstore to detect crashes, using modern uImage.FIT capabilities, ...).

> 5. safe flashing:
> 
> i discarded this idea when i was thinking of safe flashing because it is dangerous: not only flashing ceases to be safe if the env is written, but also there are not a lot of erase blocks allocated to it (only 4, and you could have a factory defect there or 2 already) so it's creepy.

You are absolutely right about that. If U-Boot environment is written frequently it has to be redundant and live in UBI as well. Unfortunately it's not done like that by many vendors.

Regarding your ideas for atomic A/B dualboot using UBI volume rename: Yes, this is an interesting idea for devices where UBI is supported by the stock loader and indeed rather easy to implement. With commit 93259e8ca261c7965618fe11c2d385638da5cfa6 it should also be possible to implement detecting of current `rootfs_data` content belongs to the running firmware (or got to be wiped).

> 6. separate UBI for rootfs_data:
>
> i could do it, but seems pretty pointless to have a large rootfs_data if there is no way not to rewrite it on every sysupgrade. i rather have a safe sysupgrade [...]

Having a larger UBI device can also be useful for user-managed persistent volumes other than `rootfs_data`, see my remark about `uvol` as a possible way to use/manage it (you may of course also just use ubimkvol and friends directly).

Lanchon, 2022-03-26:
thanks!!

> Relying only on the success of bootm without checking the integrity of the rootfs as well may easily lead to a boot loop

yes, if flash gets corrupted after successful programming (rare). but interrupted flashes could be detected:

> truncate/delete kernel volume, sync, write rootfs volume, sync, write kernel

or better:
```
rm-vol kernel_new rootfs_new rootfs_data_new
if (!safe_sysupgrade) rm-vol kernel rootfs rootfs_data; sync?
mk-vol kernel_new
[...]
sync?
rename-vol *_new -> *
```

i think some strategy similar to these should be implemented for all devices if possible, and i favor a variation of the second as it allows safe upgrades.

---

so to wrap up:

i got these usable real partitions: 64 MB, 64 MB, 80 MB (and 8 MB, 1 MB, 2 MB):
```
[    1.803842] 0x000000000000-0x000000040000 : "0:SBL1"
[    1.810930] 0x000000040000-0x000000180000 : "0:MIBIB"
[    1.819310] 0x000000180000-0x0000002c0000 : "0:SBL2"
[    1.827320] 0x0000002c0000-0x000000540000 : "0:SBL3"
[    1.834944] 0x000000540000-0x000000660000 : "0:DDRCONFIG"
[    1.843184] 0x000000660000-0x000000780000 : "0:SSD"
[    1.851010] 0x000000780000-0x000000a00000 : "0:TZ"
[    1.862091] 0x000000a00000-0x000000c80000 : "0:RPM"
[    1.873377] 0x000000c80000-0x000001180000 : "0:APPSBL"                         5 MB
[    1.891176] 0x000001180000-0x000001200000 : "0:APPSBLENV"                    0.5 MB
[    1.897889] 0x000001200000-0x000001340000 : "0:ART"
[    1.906103] 0x000001340000-0x000005340000 : "rootfs"             64 MB
[    2.026199] 0x000005340000-0x0000053a0000 : "0:BOOTCONFIG"
[    2.033059] 0x0000053a0000-0x0000054e0000 : "0:SBL2_1"
[    2.041514] 0x0000054e0000-0x000005760000 : "0:SBL3_1"
[    2.049552] 0x000005760000-0x000005880000 : "0:DDRCONFIG_1"
[    2.056518] 0x000005880000-0x0000059a0000 : "0:SSD_1"
[    2.064463] 0x0000059a0000-0x000005c20000 : "0:TZ_1"
[    2.072265] 0x000005c20000-0x000005ea0000 : "0:RPM_1"
[    2.088589] 0x000005ea0000-0x000005f00000 : "0:BOOTCONFIG1"
[    2.095149] 0x000005f00000-0x000006400000 : "0:APPSBL_1"
[    2.112002] 0x000006400000-0x00000a400000 : "rootfs_1"           64 MB
[    2.251371] 0x00000a400000-0x00000a500000 : "fw_env"                           1 MB
[    2.258742] 0x00000a500000-0x00000ad00000 : "config"              8 MB
[    2.281909] 0x00000ad00000-0x00000af00000 : "PKI"                 2 MB
[    2.291273] 0x00000af00000-0x00000b000000 : "scfgmgr"             1 MB
               0x00000b000000-0x000010000000 -not used in stock-    80 MB

erase blocksize: 128 KB
```

my options:
- the 3 volumes in 80 MB, dual boot with stock rom.
- 2 vols in 80 MB, 91 MB, or 64 MB:
  - rootfs_data in concat UBI of remaining (128/144 MB).
- 3 vols in 80 MB, 91 MB, or 64 MB:
  - extra volume (pesists through sysupgrades) in concat UBI of remaining (128/144 MB).
  - acts like a pendrive. mount it somewhere? run init hooks in extra vol?
- dual openwrt installs, each in 64MB, using advanced-reboot.
  - extra 80/91MB vol?

safe upgrades:

- when the 3 vols are in the same ubi:
  - we can do really safe upgrades by rename.
  - limits total space to half (max 32/40/45 MB).
- else when 2 vols are in the same ubi:
  - we can do semi-safe upgrades by rename, accepting loss of configuration.
  - user required to enter failsafe and erase rootfs_data.
  - limits sysupgrade image size to half (max 32/40/45 MB).

too many options. i've zero user experience with openwrt. what do you recommend?

thanks again!

Lanchon, 2022-03-29:
@dangowrt sorry to bother you one last time. you have no recommendation?

dangowrt, 2022-04-01:
I can't really answer that for you as it depends on your priorities which way you want to go.
As for OpenWrt, having less available space for the sake of having more reliability and robustness is the way to go.
So ideally you should support the vendor dual-boot scheme and use two UBI devices, each holding kernel+rootfs+rootfs_data (64MB for each is absolutely enough). Users who want additional space may setup an additional UBI device in the space unused by the vendor firmware.
As the vendor firmware makes use of several UBI devices on different MTD partitions, a safe-upgrade scheme using `ubirename` -- as great as the idea may be -- would further reduce the amount of available space, as then both old and new firmware would have to be stored in the same UBI device.
Implementing support for such a ubirename safe-upgrade scheme would be a welcome addition to OpenWrt anyway, let me know if you plan to work on that and/or I can support you with that somehow.

Regarding this particular devices: Unless you are rethinking your decision not to touch/replace the stock bootloader, I guess you are down to doing it more or less like the stock firmware would do as well.

Lanchon, 2022-04-01:
@dangowrt thanks D!

well i thought you werent showing up so i went ahead and implemented my choice.

installation:
- external backup of everything.
- the `fw_env` partition is moved (copied). (fw_env contains mac addrs, default ssid, default password.)
- a `recovery` volume is added to `rootfs` with my initramfs build. (the rootfs mtd will be called `extra` by my kernel.)
- uboot's env is updated to try boot ubi/kernel, then try extra/recovery, then try tftp.
- (up to this time stock can still be booted.)
- reboot. no ubi/kernel, so extra/recovery starts. it expects fw_env at its new location. it also sees `ubi` (155 MB mtd, no concat) and `extra` (64  MB, unused). extra can be used via your package uvol. extra will keep the recovery volume indefinitely unless the user wipes it.
- sysupgrade to `ubi`. formatting of ubi will wipe the old fw_env. the 3 openwrt partitions end up in ubi.
- all original partitions are still available with prefix `stock_*` in their names in case user wants to roll back.

safety:
- i don't trust the OEM firmware to dual boot past uboot so i won't use that command.
- safety comes in two ways: recovery/tftp and safe updates. both require changes to openwrt:
  - at a minimum ensure failing the kernel CRC before touching rootfs and until the new rootfs is completely written.
  - plus two levels of safe upgrades via ubirename.

> Implementing support for such a ubirename safe-upgrade scheme would be a welcome addition to OpenWrt anyway, let me know if you plan to work on that and/or I can support you with that somehow.

yes! i plan to try implement these 3 levels of safety later in stages. i hope they are feasible.

thanks!

Lanchon, 2022-04-01:
@dangowrt thanks!! opened pull request:
https://github.com/openwrt/openwrt/pull/9623

closed 2022-04-01 by Lanchon

Lanchon, 2023-10-18:
@dangowrt,

> Re-creating `rootfs_data` after sysupgrade is deeply embedded into many places in OpenWrt, as `rootfs_data` is really just intended to store configuration [...] Sure, it might have unforeseen consequences for users used to normal r/w-rootfs Linux rolling-release distributions, but I guess that's a matter of improving OpenWrt documentation (I also fail to find a comprehensive summary introducing the basic concepts of OpenWrt as a distribution anywhere on our web resources...)

long time Daniel!

that little thing, the missing docs, sure bit me when i started using openwrt (to port it to a new device, and you helped a lot then).

today i spent the better part of the day writing a bit of docs for openwrt.

it is all in subsections of this page: https://openwrt.org/docs/techref/flash.layout

- https://openwrt.org/docs/techref/flash.layout#nor_flash_vs_nand_flash: rewrote to explain how raw NAND flash defects are accounted for (previous text was just all wrong).

- https://openwrt.org/docs/techref/flash.layout#partitioning_of_nor_flash-based_devices: fixed the layout and text and noted that this applies to NOR devices only.

- https://openwrt.org/docs/techref/flash.layout#sysupgrade_and_rootfs_data: new section to document how sysupgrade works and what it does to rootfs_data and why.

- https://openwrt.org/docs/techref/flash.layout#partitioning_of_nand_flash-based_devices: new section to document flash layouts of typical NAND devices.

- https://openwrt.org/docs/techref/flash.layout#reserving_ubi_partition_space_for_user-defined_ubi_volumes: new section to document `rootfs_data_max`.

- https://openwrt.org/docs/techref/flash.layout#examplecreating_a_ubi_volume_for_persistent_storage_across_sysupgrades: new section to document you `uvol` package.

so at least now you got a bit of text explaining some basic concepts of openwrt, and also a usage example for `uvol`.

thanks again for all your help during porting!

(oh btw, i'm finally stepping up my use of openwrt, so i might ask the odd question now and then :) )