openwrt/openwrt #9552: how to remove X-Frame-Options: SAMEORIGIN response in luci

0neday, 2022-03-26:
I am using iframe on homeassistant recently, and I found this error 
[image]

I check openwrt, and found luci response header
 [image]

so, how to remove  X-Frame-Options?

0neday, 2022-03-26:
edit `/usr/lib/lua/luci/http.lua` and commit x-frame code 

```
                    --  if not context.headers["x-frame-options"] then
                    --          header("X-Frame-Options", "SAMEORIGIN")
                    --  end

```
, and then work fine.

closed 2022-03-26 by 0neday

TheRealMaN, 2023-06-13:
removed X-Frame-Options, but still can't get it to work inside iframe, Authorization form appeared, but
after I hit "login" button" with username and password, login page just reloads and doesn't actually enter to router interface, in Chrome network tab I get 403 error form path "luci/", without iframe, all works right