openwrt/openwrt #9555: ARP requests are not forwarded to Wifi Client after some time --> devices unreachable

JsBergbau, 2022-03-26:
OpenWrt 21.02.2, r16495-bf0c965af0 with WR802n v4 https://openwrt.org/toh/tp-link/tl-wr802n_v4

I have three of these devices and after some time, rarely a few hours, sometimes a day, mainly after a few days Wifi devices aren't reachable anymore. 

Trying to ping from 192.168.178.200

```
ping 192.168.178.51
PING 192.168.178.51 (192.168.178.51) 56(84) bytes of data.
From 192.168.178.200 icmp_seq=1 Destination Host Unreachable
From 192.168.178.200 icmp_seq=2 Destination Host Unreachable
From 192.168.178.200 icmp_seq=3 Destination Host Unreachable
```
On the AP `tcpdump arp` is running.

```
08:36:55.403719 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:36:56.443735 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:36:57.483780 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:36:58.523732 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:36:58.981987 ARP, Request who-has 192.168.178.89 (xxxxxxxxxx) (oui Unknown)) tell 192.168.178.20, length 46
08:36:58.982060 ARP, Reply 192.168.178.89 is-at b0:95:75:4a:5b:78 (oui Unknown), length 28
08:36:59.563711 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:37:00.159145 ARP, Request who-has 192.168.178.73 tell 192.168.178.39, length 46
08:37:00.213415 ARP, Request who-has 192.168.178.192 tell 192.168.178.39, length 46
08:37:00.603885 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:37:01.204425 ARP, Request who-has 192.168.178.73 tell 192.168.178.39, length 46
08:37:01.284549 ARP, Request who-has 192.168.178.192 tell 192.168.178.39, length 46
08:37:01.374026 ARP, Request who-has 192.168.178.60 tell 192.168.178.20, length 46
08:37:01.643786 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:37:01.980251 ARP, Request who-has 192.168.178.60 tell 192.168.178.20, length 46
08:37:02.244502 ARP, Request who-has 192.168.178.73 tell 192.168.178.39, length 46
08:37:02.324467 ARP, Request who-has 192.168.178.192 tell 192.168.178.39, length 46
08:37:02.683752 ARP, Request who-has 192.168.178.51 tell 192.168.178.200, length 46
08:37:02.971502 ARP, Request who-has 192.168.178.60 tell 192.168.178.20, length 46
```

Then doing `sudo arp -s 192.168.178.51 <MAC of .51>` after that 
```
 ping 192.168.178.51
PING 192.168.178.51 (192.168.178.51) 56(84) bytes of data.
64 bytes from 192.168.178.51: icmp_seq=1 ttl=255 time=24.8 ms
64 bytes from 192.168.178.51: icmp_seq=2 ttl=255 time=4.27 ms
64 bytes from 192.168.178.51: icmp_seq=3 ttl=255 time=4.11 ms
64 bytes from 192.168.178.51: icmp_seq=4 ttl=255 time=3.85 ms
64 bytes from 192.168.178.51: icmp_seq=5 ttl=255 time=4.30 ms
```

This is not a problem with the device .51. This is only an example. Different devices are affected by this problem.

JsBergbau, 2022-04-04:
Now some problem occured again 192.168.178.85 was not reachable. I used tcpdump on that device that was not reachable and I've seen that arp requests reach the device via broadcast, so also arp requests from other devices. Own arp requests were replied imediately. So it seems the ARP reply is lost on the way from wifi device via OpenWRT router to network.

thmang82, 2022-08-02:
I have exactly the same problem in my network with a Cudy WR2100, MT7621 chipset, MT7603E Wifi.
Yours has a MT7628N chipset, so maybe it's related to the MediaTek MT76 drivers?

I had this problems basically since I bought it about 2 month ago. In random intervals, sometimes hours, sometimes a few day, one or two of the Wifi clients will become unreachable in the network. The Lucy UI lists the clients as connected, but you cannot ping them from another device. On my MacBook, when I list the arp table with "arp -i en0 -l -a", those devices have no Link Layer address in their entry. The devices in question think they are connected, but also cannot communicate. It's pretty wired.
I am already re-starting the router every night automatically, but it's still happening from time to time.
It's really a shame. The hardware is quite good in terms of range and power, but those stability issues are really an issue.

Did you found a solution?

PS: I am on OpenWrt 21.02.3, r16554-1d4dea6d4f

JsBergbau, 2022-08-02:
Saidly I have no solution so far. When I notice the problem, I connect via ssh execute `wifi` and it works again. But still this is very annoying.

thmang82, 2022-08-03:
Sorry to hear that you still have those issues, too.

It happened again today (as almost every day). I logged into the router in order to check if I can find something in the logs that indicates that the unreachable device was "lost". But there was nothing. ARP table looked good, nothing in dmesg, nothing in logread. May be logread output was too short, I am not sure how long the device was unreachable.

Anybody has an Idea why this happens? Is it related to the MT76 drivers/chipset? Or something different?

JsBergbau, 2022-08-03:
It is quite a long time ago, but I also couldn't find anything in the logs when this problem happens.

thmang82, 2022-08-11:
A few days ago I changed the "Time interval for rekeying GTK" to 0, because I read a post from a guy who said it helps with MT76 connectivity issues.
 
But unfortunately, it happened again today with a Raspberry Pi CM4 that is located 2 meters away from the router :-(

There is an always open websocket connection from that device to a wired server, so I have a pretty good indication when the disconnect happened from the server logs:
10:45:04.879  ERROR: Websocket [ v0cX5BvJ ] stalled!

Interesting with that regard is the logread output from the router:

Thu Aug 11 10:32:00 2022 daemon.info dnsmasq-dhcp[3588]: DHCPREQUEST(br-lan) 192.168.1.156 e4:5f:01:91:xx:xx
Thu Aug 11 10:32:00 2022 daemon.info dnsmasq-dhcp[3588]: DHCPACK(br-lan) 192.168.1.156 e4:5f:01:91:xx:xx cm4test3
Thu Aug 11 10:59:06 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx
Thu Aug 11 11:04:07 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx
Thu Aug 11 11:10:37 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx
Thu Aug 11 11:15:56 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx
Thu Aug 11 11:21:01 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx
Thu Aug 11 11:26:09 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx
Thu Aug 11 11:34:40 2022 daemon.notice hostapd: wlan0: AP-STA-POLL-OK e4:5f:01:91:xx:xx

Why are they API-STA-POLL-OK messages after the client became unreachable? Even 30 Minutes later there are AP-STA-POLL-OK logs.

Still I cannot ping the 192.168.1.156 IP any more!
Also, the IP and the MAC address are gone from "arp" command output and from arp scan via 'arp-scan --interface=br-lan --localnet -t 3000 -x -q'
The Raspberry Pi still believes it is connected. "ifconfig" returns an IP for the Wifi device!

This is definitely strange. I am unfortunately no expert regarding the hostapd logs. Can some of the developer help here?

anfractuosity, 2022-11-06:
I think I'm also getting a similar issue to this, I use a UniFi6 Lite with Openwrt.  Also after a few days or less, I become unable to ping a certain host via wifi.

I found if I connect to the OpenWrt router, I can ping the host from the router itself though.

```
pinging from laptop -----> openwrt router/AP --------> raspberry pi can fail
```

But then if I do -

```
ssh from openwrt router/AP ---> raspberry pi
And
ping from raspberry pi --> laptop
```

Then
```
pinging from laptop ----> openwrt router/AP -------> raspberry pi can work
```

wlanrouter, 2023-01-19:
Same issue here:
TP-Link Archer C7 v4 (Qualcomm Atheros QCA956X)
OpenWrt 22.03.2 r19803-9a599fee93

Multiple devices frequently can't reach raspberry pi 3B
```
pinging from laptop ---> openwrt router/AP ---> raspberry pi = fails
pinging from smartphone ---> openwrt router/AP ---> raspberry pi = works
```
Typing `wifi `solves it.

next day
```
pinging from laptop ---> openwrt router/AP ---> raspberry pi  = works
pinging from smartphone ---> openwrt router/AP ---> raspberry pi = fails
```

On Router: `tcpdump arp`
```
07:53:07.566272 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has raspberry.lan tell laptop.lan, length 46
07:53:08.572867 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has raspberry.lan tell laptop.lan, length 46
07:53:09.563646 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has raspberry.lan tell laptop.lan, length 46
07:53:10.565373 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has raspberry.lan tell laptop.lan, length 46
07:53:11.570352 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has raspberry.lan tell laptop.lan, length 46
```

On Raspberry: `tcpdump arp`
```
nothing
```

Of course just for the device which can not connect. On all other arp works fine,

khanjui, 2023-01-30:
Same problem here, wireless devices (both 2.4 GHz MT7603E and 5 GHz MT76x0E) become unreachable / can't reach LAN or WAN from them even though they show up as connected. It happens very randomly, sometimes once in a week and sometimes few times per day. This doesn't happen on wired devices. Using Xiaomi Mi 4A Gigabit Edition latest snapshot.

fluboi, 2023-02-19:
I see the same issue with :
- MikroTik hAP ac2 (5Ghz, Qualcomm Atheros IPQ4018) 
- TP Link Archer C6 v2 (5Ghz, Qualcomm Atheros QCA9886)

OpenWRT 22.03

The station says it is connected. OpenWRT shows it in Luci. 
Tcpdump shows that packets from the Station to the AP are correctly forwarded. But Packets from AP to Station are not. 

Running `wifi` command on the OpenWRT AP temporarily solve the connectivity issue.

The problem seems to appear more often when the station is in a poorly covered area. Once the problem is present, even if we move right next to the AP, the connectivity remains broken.

JsBergbau, 2023-04-27:
> Cable bound devices can reach wifi devices at all times over IPv4 and vice versa but wifi devices are not always able to reach each other. 

In my case also cable bound devices can't reach wifi devices. However what I can confirm, that it seems not to affect all devices, but maybe thats because some devices have the arp entry cached and thats why some devices are still working.

Hope this issue will be resolved soon, since it is really annoying.

F-i-f, 2023-05-06:
I'm seeing the same problem on a [Dynalink DL-WRX36](https://dynalink.life/products/dynalink-wifi-router-wrx36) running [OpenWRT snapshot](https://openwrt.org/toh/dynalink/dl-wrx36) r22699-45f5115253.
Ethernet broadcast does not seem to work "sometimes".

Here's my set-up:
I have 3 DL-WRX36 functioning in dumb AP mode.
They are all bridged together for roaming but none of 802.11r, 802.11k or 802.11v are configured.
All 3 are configured on different 2.5 GHz and 5 GHz channels and have different 802.11ax colors.
I have a rather complex set up with VLANs and use new-style VLANs (bridging the untagged interface).
All three units have exactly the same configuration, they all get their address via DHCP, and the /etc/config/* files are all the same, except for hostnames and wifi channels and colors.

Now to the weird behavior:
I've noticed that while the station to AP traffic always works perfectly fine, station to station traffic sometimes didn't work. I narrowed it down to ARP traffic not flowing properly.
After investigation, I've found out that on stations connected to the 5 GHz band of one of the three units, Ethernet broadcasts were not being received. If Ethernet broadcasts are not working, then ARP doesn't work, and that explains why station-to-station traffic does not work.
To confirm the problem was Ethernet broadcast, if I add static ARP entries to the stations, then station-to-station traffic works fine.

I started investigating the problem with broadcasts. This is how I test:
The router behind the bridged/switched APs runs a ping to the wireless LAN broadcast address.
Tcpdump on all three APs see the broadcast ping arriving on the ethernet port.
Tcpdump on all three APs see the broadcast ping being emitted on both wireless LANs (2.5 GHz and 5 GHz) bridged to the ethernet port.
All the stations connected to two of the APs received the broadcast.
But on the last "problem" AP, the broadcast is only received on the stations connected to the 2.5 GHZ wireless network. Any station connected to the 5 GHz network never receives the broadcast ethernet frame.

I've also tried with multicast traffic. It does goes through.
I've tried changing channels on the problem AP, it does not help.

I'm wondering if the issue couldn't be worked around with [proxy_arp_pvlan](https://serverfault.com/questions/212074/is-it-possible-to-use-proxy-arp-back-to-the-same-interface).

rany2, 2023-05-18:
This issue also impacts me on MT76x02 and MT7915.

F-i-f, 2023-05-20:
> I'm wondering if the issue couldn't be worked around with [proxy_arp_pvlan](https://serverfault.com/questions/212074/is-it-possible-to-use-proxy-arp-back-to-the-same-interface).

The ping between hosts can be made to work by setting `proxy_arp_pvlan` on the ethernet interface bridged to the wifi interface showing the lost traffic. However that does not solve the more general problem of broadcast/multicast traffic not being sent back and that breaks lots of L2 protocols (avahi/mdns for one).

rany2, 2023-05-29:
I resolved my issue by switching from "FT over the Air" to "FT over the DS." Works properly now.

JsBergbau, 2023-05-30:
This setting should have only an effect, when Fast Roaming is enabled. I don't have fast roaming enabled and still have the problem.
Maybe it is a coincidence? Sometimes this error only occurs every few weeks, sometimes multiple times per day.

rany2, 2023-06-01:
@JsBergbau Indeed it appears to be a coincidence, though switching to FT over DS fixed an unrelated problem for me which I presume is related to low AP density in some areas.

rany2, 2023-06-01:
@JsBergbau Do you have WNM sleep mode enabled, by any chance?

JsBergbau, 2023-06-02:
The checkbox for "WMM Mode" is checked.

rany2, 2023-06-02:
@JsBergbau No, WMM Mode should always be enabled. At any rate it wasn't the cause, I worked around the issue for the time being by enabling multicast_to_unicast (it's multicast_to_unicast_all in newer OpenWRT versions).

larena1, 2023-06-02:
@nbd168 do you maybe have an idea about this issue that affects so many people and could be kernel related?

tperalta82, 2023-06-02:
Having exactly the same issue, but on a Xiaomi AX3600, which is not a mediatek device, but a Qualcomm IPQ807! Running wifi seems to fix it, or adding fixed arp tables on the devices (which we all want to avoid) , and it's pretty random!

larena1, 2023-06-03:
> I worked around the issue for the time being by enabling multicast_to_unicast (it's multicast_to_unicast_all in newer OpenWRT versions).

@rany2 https://github.com/openwrt/openwrt/commit/09ea1db93b53d2c1e4a081f20fbbddd4bffd451d

This is the commit that renamed the flag and the commit message is quite interesting, especially this bit

> It also has been observed to cause ARP lookup failure between stations
connected to the same interface.

So enabling the flag you mentioned can lead to ARP issues, but with the flag not enabled, ARP and multicast issues can also be observed?

I think this issue would definitely need to be examined by an experienced dev, who is familiar with the Kernel and its mac80211 subsystem.

rany2, 2023-06-03:
@larena1 It depends on the client behavior, if it works with all your clients there should be no cause for concern. It won't stop working out of nowhere

larena1, 2023-06-03:
@rany2 I get that. Just thought it's interesting that enabling the flag can also cause ARP issues while lots of people are experiencing these as well with the flag disabled.

For unknown reasons, after a while the kernel seemingly just stops sending multicast traffic received from stations to other stations connected to the same interface. I wonder why not everybody is having this or maybe everybody is but not everybody notices it as it only affects connectivity between stations?

larena1, 2023-06-30:
I'm still seeing this issue on the Qualcomm AX platform with a snapshot from just a few days ago so unfortunately it's still unsolved. It usually occurs immediately after the device booted until running `wifi` twice. After that it'll randomly occur when a device connects back to the station, like a phone. No WiFi devices reachable then and no IPv6 too. Enabled the multicast to unicast option but that didn't make a difference either. Really big issue 😐

larena1, 2023-06-30:
This is extremely weird. I have 2 stations, one 2g and one 5g.

Both stations enabled on boot: 2g will have multicast, meaning IPv6 and ARP works properly, 5g will not
2g disabled on boot: 5g will have multicast, 2g will also have multicast when enabled after

So something doesn't go right and apparently only the first station to come up will work properly.

When 5g fails and my TV and mobile phone are both connected to it, they can't reach other. With one device connected to each, they can.

Kernel code for packet relaying looks fairly simple. Not sure if and what could go wrong there

https://github.com/torvalds/linux/blob/master/net/mac80211/rx.c#L2646

damige, 2023-07-05:
I have been struggling with this issue for a while. Tried all the mentioned work arounds. 23.05 does not fix it either.
My platform is ramips/mt7621. Judging by the amount of platforms that have this problem this might be a more generic openwrt issue. I also had this problem before with my Archer C7 routers. But not from the start. Might have become an issue after 22.03 (when that platform got DSA) [https://openwrt.org/docs/guide-user/network/dsa/start](url)

thmang82, 2023-07-16:
@damige That does not sound too good regarding the "kernel: remove bridge offload hack" and 23.05 :-(

I really hope this will be fixed in 23.05 before release. The 22.03 release was really pain in the ass due to this bug. Such stability issues can really ruin the experience of OpenWrit for a lot of users.

damige, 2023-07-16:
After an extra week of testing i have not had this ARP issue reappear (after updating to 23.05 rc2). This used to happen at least once a day for me.
Why it did happen again (just once) after updating is a mystery to me. It may be unrelated. I am still monitoring the situation. But I am carefully optimistic! 
For those with this issue trying 23.05 rc2 might solve it..

damige, 2023-07-22:
Update: unfortunately it still keeps happening. Currently testing with AX off "iwlwifi.disable_11ax=true" on client.

fluboi, 2023-07-22:
Yes this issue is still there in 23.05.0-rc2. 
Exact same issue seen on several different routers:
- MikroTik hAP ac2 (5Ghz, Qualcomm Atheros IPQ4018)
- TP Link Archer C6 v2 (5Ghz, Qualcomm Atheros QCA9886)
- Netgear WAX206 (5Ghz MediaTek MT7915AN)

And with different clients, at least: 
- Laptop using Intel Wi-Fi 6 AX200NGW NIC with iwlwifi driver on Debian
- Smartphone Xiaomi Mi 11 Lite 5G running android 12

At some point the communication between the AP and the client become unidirectional (client do not receive traffic from the AP. AP still receive traffic from the client.)
It happen several times a day. 
With stock firmware and comparable wifi settings the issue do not appear at all.

larena1, 2023-07-22:
Are you also experiencing this immediately after the router booted on the 5ghz interface while the 2ghz works just fine?

This is a huge issue and apparently nobody is looking into it. 

Next steps I thought about could be capturing wifi frames on another device to make sure ARP packets are never actually relayed to other stations. If they ain't, add some code to the driver's tx routine, bisect the skbs and log all ARP packets received from the kernel for transmission. If the driver never receives the according ARP packets, then most likely the packets are being lost somewhere in the kernel's network stack.

I can see the packets though when tcpdumping the wifi interface. It'll show received ARP response from a wifi station and shortly after the same response that's then supposed to be relayed back to other wifi stations connected to the same interface but never makes it there.

fluboi, 2023-07-22:
Indeed my SSID is configured only on 5Ghz interface. 
I'm not 100% sure if it's affect 2.4Ghz. I'm usually not using it, as in my urban area this band is saturated so I  get poor bandwidth and random latency issues.

The issue also appear when sitting right next to the AP. So not link with poor coverage as I assumed first. 

I'm not able to do more than tcpdump. Debugging kernel network stack clearly require some hard skills !

However, I'm wondering if we are not mixing different issue in this thread. Is your issue only affecting ARP traffic ? 
I mean, if you set ARP entry by hand, does it fix the issue ? (using `ip neigh add` on the client if its a linux thing)

In my case, it is not specific to ARP. Of course, we the issue happen we see a lot of ARP request not getting any response, but manually adding ARP entry do not help. Ethernet connectivity is broken.

damige, 2023-07-23:
Some Observations:

- Issue occurs since 21.02 until 23.05-rc2
- Occurs on 5Ghz
- Distance to AP is not a factor
- Occurs on multiple platforms (at least MT79)
- 'Layer 1' connection remains up when issue occurs
- Occurs seemingly randomly (I have had it 30seconds after boot, sometimes not for hours)
- Multiple reports on the AX200 nic, though other devices are reported too.
- When issue occurs TX (from the AP) unicast traffic recovers before RX broadcast does. Seen in de dumps where:
1.  Unicast traffic from the GW **is** received. 
2.  ARP requests (broadcasts) are sent, for the GW address 
3.  unicast ARP responses are generated on the GW (and observed by me there, and by @larena1 on the router wifi itself)
4.  ARP responses are not received on the client

Key question 1: 

- During the issue: Why is the unicast ARP reply dropped while the unicast UDP/TCP (retries) are still received? Is openwrt handling ARP packets differently? And why should it do that when (in my case) its just a bridge.

New (to me) information:

- ARP on linux does not refresh its entries on a timer, rather it gets refreshed when a matching packet is received. [https://www.google.com/search?q=how+is+arp+refreshed+in+linux](url)
- When a matching packet is not received the entry is flushed between 15-45 sec. [https://www.baeldung.com/linux/arp-settings](url)
- When tcpdumping my client i can confirm no ARP requests are generated. 
- ARP is generated when flushing 'ip -statistics neighbor flush all'

This leads me to conclude that its not a broadcast issue, but rather a whole layer 2 failure in (at least) the transmit direction from the AP.

Key question 2:
- What causes AP TX traffic to stop for at least 15-45 seconds?

Like mentioned before i'm currently testing disabling AX on the client and have not had this issue (yet). Would someone kindly confirm this? Kernel setting " iwlwifi.disable_11ax=true " works for my AX200 intel nic. 
Speed has not been impacted much in my case by dropping down to AC.

larena1, 2023-07-23:
I'm at a loss here. Added some debug logging to mac80211 to find out if it hands over the ARP request to the driver and it does. Added some logging to ath11k to find out if it receives the ARP request and it also does.

```
 kern.warn kernel: [  190.864427] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  190.876539] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  191.665545] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  191.677680] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  192.475965] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  192.488070] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  193.262908] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  193.275055] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  194.064041] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  194.076191] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  196.989467] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  197.001603] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  197.743986] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  197.756111] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  198.548281] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  198.560419] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  199.352533] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  199.364633] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  200.142779] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  200.154905] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  200.948989] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  200.961113] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  202.493252] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  202.505376] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  203.261556] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  203.273686] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  204.063905] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  204.076034] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  204.866737] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  204.878856] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  205.665799] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  205.677917] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  206.464954] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  206.477072] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  208.007644] ATH11K ARP OP REQUEST Device: phy1 Source: MAC2 IP2 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  208.007734] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC2 IP2 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  208.093028] ATH11K ARP OP REPLY Device: phy1 Source: MAC2 IP2 Destination: MAC3 TIP1
 kern.warn kernel: [  208.093103] ATH11K 3 ARP OP REPLY Device: phy1 Source: MAC2 IP2 Destination: MAC3 TIP1
 kern.warn kernel: [  209.391669] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  209.403788] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  210.133829] ATH11K ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
 kern.warn kernel: [  210.145951] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
[... cut ...]
 kern.warn kernel: [  211.762626] ATH11K 3 ARP OP REQUEST Device: phy1 Source: MAC1 SIP1 Destination: 00:00:00:00:00:00 TIP1
```

Somewhere inbetween MAC2, a wired device, sends an ARP request that's being answered just fine while all the ARP requests from a WiFi device to another WiFi device remain unanswered and can't be seen on any WiFi device connected to that same 5ghz radio, while they can be seen on the 2ghz radio. If ath11k hands over the frames to the firmware, then at least in the case of ath11k, this could be potentially be a firmware issue of some sort.

damige, 2023-07-23:
@larena1 what client wifi chipset are you using?

larena1, 2023-07-23:
@damige different ones. Issue is the same on multiple computers, phone, TV etc.

Just tried setting frame_mode=2 for ath11k to force hardware encap but that didn't change anything. Still no ARP immediately after boot on the 5ghz radio. Decided to give crypto_mode=1 a try to disable hardware encryption and that worked. Rebooted the router twice and both ARP and IPv6 immediately work flawlessly on my mobile phone and other wireless devices whereas they never did before. Software encryption is definitely not ideal but I rather prefer the performance hit then the constant multicast issues.

Still a bit too early to be sure but will see how it goes.

damige, 2023-07-23:
> @damige different ones. Issue is the same on multiple computers, phone, TV etc.
> 
> Just tried setting frame_mode=2 for ath11k to force hardware encap but that didn't change anything. Still no ARP immediately after boot on the 5ghz radio. Decided to give crypto_mode=1 a try to disable hardware encryption and that worked. Rebooted the router twice and both ARP and IPv6 immediately worked flawlessly on my mobile phone and other wireless devices whereas they never did before. Software encryption is definitely not ideal but I rather prefer the performance hit than the constant multicast issues.
> 
> Still a bit too early to be sure but will see how it goes.

disabling ax on my intel ax200 has been doing well for 2 days. "iwlwifi.disable_11ax=true" in kernel parameters. Could you try that?

larena1, 2023-07-23:
I'm not using Linux on any of my computers so I can't set that. I experienced it always immediately after boot but was able to get it work when running `wifi` once or twice on the router. After that, the issue would manifest again sometimes after hours or even up to 2 weeks later.

Recently I was almost completely unable to fix the after boot issue and multicast just remained unusable on the 5ghz radio no matter how many times I rebooted the device or ran `wifi`. It was driving me absolutely nuts as I use to control my TV from my phone which obviously didn't work as they're both connected to 5ghz. Might be a coincidence but after disabling hardware encryption it's all working now. That'd also explain why the frames reach the tx path of ath11k but never the stations.

Will leave it like this now and see if the issue ever reappears. Definitely my last Qualcomm router anyway.

JsBergbau, 2023-07-24:
> **Observations:**
> 
>     * Issue occurs since 21.02 until 23.05-rc2
> 
>     * Occurs on 5Ghz
> 
>     * Distance to AP is not a factor
> 
>     * Occurs on multiple platforms (at least MT7x)
> 
>     * 'Layer 1' connection remains up when issue occurs
> 
>     * Occurs seemingly randomly (I have had it 30seconds after boot, sometimes not for hours)
> 
>     * Multiple reports on the AX200 nic, though other devices are reported too.
> 
>     * No significant logs occur on openwrt, nor on client dmesg/NetworkManager/wpa_supplicant
> 
>     * When issue occurs TX (from the AP) unicast traffic recovers before RX (broadcast) does. Seen in de dumps where:
> 
> 
>     1. ARP requests (broadcasts) are sent, for the GW address
> 
>     2. Unicast traffic from the GW **is** received.
> 
>     3. Unicast ARP responses are generated on the GW (and observed by me on the GW, and by @larena1 on the openwrt AP wifi itself)
> 
>     4. ARP responses are not received on the client
> 
> 
> **Key question 1:** _During the issue: Why is the unicast ARP reply lost while the unicast UDP/TCP (retries) are still received?_
> 
> **Key question 2:** _Do ARP replies use the GTK key? I don't believe this ought to be the case, as they are unicast_ _If they do use GTK however, could it be the that when recovering from this issue the PTK key is correct but the GTK key is wrong? [link](http://giantsnerdwifi.blogspot.com/2017/04/broadcast-key-rotation-in-wpa2.html)_ _[link](https://forum.openwrt.org/t/driver-cannot-safely-rekeying-with-801-1x/94109/10) states that "GTK rekeying can happen when a station drops from the network"_
> 
> **New (to me) information:**
> 
>     * ARP on linux does not refresh its entries on a timer, rather it gets extended when a matching packet is received. [link](https://man7.org/linux/man-pages/man7/arp.7.html)
> 
>     * When a matching packet is not received the entry is flushed between 15-45 sec. [link](https://www.baeldung.com/linux/arp-settings)
> 
>     * It gets flushed faster if higher level protocols (e.g TCP) start to fail [link](https://stackoverflow.com/questions/10225800/what-is-the-base-reachable-time-arp-related-proc-entry)
> 
>     * When tcpdumping my client (on a working connection) i can confirm no additional ARP requests are generated for the GW mac.
> 
>     * Additional ARP are generated when flushing 'ip -statistics neighbor flush all'
> 
> 
> This leads me to conclude that its not just a broadcast(ARP) issue, but rather a whole layer 2 failure in (at least) the transmit direction from the AP.
> 
> **Key question 3:** _What causes Layer2 AP WIFI TX traffic to stop?_
> 
> **TLDR:**
> 
>     1. AP TX Layer 2 traffic stops. (which causes the GW ARP entry to be stale/invalidated, due to (lack of?) signalling from "higher level protocols" [link](https://stackoverflow.com/questions/10225800/what-is-the-base-reachable-time-arp-related-proc-entry)
> 
>     2. AP TX (unicast) recovers (without user intervention)
> 
>     3. AP TX (unicast,GW ARP reply) recovers later (without user intervention) and ARP reply from GW is received.
>        Time between 1 and 2 is significant. 50-200+ sec observed. (which could be the GTK re-key time, if ARP unicast replies used GTK?)

This issue also affects 2,4 GHz. I have only 2,4 GHz devices since they significantly use less power than the other devices I found.

The connectivity problem was fixed when I added a static ARP table entry for the router. But I did this only for testing purposes, because when replacing the router probably I won't think about it which would lead to other problems. Still it could be that it is a whole Layer2 problem. I also had the case where I couldn't reach the wifi devices from my PC. Adding the static ARP table entry "fixed" it as well.

fluboi, 2023-07-27:
Indeed, `iwlwifi disable_11ax=Y` on the client seems to be a working workaround. 

I've seen the issue using MikroTik hAPac2, witch is an AC only access-point... 

It might be an issue on the client client side more than on the AP side. 
However, with stock firmware on the AP, no problem...

damige, 2023-07-27:
> Indeed, `iwlwifi disable_11ax=Y` on the client seems to be a working workaround.
> 
> I've seen the issue using MikroTik hAPac2, witch is an AC only access-point...
> 
> It might be an issue on the client client side more than on the AP side. However, with stock firmware on the AP, no problem...

Interesting. So then it seems to be:
_An AX capable client (such as the intel ax200) does something "AX", that creates this issue on multiple openwrt (AC and AX) platforms/architectures._
I have had no problems for 5 days now with disabling ax in iwlwifi.

kristskrilovs, 2023-08-07:
I also have this issue on wrx36 with most recent stable ath11k firmware WLAN.HK.2.9.0.1-01862-QCAHKSWPL_SILICONZ-1.

I use 'multicast_to_unicast_all' as a workaround.

rmandrad, 2023-08-07:
for those on the ipq8074x target ... I went back to the WLAN.HK.2.7.0.1-01744 firmware and this issue has gone (now for 2 days ...) worth perhaps if you can try to check on your builds

a7ypically, 2023-08-07:
Same issue here on ipq8074.
I noticed wpa rekey messages at the same time the issue starts. So for testing purposes I disabled wpa group rekey and for the last 3 weeks the issue has not occurred.
This does weakens security, so enable with care. If this prevents the issue for others it may help pin point the cause.
To disable rekey add in /etc/config/wireless under each -iface section
option wpa_group_rekey '0'
And restart router.

kristskrilovs, 2023-08-07:
> option wpa_group_rekey '0'

Great find!

It may have fixed [this issue](https://forum.openwrt.org/t/ath11k-possible-firmware-bug-traffic-interruptions-when-a-client-disconnects-from-wlan/167400/31) as well. Too early to tell but I can't seem to repro that one with this change.

I'll take weaker security over firmware/driver bugs at this point.

JsBergbau, 2023-08-08:
I wanted to try this setting, but need some information first.

Where is this multicast_to_unicast_all setting documented? A google search reveals very little, including this thread. 

Where is it in the Luci Configuration GUI? Where exactly has it to be set? According to https://forum.openwrt.org/t/multicast-to-unicast/44318/7 it has to be enabled on LAN interface and not the Wifi. The only setting that I can find in the GUI is "Enable multicast support" on the "br-lan" device.

kristskrilovs, 2023-08-08:
`config wifi-iface 'default_radio0'`
`  option multicast_to_unicast_all '1'`

Not sure about LuCi, but you can check /lib/netifd/hostapd.sh. It uses this option to set multicast_to_unicast=1 in generated hostapd conf file. Documentation is in [hostapd.conf example](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf).

JsBergbau, 2023-08-09:
Thanks, you can check if it works in /tmp/run/hostapd-phy0.conf 
If so, there must be a `multicast_to_unicast=1`
So you have to to this for every wifi you are running. Very confusing that the option has been renamed to multicast_to_unicast_all whereas in hostapd-phy0.conf it remains just `multicast_to_unicast`

JsBergbau, 2023-08-09:
From https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf

> # Multicast to unicast conversion
> # Request that the AP will do multicast-to-unicast conversion for ARP, IPv4, and
> # IPv6 frames (possibly within 802.1Q). If enabled, such frames are to be sent
> # to each station separately, with the DA replaced by their own MAC address
> # rather than the group address.
> 

So if `multicast_to_unicast=1` solves the problem it looks like there is a problem with the group key. If devices have wrong group key, they can't decode the ARP request and thus causing the trouble.

I had group key interval of 3600 seconds. Did anyone try with shorter periods? Maybe that prevents the issue?

kristskrilovs, 2023-08-09:
> So you have to to this for every wifi you are running.

You don't have to. You only have to if you want this functionality (or workaround in this case). For example I have IPv4 only guest network where I isolate clients.

> Very confusing that the option has been renamed to multicast_to_unicast_all whereas in hostapd-phy0.conf it remains just `multicast_to_unicast`

This was done on purpose: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=ab61232b0abe2158c346f9fbf17f34e64de5288a
There are a lot of options where openwrt config name differs from name in respective generated config file.

Btw, I just checked LuCi and the option is there under advanced settings - "Multi To Unicast".

rany2, 2023-08-10:
@kristskrilovs The option in Luci is wrong unless you're on a very recent SNAPSHOT build: https://github.com/openwrt/luci/issues/6472

larena1, 2023-08-10:
Setting wpa_group_rekey to 0 on ath11k has no effect for me. After reboot, 5g interface comes up with no IPv6 and ARP. Enabling multicast_to_unicast_all works reliably though. 

Could in fact be an encryption issue. The fact that it also works with hardware encryption disabled also suggests that. 

@JsBergbau try opening an issue on the ath11k bug tracker. This might need to be fixed by Qualcomm in their firmware, although it is unclear if and when truck they'll ever look into it.

JsBergbau, 2023-08-10:
I am using since about 24 hours the multicast to unicast option. Still I had to use the `wifi` command several times this day. So obviously this is not the solution.

larena1, 2023-08-10:
Are you using multicast_to_unicast or multicast_to_unicast_all?

kristskrilovs, 2023-08-10:
> for those on the ipq8074x target ... I went back to the WLAN.HK.2.7.0.1-01744 firmware and this issue has gone (now for 2 days ...) worth perhaps if you can try to check on your builds

I can confirm this. Downgraded ath11k firmware files on my wrx36 device to WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1. So far broadcast/multicast works fine with default config, without any workarounds. Will keep testing with this older firmware. Hopefully it's stable.

So this issue appears to be regression in 2.9.0.1 firmware starting with WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1. The only workaround that works for me with 2.9.0.1 is multicast_to_unicast_all as documented earlier.

JsBergbau, 2023-08-11:
> Are you using multicast_to_unicast or multicast_to_unicast_all?

I'm using multicast_to_unicast and veryfied via 
``` 
cat /tmp/run/hostapd-phy0.conf | grep unicast
multicast_to_unicast=1
multicast_to_unicast=1
multicast_to_unicast=1
multicast_to_unicast=1
```
that the option is used.

larena1, 2023-08-11:
Either way it seems that Qualcomm would have to be made aware of this to investigate and hopefully fix the issue in their firmware.

JsBergbau, 2023-08-11:
I have MediaTek MT76x8 802.11bgn so nothing to do with Qualcomm.

[115 more replies]