openwrt/openwrt #9556: openwrt编译出错

cchuycchuy, 2022-03-26:
Enabling system
Enabling ucitrack
Enabling uhttpd
Enabling umount
Enabling urandom_seed
Enabling urngd
Enabling wpad
sed -i "s/Installed-Time: .*/Installed-Time: 1648256565/" /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/boot /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/* /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst* /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/* /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
find /home/cchuycchuy/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/ -mindepth 1 -execdir touch -hcd "@1648256565" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2020.824.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/cchuycchuy/openwrt'
package/Makefile:111: recipe for target '/home/cchuycchuy/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/cchuycchuy/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/cchuycchuy/openwrt'
Build failed - please re-run with -j1 to see the real error message
/home/cchuycchuy/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1

aparcar, 2022-03-26:
Please write issues in English.

Paull, 2022-03-28:
```
'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2020.824.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find. Please remove that entry from $PATH
```

981213, 2022-03-31:
The solution to this issue has been provided above by @Paull . Issue closed.

closed 2022-03-31 by 981213