openwrt/openwrt #9559: Luci Attended-Sysupgrade: Resource not Found error

t3dium, 2022-03-26:
I'm trying to update openwrt in case it fixes #9539, but attended sysupgrade stopped working, and gives a "not found error" I was able to use it in the past..

screenshot of error:
[image]

when pinging the sysupgrade domain, no issues there:
[image]

ynezz, 2022-04-02:
https://github.com/openwrt/luci is the correct place.

closed 2022-04-02 by ynezz

Hielyr, 2022-09-03:
> https://github.com/openwrt/luci is the correct place.

Y'know, I don't think it is. 

I have this same error and if https://github.com/openwrt/luci is the correct address for Attended Sysupgrade, why does https://sysupgrade.openwrt.org/ still exist and have this text on the page?

> This server listens for requests to build custom OpenWrt sysupgrade images. It starts an ImageBuilder process for the specified target, incorporates any requested packages, and returns that image to the requestor.